lvgl/lvgl #2034: Segmentation fault when cleaning list during scroll

deltabeard, 2021-01-23:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Cleaning all buttons within a list within a button click event callback, whilst the list is still scrolling after a flick-up or flick-down input gesture, causes a segmentation fault due to a use-after-free of `drag_obj`.

```
=================================================================
==303784==ERROR: AddressSanitizer: heap-use-after-free on address 0x60f00004006c at pc 0x55cb42fed289 bp 0x7ffd552dd9b0 sp 0x7ffd552dd9a0
READ of size 1 at 0x60f00004006c thread T0
    #0 0x55cb42fed288 in scrlbar_refresh inc/lvgl/src/lv_widgets/lv_page.c:1294
    #1 0x55cb42feac3d in lv_page_scrollable_signal inc/lvgl/src/lv_widgets/lv_page.c:958
    #2 0x55cb42fc0087 in lv_cont_refr_autofit inc/lvgl/src/lv_widgets/lv_cont.c:795
    #3 0x55cb42fbdadd in lv_cont_signal inc/lvgl/src/lv_widgets/lv_cont.c:278
    #4 0x55cb42fea6d0 in lv_page_scrollable_signal inc/lvgl/src/lv_widgets/lv_page.c:913
    #5 0x55cb42fc00d2 in lv_cont_refr_autofit inc/lvgl/src/lv_widgets/lv_cont.c:798
    #6 0x55cb42fbdadd in lv_cont_signal inc/lvgl/src/lv_widgets/lv_cont.c:278
    #7 0x55cb42fb309b in lv_btn_signal inc/lvgl/src/lv_widgets/lv_btn.c:267
    #8 0x55cb42fe0980 in lv_list_btn_signal inc/lvgl/src/lv_widgets/lv_list.c:797
    #9 0x55cb42fdeb4d in lv_list_add_btn inc/lvgl/src/lv_widgets/lv_list.c:232
    #10 0x55cb42f1f4ec in fill_fp_list src/main.c:273
    #11 0x55cb42f1ef2f in recreate_filepicker src/main.c:204
    #12 0x55cb42f1efd5 in btnev_chdir src/main.c:220
    #13 0x55cb42f36d05 in lv_event_send_func inc/lvgl/src/lv_core/lv_obj.c:1968
    #14 0x55cb42f368f8 in lv_event_send inc/lvgl/src/lv_core/lv_obj.c:1880
    #15 0x55cb42f2b741 in indev_proc_release inc/lvgl/src/lv_core/lv_indev.c:1038
    #16 0x55cb42f26f30 in indev_pointer_proc inc/lvgl/src/lv_core/lv_indev.c:417
    #17 0x55cb42f25993 in _lv_indev_read_task inc/lvgl/src/lv_core/lv_indev.c:110
    #18 0x55cb42f9a0f2 in lv_task_exec inc/lvgl/src/lv_misc/lv_task.c:386
    #19 0x55cb42f99615 in lv_task_handler inc/lvgl/src/lv_misc/lv_task.c:134
    #20 0x55cb42f217d3 in ui_render src/main.c:766
    #21 0x55cb42f218d2 in main src/main.c:795
    #22 0x7f7e9cb0e151 in __libc_start_main (/usr/lib/libc.so.6+0x28151)
    #23 0x55cb42f1d55d in _start (/home/mahyar/projects/haiyajan-lvgl/Haiyajan-lvgl+0x1955d)

0x60f00004006c is located 108 bytes inside of 176-byte region [0x60f000040000,0x60f0000400b0)
freed by thread T0 here:
    #0 0x7f7e9cf180e9 in __interceptor_free /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:123
    #1 0x7f7e9cd505ce  (/usr/lib/libSDL2-2.0.so.0+0x7f5ce)
    #2 0x55cb42f3d77e in obj_del_core inc/lvgl/src/lv_core/lv_obj.c:3827
    #3 0x55cb42f31423 in lv_obj_del inc/lvgl/src/lv_core/lv_obj.c:485
    #4 0x55cb42f315ed in lv_obj_clean inc/lvgl/src/lv_core/lv_obj.c:532
    #5 0x55cb42fde67e in lv_list_clean inc/lvgl/src/lv_widgets/lv_list.c:134
    #6 0x55cb42f1ef23 in recreate_filepicker src/main.c:203
    #7 0x55cb42f1efd5 in btnev_chdir src/main.c:220
    #8 0x55cb42f36d05 in lv_event_send_func inc/lvgl/src/lv_core/lv_obj.c:1968
    #9 0x55cb42f368f8 in lv_event_send inc/lvgl/src/lv_core/lv_obj.c:1880
    #10 0x55cb42f2b741 in indev_proc_release inc/lvgl/src/lv_core/lv_indev.c:1038
    #11 0x55cb42f26f30 in indev_pointer_proc inc/lvgl/src/lv_core/lv_indev.c:417
    #12 0x55cb42f25993 in _lv_indev_read_task inc/lvgl/src/lv_core/lv_indev.c:110
    #13 0x55cb42f9a0f2 in lv_task_exec inc/lvgl/src/lv_misc/lv_task.c:386
    #14 0x55cb42f99615 in lv_task_handler inc/lvgl/src/lv_misc/lv_task.c:134
    #15 0x55cb42f217d3 in ui_render src/main.c:766
    #16 0x55cb42f218d2 in main src/main.c:795
    #17 0x7f7e9cb0e151 in __libc_start_main (/usr/lib/libc.so.6+0x28151)

previously allocated by thread T0 here:
    #0 0x7f7e9cf18459 in __interceptor_malloc /build/gcc/src/gcc/libsanitizer/asan/asan_malloc_linux.cpp:145
    #1 0x7f7e9cd504e3  (/usr/lib/libSDL2-2.0.so.0+0x7f4e3)
    #2 0x55cb42f937a9 in _lv_ll_ins_head inc/lvgl/src/lv_misc/lv_ll.c:74
    #3 0x55cb42f3033d in lv_obj_create inc/lvgl/src/lv_core/lv_obj.c:308
    #4 0x55cb42fbcc7d in lv_cont_create inc/lvgl/src/lv_widgets/lv_cont.c:77
    #5 0x55cb42fb27e3 in lv_btn_create inc/lvgl/src/lv_widgets/lv_btn.c:65
    #6 0x55cb42fde753 in lv_list_add_btn inc/lvgl/src/lv_widgets/lv_list.c:162
    #7 0x55cb42f1f4ec in fill_fp_list src/main.c:273
    #8 0x55cb42f1ef2f in recreate_filepicker src/main.c:204
    #9 0x55cb42f1f701 in btnev_dirup src/main.c:292
    #10 0x55cb42f36d05 in lv_event_send_func inc/lvgl/src/lv_core/lv_obj.c:1968
    #11 0x55cb42f368f8 in lv_event_send inc/lvgl/src/lv_core/lv_obj.c:1880
    #12 0x55cb42f2b8f5 in indev_proc_release inc/lvgl/src/lv_core/lv_indev.c:1057
    #13 0x55cb42f26f30 in indev_pointer_proc inc/lvgl/src/lv_core/lv_indev.c:417
    #14 0x55cb42f25993 in _lv_indev_read_task inc/lvgl/src/lv_core/lv_indev.c:110
    #15 0x55cb42f9a0f2 in lv_task_exec inc/lvgl/src/lv_misc/lv_task.c:386
    #16 0x55cb42f99615 in lv_task_handler inc/lvgl/src/lv_misc/lv_task.c:134
    #17 0x55cb42f217d3 in ui_render src/main.c:766
    #18 0x55cb42f218d2 in main src/main.c:795
    #19 0x7f7e9cb0e151 in __libc_start_main (/usr/lib/libc.so.6+0x28151)

SUMMARY: AddressSanitizer: heap-use-after-free inc/lvgl/src/lv_widgets/lv_page.c:1294 in scrlbar_refresh
Shadow bytes around the buggy address:
  0x0c1e7fffffb0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fffffc0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fffffd0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7fffffe0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c1e7ffffff0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c1e80000000: fd fd fd fd fd fd fd fd fd fd fd fd fd[fd]fd fd
  0x0c1e80000010: fd fd fd fd fd fd fa fa fa fa fa fa fa fa fd fd
  0x0c1e80000020: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c1e80000030: fd fd fd fd fa fa fa fa fa fa fa fa fd fd fd fd
  0x0c1e80000040: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c1e80000050: fd fd fa fa fa fa fa fa fa fa fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==303784==ABORTING
```

### To Reproduce

Writing a minimum example hasn't been successful mainly because user interaction is required or emulated.

```c
/* List button event handler. */
void btnev_chdir(lv_obj_t *btn, lv_event_t event)
{
	const char *dir = lv_list_get_btn_text(btn);
	struct ui_ctx *ui_ctx = lv_obj_get_user_data(btn);

	if(event != LV_EVENT_CLICKED)
		return;

	/* ui_ctx contains a pointer to the list object. */
	lv_list_clean(ui_ctx->fp_list);
	/* Segmentation fault occurs here if the list is still scrolling. */

	return;
}
```

### Expected behavior
To be able to remove all of the buttons, and then add new buttons within the event handler.

### Workaround

```c
void btnev_chdir(lv_obj_t *btn, lv_event_t event)
{
	const char *dir = lv_list_get_btn_text(btn);
	struct ui_ctx *ui_ctx = lv_obj_get_user_data(btn);

	if(event != LV_EVENT_CLICKED)
		return;

	lv_scrollbar_mode_t scrollbar_mode = lv_list_get_scrollbar_mode(ui_ctx->fp_list);
	lv_list_set_scrollbar_mode(ui_ctx->fp_list, LV_SCROLLBAR_MODE_OFF);
	lv_task_handler();

	lv_list_clean(ui_ctx->fp_list);
	lv_task_handler();

	fill_fp_list(ui_ctx);
	lv_list_set_scrollbar_mode(ui_ctx->fp_list, scrollbar_mode);
	lv_task_handler();
}
```

### Context
I'm using a list object as a file picker. When a folder is selected, the list is cleaned and folders and files within the accessed folder is added to the list.

embeddedt, 2021-01-23:
The problem here appears to be that events don't check for object deletion till you return from the event handler. That is probably difficult to fix unless we check for any references to the object from input devices in `lv_obj_del`.

A cleaner workaround is probably to use [`lv_async_call`](https://docs.lvgl.io/latest/en/html/overview/task.html#asynchronous-calls) and create the buttons on the next tick.

deltabeard, 2021-01-24:
Thanks! That's a better workaround.

kisvegabor, 2021-01-25:
As the issue seems to be solved I close it.

closed 2021-01-25 by kisvegabor