lvgl/lvgl #2036: Matrix Multiplication

oras, 2021-01-25:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

oras, 2021-01-25:
I want to run filter such as Kenny on an image
Thanks

embeddedt, 2021-01-25:
The canvas is probably the easiest to run custom filters on. We do not directly support matrix transforms on widgets right now.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2021-01-25 by embeddedt