lvgl/lvgl #2038: Double Buffer optimization for SPI interface screen

FASTSHIFT, 2021-01-25:
### Introcude the problem
When using the SPI screen, the benchmark score of two full-screen buffers is lower than that of using two half-screen buffers.

### Examples and cases
Two half-screen buffers:
[image]

Two full-screen buffers:
[image]

### Suggested solution
I modified a function of lv_hal_disp.c, and the benchmark score of the two full-screen buffers has been improved.
```C
bool lv_disp_is_true_double_buf(lv_disp_t * disp)
{
    return false;
    /*uint32_t scr_size = disp->driver.hor_res * disp->driver.ver_res;

    if(lv_disp_is_double_buf(disp) && disp->driver.buffer->size == scr_size) {
        return true;
    }
    else {
        return false;
    }*/
}
```
[image]

Is it possible to add a macro switch to distinguish whether it is an SPI screen and give full play to the performance of the two full-screen buffers.

kisvegabor, 2021-01-27:
I agree it'd be less confusing to add a config option. 

However, for backward compatibility, we should keep the current behavior in v7.

I'll add it in v8.

FASTSHIFT, 2021-04-14:
@kisvegabor 
Does lvgl v8 support this feature now?
It is optimized for the dual full-screen buffer of the SPI screen.

kisvegabor, 2021-04-14:
I've just added the `full_refresh` flag to the display driver. IF enabled the whole screen will be redrawn even if only 1 pixel has changed. 
By default it's false.

commit e4345bd7 (2021-04-14): feat(disp) add full_refresh flag to the display driver
It replaces the previous concept of true double buffering when the whole screen was refreshed if the 2 draw buffers had the same size

Fixes: #2038
closed 2021-04-14 by kisvegabor