lvgl/lvgl #2041: issue for cpicker

ZhaoQiang-b45475, 2021-01-27:
Hi,
I create a cpicker and set it to rect, met a issue shown as capture(for detail, please refer to zip which is a video).
The color selecting area become the same color sometimes and couldn't change back. Is it the issue of lvgl or system's cpicker?
[image]
[color_media.zip](https://github.com/lvgl/lvgl/files/5877541/color_media.zip)

embeddedt, 2021-01-27:
Are you referring to how it switches to saturation mode instead of the default hue mode? That is triggered by holding down the knob for a long time, and can be disabled using `lv_cpicker_set_color_mode_fixed(cpicker, true)`.

ZhaoQiang-b45475, 2021-02-01:
Thanks for your explanation!

kisvegabor, 2021-02-01:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2021-02-01 by kisvegabor