lvgl/lvgl #2043: Contribute a project for building native Windows app with LVGL [WIP]

MouriNaruto, 2021-01-27:
### Introcude the problem

I had found there were something I can make some improvement after trying simulator projects which support Windows under the LVGL organization. I want to contribute a separate project called [lv_port_windows](https://github.com/MouriNaruto/lv_port_windows) because the following reasons.

- I don't want to make a pure simulator only. I want to make a project for helping more people to build native Windows app with LVGL.
- I usually use Visual Studio. But I don't want to contribute my implementation to [lvgl/lv_sim_visual_studio_sdl](https://github.com/lvgl/lv_sim_visual_studio_sdl) because my implementation don't depend on SDL.

Here are some differences from [lvgl/lv_sim_visual_studio_sdl](https://github.com/lvgl/lv_sim_visual_studio_sdl).

- Only depends on Win32 API, C Runtime and C++ STL only.
- Support compiling with [VC-LTL](https://github.com/Chuyu-Team/VC-LTL) toolchain make the binary size as smaller as using MinGW.
- Support resizing the Window in the HAL level.
- Support Windows keyboard and mouse wheel event in the HAL level.

But lv_port_windows and LVGL itself still have the following things to do.

- lv_port_windows should support Per Monitor DPI Aware in the HAL level.
- LVGL should add the support for adaptive UI. (I had tested with lv_demo_widgets, but I have found LVGL doesn't support widgets resizing (width, height and dpi) after I called lv_disp_drv_update to update. I will provide some screenshots in this issue for reference.)
- LVGL should add the support for customizing the keyboard and mousewheel driver without any modifications in lv_demo_keypad_encoder. (I don't want to use indev implementation in lv_drivers project because my implementation don't depend on SDL.)

### Examples and cases

Here is the screenshot of lv_port_windows.
[image]

Here is the screenshot of I mention "I have found LVGL doesn't support widgets resizing".
[image]

### Suggested solution

I want move lv_port_windows repository under LVGL organization because I want to help more people.

Kenji Mouri

embeddedt, 2021-01-27:
The resizing issue is not an LVGL limitation, but rather a limitation of the examples. They don't check for the screen resizing because that almost never happens in an embedded product (barring rotation). In a real application this could easily be solved by resizing/repositioning widgets when the screen size changes.

Do you think you could merge your Win32 API driver changes to https://github.com/lvgl/lv_drivers/blob/master/win_drv.c? At first glance it looks like your driver can easily be adapted to C as it's not using any major C++ features. That driver is used by https://github.com/lvgl/lv_sim_codeblocks_win which depends on CodeBlocks and MinGW only (SDL is not required).

Since the project itself is targetting Windows only, I have no problem with the rest remaining as C++. I'm just hoping to have one unified driver.

MouriNaruto, 2021-01-27:
@embeddedt Thank you for your apply.

Yes, I can create a PR for updating win_drv in lv_drivers. 

But I hope LVGL can provide a universal way to use the keyboard and mouse wheel driver. (Because I also want to add the keyboard and mouse wheel support for win_drv.)

I found lv_demo_keypad_encoder example create a keyboard and a mouse wheel driver by itself. (It includes indev.h directly and it depends on the SDL.)

embeddedt, 2021-01-27:
Keyboards and mice can be registered [the same way](https://lvgl.io/docs/latest/en/html/porting/indev.html) as touch/mouse devices; the approach that demo uses is really a hack.

MouriNaruto, 2021-01-27:
> Keyboards and mice can be registered [the same way](https://lvgl.io/docs/latest/en/html/porting/indev.html) as touch/mouse devices; the approach that demo uses is really a hack.

@embeddedt I need to know how to provide the lv_indev_t handle for user elegantly. (Because the user need to use lv_indev_set_group with that.)

embeddedt, 2021-01-27:
The best solution I can think of is to have APIs in the Windows driver like `lv_indev_t *lv_windows_keyboard_init(void);` and `lv_indev_t *lv_windows_mousewheel_init(void);`.

MouriNaruto, 2021-01-27:
> The best solution I can think of is to have APIs in the Windows driver like `lv_indev_t *lv_windows_keyboard_init(void);` and `lv_indev_t *lv_windows_mousewheel_init(void);`.

@embeddedt If the LVGL itself can provide the similar APIs for user, it will make the driver developer easier to adapt LVGL to other platforms. (For example, I also want to provide a driver for using LVGL in a UEFI application.)

embeddedt, 2021-01-27:
Users can also loop through all registered input devices with `lv_indev_get_next`, so if someone wanted to attach a group to a keyboard, they could simply find the first registered keyboard and attach it.

MouriNaruto, 2021-01-27:
> Users can also loop through all registered input devices with `lv_indev_get_next`, so if someone wanted to attach a group to a keyboard, they could simply find the first registered keyboard and attach it.

@embeddedt 
Looks nice. I will try it and make a PR for lv_examples project for that if it's OK. (Because I need to make it adaptable first.)
And I also hope the lv_port_windows can be moved to LVGL organization in the future.

MouriNaruto, 2021-01-27:
@embeddedt I have create the PR for lv_demo_keypad_encoder demo. I hope it can be merged.

MouriNaruto, 2021-01-27:
@embeddedt Thank you for merging it. I will create a PR for Windows driver in recent days because some issues need to be solved.

But I wonder to ask some questions.

- Can I write the Windows driver with C++?
- Can I move the Windows driver to a separate folder in lv_drivers project?

embeddedt, 2021-01-27:
> Can I write the Windows driver with C++?

The other drivers use C, so I think it would be preferable to keep that for consistency. At first glance all the APIs in your current driver exist in C as well; just remove the `::` from the beginning.

> Can I move the Windows driver to a separate folder in lv_drivers project?

Sure; that's fine. I assume you want to split the display and input device components into the proper folders?

MouriNaruto, 2021-01-27:
> Sure; that's fine. I assume you want to split the display and input device components into the proper folders?

@embeddedt I only split the Windows drivers to the proper folder (maybe it called win_drv) and create a read me document.

MouriNaruto, 2021-01-27:
> At first glance all the APIs in your current driver exist in C as well; just remove the :: from the beginning.

@embeddedt Because I haven't made the code tidy. (I will try my best to use the pure C only.)

MouriNaruto, 2021-01-27:
@embeddedt
Also, I need some encouragement via moving lv_port_windows repository under LVGL organization. 😀
If it can be came true, I will very happy to contribute a better Windows driver to lv_drivers. Also I will create a UEFI driver and UEFI application demo.

embeddedt, 2021-01-27:
> Also, I need some encouragement via moving lv_port_windows repository under LVGL organization.

I can't do that without feedback from @kisvegabor. :slightly_smiling_face:

MouriNaruto, 2021-01-27:
> > Also, I need some encouragement via moving lv_port_windows repository under LVGL organization.
> 
> I can't do that without feedback from @kisvegabor. 🙂

Thanks, I will wait for that. I hope he can agree with that.

kisvegabor, 2021-01-28:
Hi @MouriNaruto 

Thanks for your interest in contributing to LVGL!

Would it be possible to merge your changes into the `lvgl/lv_sim_visual_studio_sdl` repo?
This repo can work only on widows anyway and I believe people don't care much whether it works with SDL or native drivers. 
So we can remove the `_sdl` part from the repo name and use your drivers by default. What do you think?

cc @MCF

MouriNaruto, 2021-01-28:
> Hi @MouriNaruto
>
> Thanks for your interest in contributing to LVGL!
>
> Would it be possible to merge your changes into the `lvgl/lv_sim_visual_studio_sdl` repo?
> This repo can work only on widows anyway [...]

@kisvegabor

In my opinion, I don't think removing the `_sdl` part from the repo name is a good choice because I'm afraid that it will cause lots of third-party LVGL tutorials may have dead links about that. (I don't think it's good for LVGL users.) I also think the LVGL should have a new mono repo for managing all platform ports. (It makes LVGL users easier to find the solution about the existed porting and save the time for reinventing the wheels.)

kisvegabor, 2021-01-28:
> In my opinion, I don't think removing the _sdl part from the repo name is a good choice because I'm afraid that it will cause lots of third-party LVGL tutorials may have dead links about that. (I don't think it's good for LVGL users.)

Fortunately, GitHub manages it very well. References to the old link will be redirected automatically to the new one. See [here](https://docs.github.com/en/github/administering-a-repository/renaming-a-repository):
*"In addition to redirecting web traffic, all git clone, git fetch, or git push operations targeting the previous location will continue to function as if made on the new location."*

> I also think the LVGL should have a new mono repo for managing all platform ports. (It makes LVGL users easier to find the solution about the existed porting and may save the time for reinventing the wheels.)

About a year ago we had a mono repo for ports but it was quite difficult to manage. Maybe now, with more experience, we could manage it better.  Let's discuss it as a separate topic.

MouriNaruto, 2021-01-28:
> Fortunately, GitHub manages it very well. References to the old link will be redirected automatically to the new one. See [...]

It looks good, but there are some issues about that.

First, I am worry about that he will mind about that I had changed his repository a lot. (This is why I want to make a new repository. I don't want to make trouble.) 

Second, I need the permission of commit to that repo directly. Because I don't like waiting for replying. Some open source project including the famous one will have 0 replies for a PR from a indirecly commiter until waiting for several days, months or years.  That makes me very sad, and I don't want to commit to other open source project anymore. (But I'm still an open source lover because I created several open source projects.) But when I read the documents of LVGL, I think maybe I can do something, so I create a PR for other open source projects again.

kisvegabor, 2021-01-28:
> First, I am worry about that he will mind about that I had changed his repository a lot. (This is why I want to make a new repository. I don't want to make trouble.)

I'm sure it won't be a problem. Changing the repos is natural and nothing is written in stone. And we are definitely open to reasonable changes. 

I prefer using a single repo for windows because it's easier to maintain 1 repo than 2 and it's a clearer message for the users: "If you are on Windows go to this repo"

> Second, I need the permission of commit to that repo directly.

It's everybody's interest to have more active people around the project and repos so I'd love to give write access to as many people as possible. Having dedicated maintainers on specific repos/topic is very important and we are looking for people who are willing to contribute on such a serious level.

>  Because I don't like waiting for replying.

If you look at any random issue in lvgl you can see that long wait times are absolutely not typical here. We actively monitor issues every day.

MouriNaruto, 2021-01-28:
> I'm sure it won't be a problem. Changing the repos is natural and nothing is written in stone. And we are definitely open to reasonable changes.
>
> I prefer using a single repo for windows because [...]

I agree with that. Could you rename and give me write access in that repository? I want to commit my changes ASAP.

kisvegabor, 2021-01-28:
> I agree with that. Could you rename that repository and give me write access? I want to commit my changes ASAP.

Just one thing: please confirm that the original functionality will be maintained. I.e. the project will be usable with Visual Studio out of the box, and the examples and demos will work the same way.

If so, let's wait a little for @mcf and/or @embeddedt to agree. (No worries it should happen quickly :slightly_smiling_face: ) 
Until that, you can file PR and hopefully merge it yourself.

MouriNaruto, 2021-01-28:
> > I agree with that. Could you rename that repository and give me write access? I want to commit my changes ASAP.
> 
> Just one thing: please confirm that the original functionality will be maintained. I.e. the project will be usable with Visual Studio out of the box, and the examples and demos will work the same way.
> 
> If so, let's wait a little for @MCF and/or @embeddedt to agree. (No worries it should happen quickly 🙂 )
> Until that, you can file PR and hopefully merge it yourself.

I will wait for that.

embeddedt, 2021-01-28:
If we can get rid of the SDL dependency from the Visual Studio project, while maintaining the same behavior, let's do it. :+1:

kisvegabor, 2021-01-28:
Great! Renamed and write access granted. :slightly_smiling_face:

MouriNaruto, 2021-01-28:
> Great! Renamed and write access granted. 🙂

I have accepted your invitation. Thanks.

MouriNaruto, 2021-01-28:
@kisvegabor I have found the similar issue as https://github.com/lvgl/lv_sim_visual_studio/issues/17.

kisvegabor, 2021-01-29:
Actually, I'm on Linux and only rarely use Windows.
So it's better to give it to the hands of people who really use it :slightly_smiling_face: 

Let's close this issue and continue the discussion in the [lv_sim_visual_studio](https://github.com/lvgl/lv_sim_visual_studio) if somethings come up.

Thank you @MouriNaruto and @MCF.

closed 2021-01-29 by MouriNaruto