lvgl/lvgl #2045: Roller widget not working when options>194 on ESP32

s1618, 2021-01-28:
Hi,
I try to put a simple Roller widget and it works until I put over 194 options.
The widget mid part is halfway between the last number and an empty string.
I don't think is a character count problem because when I put a lot of char on the 194th option, it still works.

Is this a bug or am I not using the widget correctly?

I tried with char* and String converter to char*.
Each option has '\n' but the last one which has '\0'.
I also update to the last version of LVGL.

Platform : TTGO watch 2020 (ESP32)
IDE : VScode w/ PlatformIO

kisvegabor, 2021-01-29:
Hi,

Please add an exact code to reproduce.

You can also try to use `typedef int32_t lv_coord_t;` (`int16_t` to `int32_t`) in `lv_conf.h`.

s1618, 2021-01-29:
`typedef int32_t lv_coord_t;` did the trick. It works. I guess this wasn't an issue per say.

`
    String test =
                "1\n"
                "2\n"
                "3\n"
                "4\n"
                "5\n"
                "6\n"
                "7\n"
                "8\n"
                "9\n"
                "10\n"
                "11\n"
                "12\n"
                "13\n"
                "14\n"
                "15\n"
                "16\n"
                "17\n"
                "18\n"
                "19\n"
                "20\n"
                "21\n"
                "22\n"
                "23\n"
                "24\n"
                "25\n"
                "26\n"
                "27\n"
                "28\n"
                "29\n"
                "30\n"
                "31\n"
                "32\n"
                "33\n"
                "34\n"
                "35\n"
                "36\n"
                "37\n"
                "38\n"
                "39\n"
                "40\n"
                "41\n"
                "42\n"
                "43\n"
                "44\n"
                "45\n"
                "46\n"
                "47\n"
                "48\n"
                "49\n"
                "50\n"
                "51\n"
                "52\n"
                "53\n"
                "54\n"
                "55\n"
                "56\n"
                "57\n"
                "58\n"
                "59\n"
                "60\n"
                "61\n"
                "62\n"
                "63\n"
                "64\n"
                "65\n"
                "66\n"
[... 223 lines cut ...]
                "290\n"
                "291\n"
                "292\n"
                "293\n"
                "294\n"
                "295\n"
                "296\n"
                "297\n"
                "298\n"
                "299\0";

    // parent is the widget's container
    char data[test.length()+1];
    test.toCharArray(data, test.length()+1);

    lv_obj_t *roller = lv_roller_create(parent, NULL);
    lv_roller_set_auto_fit(roller, false);
    lv_roller_set_align(roller, LV_LABEL_ALIGN_CENTER);
    lv_roller_set_visible_row_count(roller, 4);
    lv_roller_set_options(roller, data, LV_ROLLER_MODE_INFINITE);
    lv_obj_set_ext_click_area(roller, CLICKABLE_PADDING, CLICKABLE_PADDING, CLICKABLE_PADDING, CLICKABLE_PADDING);
    lv_obj_set_width(roller, 90);
    lv_obj_align_mid(roller,parent,LV_ALIGN_CENTER,0,0);
    lv_roller_set_selected(roller,59,false);
`

kisvegabor, 2021-02-01:
Great! For such large coordinates, the default `int16_t` coordinates can overflow.

closed 2021-02-01 by kisvegabor