lvgl/lvgl #2046: [Feature] Rotation awareness

C47D, 2021-01-29:
### Introduce the problem
LVGL doesn't know how to track the rotation of the display and/or indev.

### Examples and cases
While updating the `lvgl_esp32_drivers` repo I found difficult to know the rotation of the display without using a global variable.

### Suggested solution
Add a member on `lv_disp_drv_t` named rotation, which can have values from 0 to 3, the meaning of this member is up to the user, because it doesn't change anything inside LVGL (aka LVGL will not rotate the image, the user must do it on it's callbacks).

@kisvegabor suggested the following API
```c
lv_disp_set_orientation(disp, 0..3 or enum)
{
   disp->drv->orientation = x; //0..3

   if (disp->drv->set_orientation_cb) {
      disp->drv->set_orientation_cb(disp->drv, x);
   }

   lv_indev_refr_orientation(disp, x);   //To be implemented, swap axis
   lv_disp_refr_orientation(disp, x);     //To be implemented, swap screen dimensions
}
```

On the above snippet, the user should provide the `set_orientation_cb` callback.

What do yall think?

kisvegabor, 2021-02-01:
Thanks for opening this issue.
If there is no objection I can add it in v8.

C47D, 2021-02-03:
I don't know how complex it is, if you need me to test anything please let me know!

embeddedt, 2021-02-15:
@C47D I think the extended rotation support added in #2055 should implement everything you need; can you take a look at it? It's available in `dev`.

C47D, 2021-02-16:
@embeddedt Thanks for working on it, I will test it on the `lvgl_esp32_drivers` repo.

turoksama, 2021-03-04:
Great.
What I know about LCDs which have driver(NOT RGB) may have problems in rotations:
1.The size of  (X, Y) window is fixed at initialed stage, (X,Y) cannot be lager than presets, otherwise will be ignored.(call_back should be reserved to re-init LCD)
2.If not 1, then RAM map problem?

Beware inputs of touch screen is also rotated.

embeddedt, 2021-03-04:
@C47D

GitHub doesn't notify me when you edit the message, sorry about that...

There is no callback at the moment for detecting the rotation change. I can add one before the 7.11 release if it's needed.

@turoksama 

Touch screen rotation is [handled automagically](https://github.com/lvgl/lvgl/blob/a4c33bb16eb3b021883d049f3034d46a7f6e1809/src/lv_core/lv_indev.c#L417-L425) as of 7.11.

C47D, 2021-03-04:
Don't worry @embeddedt, I think that callback might be useful so we can use it to send a configuration command to the display, e.g. the ILI displays needs to update a register to configure how the graphical memory is dumped into the display.

fvanroie, 2021-03-28:
> Touch screen rotation is [handled automagically](https://github.com/lvgl/lvgl/blob/a4c33bb16eb3b021883d049f3034d46a7f6e1809/src/lv_core/lv_indev.c#L417-L425) as of 7.11.

I'm trying this functionality and I see some issues:

https://user-images.githubusercontent.com/15969459/112765747-dd334a00-900e-11eb-8b8b-07c741e219a9.mp4

The mouse pointer runs in circles when it's not pressed. This seems to be because `i->proc.types.pointer.last_point.x` and `.y` are set to the rotated values. The next iteration will rotate this saved `last_point` variable again and update it... etc. Making the pointer *run in circles*, 90° each pass.

Only updating the cursor coordinates to the newly calculated point without affecting `i->proc.types.pointer.last_point` seems to work for the cursor position, but it has no impact on the hit-test position. Buttons are still being pressed on the raw coordinates.

embeddedt, 2021-03-28:
I am not sure exactly what's causing this. The rotation is done [inside `indev_pointer_proc`](https://github.com/lvgl/lvgl/blob/1748798aca03a5eeca07faa1efcf49678bbf1566/src/lv_core/lv_indev.c#L417-L425) which is called [only when a new point has just been read from the touchscreen](https://github.com/lvgl/lvgl/blob/release/v7/src/lv_core/lv_indev.c#L110).

fvanroie, 2021-03-28:
`_lv_indev_read` initialises the previous position with the rotated coordinates before read_cb is called:

```c
    if(indev->driver.type == LV_INDEV_TYPE_POINTER) {
        data->point.x = indev->proc.types.pointer.act_point.x;
        data->point.y = indev->proc.types.pointer.act_point.y;
    }
```

When no touch is detected in the `read_cb` I keep the previous data->point set as is, but the `indev_pointer_proc` function still rotates it again and again.

embeddedt, 2021-03-28:
Ah... now I see what you mean. In my touch driver I always overwrite the old value; that's probably why I never saw this problem.

I wonder if it would be best to rotate immediately following the call to `read_cb` only if the new values don't match the old ones. That seems like the least invasive change.

fvanroie, 2021-03-28:
I've been wondering about that too... but even if the values seemingly didn't change, there is a very, very slim chance that the *new* raw values are *"accidentaly"* the same as the previous rotated ones. (ok, highly unlikely but playing devils advocate here).

The raw `read_cb` values and calculated values shouldn't mix because you can't tell them apart. I think there needs to be a clear separation between raw and calculated values.

As a workaround, the previous raw point can be kept in the callback or indev and always set `data->point` with its raw values.
I'm not sure where the calculation is supposed to happen for everything to adhere to the new values.

embeddedt, 2021-04-10:
@fvanroie @C47D As far as I know, rotation support should now work without problems. Do you think this issue can be closed?

fvanroie, 2021-04-10:
I still need to test this again, but that won't happen for another week or so...
For me there is no need to keep this issue open.

C47D, 2021-04-12:
I am in almost the same position as @fvanroie , I will have some free time until next weekend, let's close this and open a new issue if we found any problems.

closed 2021-04-12 by C47D

fvanroie, 2021-05-31:
Sorry for the late response. I have updated my project to include this PR and it is working as expected now.
This takes the burden of juggling input coordinates from the indev driver. Thank you very much!

Will there be an 7.11.x (or 7.12) official release containing this fix?

embeddedt, 2021-05-31:
As far as I know, another 7.x release is not planned, however, you can use the `release/v7` branch which includes 7.11 plus any bugfixes since then.

fvanroie, 2021-05-31:
That's what I'm doing now, but the `releases/v7` branch shows up as `7.8.0` in platformio... library.json and library.properties aren't up-to-date.

embeddedt, 2021-06-02:
I've tried to bump the version numbers in those files [here](https://github.com/lvgl/lvgl/commit/3bafa5a8a385d482d4c600d4e0f233f34f0407d3). I'm not sure if it will help or not.

fvanroie, 2021-06-02:
That fixed it, all good now. Thanks!

PR #2055 (2021-02-12): Better rotation support (plus experimental software rotation)
### Description of the feature or fix

This PR adds support for rotating the display to all 3 alternative orientations. The rotation can be done in hardware (like current versions) or can be done in software by enabling the `sw_rotate` flag on the display driver.

180 degree rotation done in software is extremely fast, at least on STM32F7, and I noticed little to no performance loss. 90 and 270 degree rotation, however, is considerably slower.

@kisvegabor Suggestions on how to optimize this would be appreciated if you have time to  [... cut]