lvgl/lvgl #2049: The Latest version would be broken while calling Create Lable Fiction.

HsiueMingLee, 2021-01-29:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2021-01-29:
Please provide a code sample which can be used to reproduce the issue in a simulator.

HsiueMingLee, 2021-01-29:
> Please provide a code sample which can be used to reproduce the issue in a simulator.

Version6 were working well.but latest version would broken down the CPU.

embeddedt, 2021-01-29:
Can you please provide some code and some more information about your platform? This issue is not very actionable in it's current state.

HsiueMingLee, 2021-01-30:
> Can you please provide some code and some more information about your platform? This issue is not very actionable in its current state.

STM32F407.When the cpu  running here [strcpy(ext->text, text);] which is in "lv_label_set_text" fuchtion,the whole program will 
 be down. And i replaced with [strcpy(ext->text, "i am text");] ,it still unworking. And the i comment this statement like /*strcpy(ext->text, text);*/,and followed by ext->text = "i am text",it's unworking,so I think the issues are in lv_label_ext_t or somwhere.

embeddedt, 2021-01-30:
It sounds like you have memory corruption on your platform; I don't think this is happening on the simulator.

HsiueMingLee, 2021-01-30:
> It sounds like you have memory corruption on your platform; I don't think this is happening on the simulator.

Stack_Size      EQU     0x00000400  ===》 Stack_Size      EQU     0x00000800.

Solve perfectly ！

embeddedt, 2021-01-30:
Great!

The general ballpark I would recommend is 0x1000 (4 kilobytes) if you have the space.

closed 2021-01-30 by embeddedt