lvgl/lvgl #2050: How to hidden the container but the content in ?

HsiueMingLee, 2021-01-31:
How to hidden the container but the content in ?

1    lv_obj_t * box_condition_text;//condition text
2   box_condition_text = lv_cont_create(tabview01_weather, NULL);
3    lv_obj_set_size(box_condition_text,200,50);
4    lv_obj_set_pos(box_condition_text,60,245);
5    lv_obj_set_hidden(box_condition_text,true);

6    label_weather_condition_text = lv_label_create(box_condition_text, NULL);//weather condition text
7    lv_obj_align(label_weather_condition_text,box_condition_text,LV_ALIGN_CENTER,-30,-6);
8    lv_label_set_text(label_weather_condition_text, "CLEAR");    /*Set an initial text*/

This is a part of my code above. My purpose is creat a container first , and fill the text.When i debug the function,i want to hidden the container and leave the text.But when i did the 5st satement,container and text are both hidden.

HsiueMingLee, 2021-01-31:
lv_style_set_bg_opa(&style_box_backgroud,LV_STATE_DEFAULT,0);
    lv_style_set_border_opa(&style_box_backgroud,LV_STATE_DEFAULT,0); 

Solved !

embeddedt, 2021-01-31:
Correct solution!

closed 2021-01-31 by embeddedt