lvgl/lvgl #2052: global-buffer-overflow

incity, 2021-02-02:
just test with lv_demo_widgets, and switch to [Selectors] tab.

[image]

[image]

kisvegabor, 2021-02-03:
Hi,

Hm, I have never seen this issue. What version of LVGL do you use? What (related) compiler flags do you use?

incity, 2021-02-03:
this maybe because you access memory that are not malloc by you, like
char* buf = (char*) malloc(5);
int i = buf[5];// global buffer overflow?

you can complile and link with -fsanitize=address -fno-omit-frame-pointer

kisvegabor, 2021-02-03:
Upps, this meant to fix your other issue: https://github.com/lvgl/lvgl/issues/2053

I'm checking `fsanitize=address `

kisvegabor, 2021-02-03:
Should be fixed here: 8b78688c

commit 0631b170 (2021-02-03): fix(indev): clear the indev's `act_obj` in `lv_indev_reset`

fixes #2052
closed 2021-02-04 by incity