lvgl/lvgl #2057: Use freetype to report an error？？

chcmq, 2021-02-05:
I am sure I have installed freetype

=================================================
libs/lv_lib_freetype/lv_freetype.h:16:22: fatal error: ft2build.h: No such file or directory

******************************************************************
* Looking for ft2build.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ft2build.h"
* Web  > https://platformio.org/lib/search?query=header:ft2build.h
*
******************************************************************

=================================================

kisvegabor, 2021-02-08:
Do you link freetype with your program?
See https://github.com/lvgl/lv_lib_freetype#add-freetype-to-your-project

chcmq, 2021-02-08:
I used apt-get install to install it.
```bash
$apt-get install libfreetype6-dev
```
- System  Linux

kisvegabor, 2021-02-08:
You also need to add the library and its include path to platformio.ini.

chcmq, 2021-02-08:
- Can you help me to see where I configured wrong?
[image]
[image]

kisvegabor, 2021-02-08:
I'm not so experienced with PIO. Maybe @puzrin can help with that.

kisvegabor, 2021-02-08:
I just saw that you have opened a [topic](https://forum.lvgl.io/t/how-to-add-freetype-to-esp32/4515) in the forum.

Let's continu there. 

But in order to get help easier, please fill teh template.

closed 2021-02-08 by kisvegabor

puzrin, 2021-02-08:
There are no `freetype` package in PIO registry. Since this is not lvgl-specific, it's better to search help on PIO forum.

chcmq, 2021-02-08:
Thank you very much @puzrin @kisvegabor