lvgl/lvgl #2058: Add GPU API for accelerated drawing directly to framebuffer/texture

deltabeard, 2021-02-06:
## Introduction
LVGL currently uses software rendering to a draw to a buffer, which is then drawn on the screen using the `flush_cb` user callback function. Two GPU callback functions are currently implemented to allow for accelerated modification of a given buffer.

The API is limited somewhat however, as it does not allow for drawing primitives to be performed by user callback functions directly onto a framebuffer or texture. I think that a increase in performance could be obtained on platforms that use textures, such as SDL2, OpenGL, etc. when a high resolution, such as 1080p (like in my application), is used.

## Example

In my application, there is significant lag in LVGL during screen transition animations at 1080p, on x86_64 Haswell platform. Profiling shows that this is attributed to colour blending and drawing functions.

## Suggested solution

Callback functions would be implemented for the following drawing directly onto a texture:

- Drawing rectangles and lines:
```c
/**
 * Draws a filled rectangle at the given coordinates and ARGB colour, blending colours as required if alpha is set.
 * \param disp_drv LVGL Display context.
 * \param rect Coordinates of rectangle to draw; NULL for full display.
 * \param color Colour of rectangle to draw.
 */
void lv_disp_draw_rect_cb(lv_disp_drv_t *disp_drv, const lv_area_t *rect, lv_color_t color)
```
In SDL2, this callback would be implemented using the `SDL_RenderFillRect` function.

- Drawing pixel:
```c
/**
 * Draw a pixel at the given coordinate.
 * \param disp_drv LVGL Display context.
 * \param point Coordinate to draw pixel.
 * \param color Colour of pixel to draw.
 */
void lv_disp_draw_px_cb(lv_disp_drv_t *disp_drv, lv_coord_t point, lv_color_t color)
```
In SDL2, this callback would be implemented using the `SDL_RenderDrawPoint` function. I'm not sure how useful this callback will be however, as I don't know how LVGL renders elements currently.

- Drawing a buffer:
```c
/**
 * Draw a buffer onto the texture.
 * \param disp_drv LVGL Display context.
 * \param buf Buffer to draw.
 * \param src Coordinates of image to render from buf.
 * \param dest Coordinates to render the image to texture.
 */
void lv_disp_draw_buff_cb(lv_disp_drv_t *disp_drv, const lv_color_t *buf, const lv_area_t *src, const lv_area_t *dest)
```
This function can be used to draw software rendered buffers onto the texture. These buffers may be transparent, and alpha colours will be blended, thereby providing similar functionality to `gpu_blend_cb` except the result of the blend is not returned but rendered onto a texture.

With regards to the problem I'm facing, I think the `lv_disp_draw_buff_cb` function would demonstrate the greatest performance benefit. 

## Question

I am happy to look into implementing such functions. I am not familiar with the LVGL codebase though. I assume that I need to look at the lv_draw source files? Is adding this feature feasable, or would it require a major rework?

Many thanks.

kisvegabor, 2021-02-08:
Hi,

It's a good idea to use SDL functions directly as a GPU interface.

Recently we have taken the approach to integrate common GPU's usage right into LVGL without callbacks because it allows the greatest flexibility.

I suggest string with `draw_rect`. It can be simply added [here](https://github.com/lvgl/lvgl/blob/master/src/lv_draw/lv_draw_blend.c#L334-L356) similarly to other GPU implementations.

Would you like to give it a try?

deltabeard, 2021-02-08:
Hi,

Thank you for your response.

### 2D Acceleration

The LVGL API is currently limited in such a fashion that doesn't allow SDL2 to utilise 2D accelerate rendering functions. For example, it is not possible to use the following function as the GPU callback, as it does not render the requested rectangle to the given buffer, but instead directly to the SDL2 rendering context.

```c
void my_gpu_fill_cb(struct _disp_drv_t *disp_drv, lv_color_t *dest_buf, lv_coord_t dest_width,
	const lv_area_t *a, lv_color_t c)
{
	struct ui_ctx *ui_ctx = disp_drv->user_data;
	SDL_Renderer *ren = ui_ctx->ren;
	SDL_Rect r;

	SDL_SetRenderDrawColor(ren, c.ch.red, c.ch.green, c.ch.blue, c.ch.alpha);
	r.h = a->y2 - a->y1;
	r.w = a->x2 - a->x1;
	r.x = a->x1;
	r.y = a->y1;
	SDL_RenderFillRect(ren, &r);
}
```

<hr>

### 2D Acceleration then copy back software rendered buffer

In order to solve this, I perform the rendering in a 2D accelerated context, and then copy the pixel back into the output buffer:

- copy the destination buffer to a new texture *srctex*
- create a new texture *targtex* as a rendering target 
- render the given buffer texture *srctex* to the target texture *targtex*
- draw the rectangle on to the target texture with the 2D accelerated SDL2 drawing function
- copy the rendered pixels back into the destination buffer using *SDL_RenderReadPixels*

**This method does not work.** I think it's because SDL2 uses batch jobs for drawing, and probably does so on *SDL_RenderPresent* which I cannot call in this callback function. Furthermore, *SDL_RenderReadPixels* is slow anyway.

```c
	/* Sends buffer to 2D accelerated context.
	 * This does not work probably because the renderer must be flushed
	 * before returning from this function. */
	struct ui_ctx *ui_ctx = disp_drv->user_data;
	SDL_Renderer *ren = ui_ctx->ren;
	SDL_Texture *targtex, *srctex;
	SDL_Surface *srcbuf;
	SDL_Rect r;
	int w = dest_width;
	int h = a->y2;

	srcbuf = SDL_CreateRGBSurfaceWithFormatFrom(dest_buf, w, h,
		SDL_BYTESPERPIXEL(SDL_PIXELFORMAT_ARGB8888),
		w * 4, SDL_PIXELFORMAT_ARGB8888);
	srctex = SDL_CreateTextureFromSurface(ren, srcbuf);
	targtex = SDL_CreateTexture(ren, SDL_PIXELFORMAT_ARGB8888, SDL_TEXTUREACCESS_TARGET, w, h);

	SDL_SetRenderTarget(ren, targtex);
	SDL_SetRenderDrawColor(ren, c.ch.red, c.ch.green, c.ch.blue, c.ch.alpha);
	SDL_RenderCopy(ren, srctex, NULL, NULL);

	r.h = a->y2 - a->y1;
	r.w = a->x2 - a->x1;
	r.x = a->x1;
	r.y = a->y1;
	SDL_RenderFillRect(ren, &r);
	SDL_RenderReadPixels(ren, NULL, SDL_PIXELFORMAT_ARGB8888, dest_buf, dest_width);
	SDL_SetRenderTarget(ren, NULL);

	SDL_DestroyTexture(srctex);
	SDL_DestroyTexture(targtex);
	SDL_FreeSurface(srcbuf);
```

<hr>

### Software only rendering

So since we can't perform any 2D accelerated rendering with SDL2 in this callback function, we have to resort to software rendering.

```c
	SDL_Renderer *ren;
	SDL_Surface *srcbuf;
	int w = dest_width;
	int h = a->y2;
	SDL_Rect r;

	srcbuf = SDL_CreateRGBSurfaceWithFormatFrom(dest_buf, w, h,
		SDL_BYTESPERPIXEL(SDL_PIXELFORMAT_ARGB8888),
		w * 4, SDL_PIXELFORMAT_ARGB8888);
	ren = SDL_CreateSoftwareRenderer(srcbuf);

	r.h = a->y2 - a->y1;
	r.w = a->x2 - a->x1;
	r.x = a->x1;
	r.y = a->y1;
	SDL_SetRenderDrawColor(ren, c.ch.red, c.ch.green, c.ch.blue, c.ch.alpha);
	SDL_RenderFillRect(ren, &r);

	SDL_DestroyRenderer(ren);
	SDL_FreeSurface(srcbuf);
```

Within SDL2, this method eventually calls *full_function* which may provide a performance benefit in the possible use of SSE and NEON instructions should the running platform support it. Otherwise, the speed of this method will be the same as the generic method used by LVGL should a callback for *gpu_fill_cb* not be provided.

<hr>

The *2D Acceleration* method described would provide the biggest performance benefit with SDL2 applications, but it's not clear how difficult it would be to implement. I understand that the current implementation in LVGL allows for incredible portability, so I wonder if a new callback function --- that is disabled by default within an ifdef --- could be added to allow for drawing rectangles directly to a renderer?

kisvegabor, 2021-02-09:
Thanks for the investigation! I hoped that SDL can manage somehow to draw by GPU and give the result back.

Once I was thinking about it when considered OpenGL integration. What about replacing LVGL's drawer (rectangle, text, line, image etc drawer) directly. I think it'd be pretty straightforward to add a rectangle and image drawer.
Text drawer would be trickier but I can add a special mode where LVGL creates an ARGB image from the text first that can be passed to SDL as an image. 

Take look at the rectangle drawer [here](https://github.com/lvgl/lvgl/blob/e6de537952c3c2d9f37096938dd5b876c6ba6802/src/lv_draw/lv_draw_rect.c#L103-L126). For first a plain one-colored rectangle with no radius and border would be great and we can add more features laters.

For the sake of simplicity it could work only with 2 full framebuffers without partial updates (i.e. always the whole frame is redrawn).

What do you think?

MouriNaruto, 2021-02-10:
It's also a clever idea for the pure Windows GDI implementations.

It may solve the performance issue when using LVGL under 4k or higher display.

MouriNaruto, 2021-02-10:
I also hope LVGL can provide the ability to write to framebuffer which the user provided directly.

For example, in the new implementations in [lvgl/lv_sim_visual_studio](https://github.com/lvgl/lv_sim_visual_studio), it has already created the Windows GDI bitmap framebuffer for LVGL. But in today's LVGL, I also need to copy pixels from LVGL's pixel buffer to Windows GDI bitmap framebuffer in the callback which I set in the `flush_cb`. And by the design of LVGL, it's hard to optimize it. Here are some reasons:

- LVGL support multiple types of color depths and the framebuffer only support one type. 
- By the design of LVGL, LVGL provide us a framebuffer of the area we need to refresh it. It makes I can't use the memcpy directly even though I set size of the LVGL pixel buffer as same as the framebuffer. I can use SIMD intrinsics to optimize it but it's trivial task and reduce the portability. I also can use the BitBlt API in Windows GDI but it increases the CPU context switches and add the call overhead.

So I hope LVGL provide a API to set the custom framebuffer. I guess it may need these:

- Color depth
- Resolution
- The address of the framebuffer
- The notify callback for caller to flush the display
- The way to change the custom framebuffer dynamically

kisvegabor, 2021-02-11:
Hi @MouriNaruto 
Take look at [Two screen-sized buffer](https://docs.lvgl.io/latest/en/html/porting/display.html#display-buffer) mode here. It makes LVGL work on frame buffers directly

MouriNaruto, 2021-02-12:
> Hi @MouriNaruto
> Take look at [Two screen-sized buffer](https://docs.lvgl.io/latest/en/html/porting/display.html#display-buffer) mode here. It makes LVGL work on frame buffers directly

@kisvegabor Hi, I have tried that. But I don't find significant performance improvement. I used the profiler and found that the LVGL will copy the content from buf1 (the pixel buffer) to buf2 (the frame buffer) internally.

https://github.com/lvgl/lvgl/blob/c4b46706c3e977e918224009623e6d1caf118151/src/lv_core/lv_refr.c#L249

It looks better if the LVGL can provide a way to write to the frame buffer *directly*.

kisvegabor, 2021-02-12:
> I used the profiler and found that the LVGL will copy the content from buf1 (the pixel buffer) to buf2 (the frame buffer) internally.

Yes. We have two options: 
1. Always redraw the whole screen even if only a few pixels have changed
2. Redraw only the changed parts, but then the 2 framebuffers need to be kept in sync.

Now 1) is used but in v8 I"m considering changing to 2) because it's more standard and makes SDL/OpenGL usage much simpler.

MouriNaruto, 2021-02-12:
@kisvegabor 
The second option looks better. And I hope the LVGL can provide a method to achieve that with one framebuffer. (Because the double buffering has been already supported in many graphics APIs.)

kisvegabor, 2021-02-15:
How do you mean "with one frame buffer"? 2 frame buffers are required anyway: 1 for drawing, 1 for displaying the previous image.

MouriNaruto, 2021-02-15:
> How do you mean "with one frame buffer"? 2 frame buffers are required anyway: 1 for drawing, 1 for displaying the previous image.

@kisvegabor 
In some cases, we can remove the frame buffer for displaying the previous image. Provide a way to let the users to implement the double buffering by themselves may improve the experience because some graphics APIs have already had the double buffering mechanism.

kisvegabor, 2021-02-17:
Ah, I see. Actually, it's possible now if you set a `rounder_cb` in your display driver and use a single display buffer.

MouriNaruto, 2021-02-17:
> Ah, I see. Actually, it's possible now if you set a `rounder_cb` in your display driver and use a single display buffer.

@kisvegabor 

Thank you very much. I have tried it. It's nice with no memory copy operations. Makes the CPU usage of the process is 0%. 🎉

But I also want to you notice the https://github.com/lvgl/lv_sim_visual_studio/issues/20 and hope to get the result. Thanks.

embeddedt, 2021-02-18:
@kisvegabor Does this force a redraw of the entire screen whenever any object gets invalidated, or is only the invalidated portion of the VDB changed?

MouriNaruto, 2021-02-19:
@embeddedt It looks like that LVGL is only the invalidated portion of the VDB changed when I am viewing the function calls from performance profiler.

kisvegabor, 2021-02-22:
> Does this force a redraw of the entire screen whenever any object gets invalidated, or is only the invalidated portion of the VDB changed?

Yes, the `rounder_cb` works on the area invalidation. If you a `rounder_cb` that sets every coordinate screen-sized, always the whole screen will be invalidated and redrawn.

MouriNaruto, 2021-02-22:
> Yes, the `rounder_cb` works on the area invalidation. If you a `rounder_cb` that sets every coordinate screen-sized, always the whole screen will be invalidated and redrawn.

Yes, but it can reduce the memory copy operations when update the framebuffer. (I know it will cause the whole screen will always be invalidated and redrawn. For example, the CPU and GPU usage of the desktop compositor of Windows (Desktop Window Manager) will increased. (Or it means most of the CPU time is used in the rendering instead of memory copying.) So, the performance of Windows implementation has been improved a lot. (I have tried the lv_demo_benchmark demo, most of items can reach up to 40,000 fps.))

I also wonder to know how to lock the frame rate dynamically. (For example, I can get the refresh rate of the monitor by functions provided via OS. If I can set the frame rate of LVGL to the refresh rate of the monitor. It can make users get a comfortable experience.)

kisvegabor, 2021-02-22:
>  So, the performance of Windows implementation has been improved a lot. (I have tried the lv_demo_benchmark demo, most of items can reach up to 40,000 fps.))

Wow, incredible!

> I also wonder to know how to lock the frame rate dynamically. (For example, I can get the refresh rate of the monitor by functions provided via OS. If I can set the frame rate of LVGL to the refresh rate of the monitor. It can make users get a comfortable experience.)

You need more than the frame rate. You need to get a trigger to VSYNC and pass the draw buffer to Windows (?) right after the VSYNC signal. On large resolutions without VSYNC ugly tearing can be visible on animations or scrolling.

MouriNaruto, 2021-03-09:
@kisvegabor 

I wonder to know which APIs in LVGL are the fundamental APIs for drawing. Because I wonder to try to implement it with Windows GDI or Direct2D for discover the GPU acceleration.

kisvegabor, 2021-03-12:
It's really unbelievable that software rendering could produce such a high FPS. 

> I wonder to know which APIs in LVGL are the fundamental APIs for drawing. Because I wonder to try to implement it with Windows GDI or Direct2D for discover the GPU acceleration.

Take a look at [this comment](https://github.com/lvgl/lvgl/issues/2058#issuecomment-775869196). I believe `lv_draw_rect/img/label/line` should be directly replaced by the "GPU drawer". I think some adjustments will be required in LVGL too (especially in label drawing) to make support GPU simpler. 

Anyway, supporting the first two cases of the benchmark (Rectangle and Rectangle + opa) seems the simplest and with them, we could have an initial comparison.

embeddedt, 2021-04-14:
Not stale, just a long-term improvement.

Miguel0101, 2021-07-12:
Will this be compatible with microcontrollers, e.g. ESP32, STM32, ...?

mariotaku, 2021-08-19:
Hi. I have tried LVGL on Raspberry Pi and really wish to see hardware acceleration support. Seeing this in v8.1 roadmap, I have a few questions:

1. Has this task started?
2. Will it use accelerated SDL_Render or OpenGL (ES)?
3. AFAIK for OpenGL with SDL, vsync is a good option. How do you handle that? Because main loop will likely being limited to be the same as frame rate.

Also I would be happy to help if contribution is needed :)

kisvegabor, 2021-08-19:
> Has this task started?

No yet.

> Will it use accelerated SDL_Render or OpenGL (ES)?

It should be modular to allow adding multiple GPU APIs.

> AFAIK for OpenGL with SDL, vsync is a good option. How do you handle that? Because main loop will likely being limited to be the same as frame rate.

I don't know (yet) how OpenGL and SDL handles VSYNC. We can decide these things is we go deeper.

I can take care of the necessary architectural change in LVGL but I don't  know the higher level GPUs well enough to deal with that part too.
So to get started with this feature we need someone who can implementation of OpenGL (ES) or SDL API that replaces LVGL's drawing routines. 

> Also I would be happy to help if contribution is needed :)

Amazing! :tada: 

As a proof of concept, I think the first step could be to make LVGL draw simple rectangles with GPU.

deltabeard, 2021-08-19:
I know how to use SDL2 and its rendering API, but I don't know how lvgl works internally. I think this new API would be similar to existing immediate-mode interfaces, such as [Nuklear](https://github.com/Immediate-Mode-UI/Nuklear) and [IMGUI](https://github.com/ocornut/imgui). So instead of lvgl using its own internal buffer, it would call functions that are implemented by the developer to draw directly to the screen. For brevity, I'll refer to this new hardware accelerated drawing API as the "new" API, in this post.

Functions that the developer would have to provide to lvgl in this new API would include, for example, drawing a line between two points, drawing a filled rectangle, etc. Functions that are typically required for immediate-mode toolkits.

For more complex UI elements, such as text, I think that the new API would have functions that would allow for the creation, drawing and deletion of textures. So lvgl would call a function to create a texture from a buffer (allocated and provided by lvgl), which on success, would return an opaque pointer or a hash (for a hash map) that lvgl can use to refer to the texture later. Whenever text would need to be drawn, lvgl would simply call a new function that would draw this texture at certain coordinates on screen. A delete texture function would be used by lvgl to free memory, on exit for example.

## Summary

### Creating textures

LVGL would first determine how large of a texture is required the text that is going to be drawn. Then, it would call gpu_create_texture to obtain a pixel buffer of the created texture. The format of the texture would be defined at build time with `LV_COLOR_DEPTH` in `lv_conf.h`.
```c
void *gpu_create_texture(void *priv, void *pixels, int width, int height);
```
This function could be implemented in SDL2 as (assuming `LV_COLOR_DEPTH` is 32, and with no error checking):
```c
struct context {
	SDL_Renderer *renderer;
};
void *gpu_create_texture(void *priv, void *pixels, int width, int height)
{
	SDL_Surface *surf;
	SDL_Texture *tex;
	struct context *ctx = priv;

	surf = SDL_CreateRGBSurfaceWithFormatFrom(pixels, width, height, 32, width*4, SDL_PIXELFORMAT_ARGB8888);
	tex = SDL_CreateTextureFromSurface(priv->renderer, surf);
	SDL_FreeSurface(surf);

	return tex;
}
```

### Drawing textures

Later on, lvgl would want to draw this texture on a frame.
```c
void gpu_draw_texture(void *priv, void *texture, int x, int y, int width, int height);
```
This function could be implemented in SDL2 as (assuming `LV_COLOR_DEPTH` is 32, and with no error checking):
```c
struct context {
	SDL_Renderer *renderer;
};
void gpu_draw_texture(void *priv, void *texture, int x, int y, int width, int height)
{
	SDL_Texture *tex = texture;
	struct context *ctx = priv;
	SDL_Rect dstrect = { .x =x, .y = y, .w = width, .h = height };
	SDL_RenderCopy(ctx->renderer, tex,
                   NULL, &dstrect);
}
```

### Freeing textures

Lastly, lvgl would delete this texture if, for example, the screen has been deleted, or the application is exiting, etc.
```c
void gpu_destroy_texture(void *priv, void *texture);
```
This function could be implemented in SDL2 as (assuming `LV_COLOR_DEPTH` is 32, and with no error checking):
```c
struct context {
	SDL_Renderer *renderer;
};
void gpu_destroy_texture(void *priv, void *texture)
{
	SDL_Texture *tex = texture;
	struct context *ctx = priv;
	SDL_DestroyTexture(texture);
}
```

## Final comments

The texture API isn't just for text, but anything else that may take a while to render but change rarely. Becuase I don't know how lvgl works internally, it's a bit of a challenge to get started with this new API. What are your thoughts? Do you think you could give a brief guide as to which source file in lvgl we should be looking at?

There may need to be some consideration as to freeing textures that haven't been for a while, in order to stop VRAM usage increasing as the user browses large LVGL applications. A possible method would be to keep only the top 1000 most recently drawn textures, or to simply delete textures that haven't been drawn in a while until the GPU is able to create a new texture.

mariotaku, 2021-08-19:
@kisvegabor @deltabeard Thanks for your replies! My question 2 and 3 is a bit vague, so I'll add more details here.
2. I would suggest to use platform independent API (OpenGL), and treat it as something similar to other GPUs (like `lv_gpu_stm32_dma2d`). But SDL2 would definitely be the easiest way to implement.
3. According to my understanding, LVGL has its own main loop. And this loop is actually running nearly thousand times per second, while interval of each frames are calculated. Typical SDL loop with OpenGL runs like this:
```
while (true) {
  while (SDL_PollEvent()) {
    // Process events
  }
  my_drawing_logic();
  SDL_GL_SwapWindow();
}
```
And I think `SDL_GL_SwapWindow` will lock main thread until next frame is needed. So the way to implement main loop might be very different.

mariotaku, 2021-08-19:
Anyway I'll try to see if I can make some changes to draw a blank page with hardware acceleration.

kisvegabor, 2021-08-19:
I had something different in mind but it's great that you have approached it from a different aspect. 

My idea was to simply replace LVGL draw functions with GPU drawers.
E.g. in [lv_draw_rect](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_rect.c#L100) we see from  [lv_draw_rect_dsc_t](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_rect.h#L30-L70) that a red rectangle needs to drawn with 2 px blue border and 10 px radius. Then we check that whether the GPU can draw it or not. If it can make it draw right into the frame buffer. If it can't fall back to LVGL's SW rendering.

As far as I know you can't draw e.g. rounded rectangles with SDL. But you can fill the background to a color very fast, or blit a screen sized  wallpaper. I believe the majority of the rendering times comes from slow fill rate or blit rate of SW rendering.

If I understand your idea correctly we should keep LVGL's SW renderer but make it render into a texture, cache these textures and make the GPU blit the textures on each other. Is it correct?

If so it fits very well to a the new LVGL feature called [object snapshot](https://docs.lvgl.io/master/others/snapshot.html?highlight=snapshot). It creates an image from an object including all its children. For the GPU probably we need the widgets (or even simpler primitives) without the children. (so a slider is built from 3 textures: background, slider, knob.)

However once we have this texture caching it'd be nice to use it on widget level too. E.g. if a complex button or chart is cached just use its image.

What do you think?

kisvegabor, 2021-08-19:
@mariotaku 
Thank's for the clarification. There is usually some delay after `lv_timer_handler()` so it's rather called ~100 times per second. With 60 FPS the  waiting for the next frame takes max 16 ms which seems acceptable.

mariotaku, 2021-08-20:
@kisvegabor Object snapshot sounds like a good option to me. Although my idea was to replace all implementations in `lv_draw_` with hardware accelerated API calls, your proposal is actually more realistic. We update texture (snapshot) of an object if needed and blend them with hardware accelerated API. I think that's also many modern UI frameworks does, like Android.

I'll first try if LVGL can work well with system vsync.

mariotaku, 2021-08-20:
PoC here: https://github.com/mariotaku/lvgl-sdl-example

However the performance is worse than software rendering... Because uploading textures to GPU takes time, no matter what the size is. I'll see if I can upload rasterized font as texture to GPU, and use them to draw.

[image]

deltabeard, 2021-08-20:
> PoC here: https://github.com/mariotaku/lvgl-sdl-example
> 
> However the performance is worse than software rendering... Because uploading textures to GPU takes time, no matter what the size is. I'll see if I can upload rasterized font as texture to GPU, and use them to draw.

Good work! I've compiled and ran the program and also draw the same conclusion. In this case, I think that a texture cache of some sort would be useful here, as that would mitigate having to redraw a texture over and over again. I think that it is the texture cache that would have a big benefit over software rendering. So `draw_letter_normal` would have some logic to check if a texture already exists for that letter, and if so uses that or creates one and stores it in an array for later.

Profiling showed that unlocking textures was the bottleneck.
[image]

Here is a diff of changes I made to your project to get it to compile at commit 4c5f449dcb2b7664783087a07b25b065e2697f3d.
[deltabeard.diff.txt](https://github.com/lvgl/lvgl/files/7022647/deltabeard.diff.txt)

mariotaku, 2021-08-20:
@deltabeard Update: I have modified font rendering logic so it went really performant.

Rendering with SW: 100fps

[image]

Rendering with HW accel: 600fps

[image]

deltabeard, 2021-08-20:
> @deltabeard Update: I have modified font rendering logic so it went really performant.

Great work on the font baking additons on your branch too. 👍  Is this improvement in rendering with font baking or without?

mariotaku, 2021-08-21:
> 
> > @deltabeard Update: I have modified font rendering logic so it went really performant.
> 
> Great work on the font baking additons on your branch too. 👍  Is this improvement in rendering with font baking or without?

Yes. It turns out updating textures are biggest performance penalty. So I'm thinking if we can cache all bitmaps into GPU directly so they will only be updated when changed.

mariotaku, 2021-08-22:
The approach of caching masks, images and fonts as GPU texture is highly effective with minimal CPU overhead. You can try my example to see how well it performs on your computer.

https://user-images.githubusercontent.com/830358/130347799-d7296bac-2902-4902-b027-6e5e74fe7c13.mp4

kisvegabor, 2021-08-24:
Amazing results!

So in summary from LVGL's side the first task is to update `lv_draw_rect/letter/img` etc function to draw into a custom buffer (i.e. texture), right? And as the next step check if a texture with the same parameters is already rendered. If so return the texture else allocate a texture and render into it.

One thing is still not clear to me. Let's say there is a very long text on a 100x100 rectangle that has some radius and `clip_corner` enabled (the content overflowing on the corners are clipped by LVGL). 
- Shall we render the whole text and blit it with an offset?
- How to handle `clip_corner`?

Or in a more simple case: If there are 2 plain rectangles, `bg` and `fg`, where `fg` is the children of `bg` and positioned out of `bg` Therefore some parts of `fg` are clipped by the bounding box of `bg`. How can we handle this?

kisvegabor, 2021-08-24:
I'm just reviewing your PR and it seems all my concerns are implemented :slightly_smiling_face:

MouriNaruto, 2021-08-25:
It's really impressive. I will try to port @mariotaku‘s sdl render to Windows GDI for `lv_port_windows` repo and `win32drv` in `lv_drivers` repo.

In a week ago, I have try to implement the similar implementation and met some issues. I think it can give me some help.

mariotaku, 2021-08-25:
@kisvegabor Seems that `clip_corner` for texts are not correctly handled yet. I'll add this later.

kisvegabor, 2021-08-25:
@MouriNaruto I suggest waiting a little while the SDL version stabilizes.

lishaoenxm, 2021-09-01:
@kisvegabor We are working on an IOT wearable project with 2 hardware graphics accelerators. Each hardware graphics accelerator has the abilities as line, bitblt, blending etc. The 2 hardware graphics accelerators performance and power consumption are different. The project requires that user can select which hardware graphics accelerator at runtime. 

We have gone over the LVGL source code. Now the _lv_disp_drv_t_ has the interfaces _gpu_fill_cb_ and _gpu_wait_cb_. It's not enough. More interfaces as drawing line, bitblt, blending should be added. At the same time, LVGL drawing functions should support the new hardware accelerator interfaces.

What's your suggestions?

kisvegabor, 2021-09-06:
@lishaoenxm 
In #2521 @mariotaku  is working on using external drawer routines.  Take a look at files in `src/gpu/sdl` folder. 

I believe in the same way we can add support to other higher level GPUs too. 

Could it work in your case too?

lishaoenxm, 2021-09-10:
> @lishaoenxm
> In #2521 @mariotaku is working on using external drawer routines. Take a look at files in `src/gpu/sdl` folder.
> 
> I believe in the same way we can add support to other higher level GPUs too.
> 
> Could it work in your case too?

We'll check it . Thank you!

kisvegabor, 2021-10-19:
Already added into `master` so I close this issue.

closed 2021-10-19 by kisvegabor