lvgl/lvgl #2060: lv_list not scrolling with LV_KEY_PREV and LV_KEY_NEXT

jkent, 2021-02-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

lv_list is not scrolling to the selected item when using LV_INDEV_TYPE_KEYPAD and LV_KEY_PREV and LV_KEY_NEXT.  Selection is off screen

### To Reproduce

### Expected behavior
Tabbing through the list should result in the list being scrolled, I.E. "Item I" being shown once it is selected.

A little off topic, but I think the way the arrow keys work with LV_INDEV_TYPE_KEYPAD is a little broken.  On screen keyboard cannot be used with such a setup.

### Screenshots or video
[image]

Even better, this repository has a simulator setup that shows the problem, using the Up/Down arrow keys:  
`https://gitlab.jkent.net/odroid-go/golauncher`

Just `cd` to simulator and do "make run"

embeddedt, 2021-02-08:
Fix proposed in #2061. Thanks for the report!

By the way, I had to apply the following patch to your Makefile to get it to build, as GCC normally requires libraries to be specified after object files, not before:

```diff
diff --git a/simulator/Makefile b/simulator/Makefile
index 47e998d..52129da 100644
--- a/simulator/Makefile
+++ b/simulator/Makefile
@@ -183,7 +183,7 @@ $(BUILD_DIR)/%.o: %.cpp
 $(TARGET): $(OBJECTS)
        @echo "  LD      $@"
        @mkdir -p $(BUILD_DIR)
-       @$(CC) $(CFLAGS) $(LDFLAGS) -o $@ $^
+       @$(CC) $(CFLAGS) -o $@ $^ $(LDFLAGS)
 
 run: $(TARGET)
        @$(TARGET)
```

jkent, 2021-02-08:
Thanks! and thanks for the patch!

PR #2061 (2021-02-17): fix(list) scroll list when button is focused using LV_KEY_NEXT/PREV
### Description of the feature or fix

Fixes #2060

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Update CHANGELOG.md
- [ ] Update the documentation
closed 2021-02-17 by embeddedt