lvgl/lvgl #2064: Weirdness with antialiasing and border

embeddedt, 2021-02-11:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

This isn't a major bug, but it's been bothering me for a while now. The antialiasing on objects with a border seems to choose the wrong color to blend with sometimes, resulting in an ugly line between the object edge and the start of the border.

This might be how it's intended to behave, but I figured it was worth clarifying.

### To Reproduce

Visit https://lvgl.github.io/lv_examples/lv_demo_widgets/index.html (or run `lv_demo_widgets`), enable Dark mode, and toggle the switch inside "Basics".

### Expected behavior

The blend should be smoother (more like the horizontal portions of the switch).

### Screenshots or video

[image]

kisvegabor, 2021-02-11:
Rendering goes like this:
1. Draw the background with an anti-antialiased edge. So the pixel on the very edge gets greenishgray.
2. Draw the outline. Its inner curve is exactly on the background's antialiased pixels. So it will blend its greenish color to the green-gray pixel resulting in green+ greenishgray.

We could solve it by making the outline's padding -1 px to be sure it overlaps the background.

kisvegabor, 2021-02-11:
Quick test:
[image]

Change this in material theme:
```c
    lv_style_set_outline_width(&styles->bar_bg, LV_STATE_DEFAULT, OUTLINE_WIDTH + 1);
    lv_style_set_outline_pad(&styles->bar_bg, LV_STATE_DEFAULT, -1);
```

PR #2067 (2021-02-12): fix(draw) overlap outline with background to prevent aliasing artifacts
### Description of the feature or fix

Fixes #2064.

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Update CHANGELOG.md
- [ ] Update the documentation
closed 2021-02-12 by kisvegabor