lvgl/lvgl #2066: Some objects don't invalidate on state change

embeddedt, 2021-02-11:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Objects don't seem to be invalidated correctly when changing states. Interestingly, this bug does NOT occur on my STM32, only on the PC simulator.

### To Reproduce

Run `lv_demo_widgets` in the PC simulator, using lvgl `master`.

### Expected behavior

The button should transition back to it's normal state when I release it.

### Screenshots or video

[image]

### Additional info

Setting `LV_STYLE_CACHE_LEVEL` to 0 in `lv_conf.h` did not fix the issue, unfortunately.

kisvegabor, 2021-02-11:
I can't see the issue on my PC.

What is the display buffer size, and resolution in your config?

embeddedt, 2021-02-11:
480x320

```c
  /*Create a display buffer*/
  static lv_disp_buf_t disp_buf1;
  static lv_color_t buf1_1[LV_HOR_RES_MAX * 120];
  lv_disp_buf_init(&disp_buf1, buf1_1, NULL, LV_HOR_RES_MAX * 120);

  /*Create a display*/
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv); /*Basic initialization*/
  disp_drv.buffer = &disp_buf1;
  disp_drv.flush_cb = monitor_flush;
  lv_disp_drv_register(&disp_drv);
```

kisvegabor, 2021-02-11:
Now I see the issue.

I've checked that it introduced in this commit: a62ecda7
Investigating...

xiaoxiang781216, 2021-02-14:
new PR(https://github.com/lvgl/lvgl/pull/2073) should fix the issue, please review it.

kisvegabor, 2021-02-15:
Fixed in https://github.com/lvgl/lvgl/pull/2073

closed 2021-02-15 by kisvegabor