lvgl/lvgl #2070: Latest changes to lv_anim gives me compilations errors for lv_micropython

uraich, 2021-02-12:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Error when compiling:
./../lib/lv_bindings/lvgl/src/lv_widgets/lv_label.c: In function ‘lv_label_set_long_mode’:
../../lib/lv_bindings/lvgl/src/lv_widgets/lv_label.c:317:24: error: cast between incompatible function types from ‘void (*)(lv_obj_t *, lv_coord_t)’ {aka ‘void (*)(struct _lv_obj_t *, short int)’} to ‘void (*)(void *, lv_anim_value_t)’ {aka ‘void (*)(void *, int)’} [-Werror=cast-function-type]
  317 |     lv_anim_del(label, (lv_anim_exec_xcb_t)lv_obj_set_x);
and many more of the same type

### To Reproduce
compile lv_micropython
-->

### Expected behavior
should be able to compile

### Screenshots or video

kisvegabor, 2021-02-12:
Thanks, for the report!

@xiaoxiang781216 I've temporally reverted the a62ecda7 commit because it caused the about mentioned compiler warning (treated as error) and this issue #2066

xiaoxiang781216, 2021-02-12:
@kisvegabor this may due to the patch:
https://github.com/lvgl/lvgl/pull/2063/commits/ab2abbed74b63f2b80c4ca25a2d24e59d1a5f6f4
which change the anmination type from int16_t to int32_t and then may introduce the compatibility issue.

xiaoxiang781216, 2021-02-14:
new PR(https://github.com/lvgl/lvgl/pull/2073) should fix the issue, please review it.

kisvegabor, 2021-02-15:
Fixed in https://github.com/lvgl/lvgl/pull/2073

closed 2021-02-15 by kisvegabor