lvgl/lvgl #2077: How to support LVGL symbol icons in all fonts

MouriNaruto, 2021-02-14:
### Introduce the problem

I had written the Windows GDI font engine for getting the font glyph at the runtime like the way of lv_lib_freetype.

But I found we can't use the LVGL symbol icons.

### Examples and cases

[image]

### Suggested solution

N/A

embeddedt, 2021-02-15:
That's a smart approach!

I'm not sure, but I think you might need to add logic to translate from FontAwesome symbols to Windows equivalents, as the glyph codes we use are specific to [FontAwesome](https://fontawesome.com/).

kisvegabor, 2021-02-15:
> I'm not sure, but I think you might need to add logic to translate from FontAwesome symbols to Windows equivalents, as the glyph codes we use are specific to FontAwesome.

Furthermore, the built-in symbols are used from this [font file](https://lvgl.io/assets/others/FontAwesome5-Solid+Brands+Regular.woff). It's a single file merged from 3 FontAwesome font files.

The symbols defined in LVGL are the UTF8 coded Unicode codes from the linked font file.

MouriNaruto, 2021-02-19:
> Furthermore, the built-in symbols are used from this [font file](https://lvgl.io/assets/others/FontAwesome5-Solid+Brands+Regular.woff). It's a single file merged from 3 FontAwesome font files.
>
> The [...]

@kisvegabor 

I have download the font file and edit it to make it only have the symbols the LVGL needed.

I also want to ask permission for a new repo for Windows application port because most people lv_sim_visual_studio should keep the same behavior as the old simulator. But I want to add some features to make people can use LVGL to develop Windows desktop applications. It makes users confused. (I had met a user ask me about that.) So I ask for the permission, I will keep the lv_sim_visual_studio like the simulator. (Or rename the repo like lv_port_windows, and I maintain two flavors(simulator and desktop application).)

kisvegabor, 2021-02-22:
@MouriNaruto Permission granted :slightly_smiling_face: 

It'd be great to ensure that LVGL can work smoothly with the Windows ecosystem.
I'm creating the repo.

kisvegabor, 2021-02-22:
Created https://github.com/lvgl/lv_port_windows/
And I also sent you an invitation for "Maintain" access. (almost like Admin)

MouriNaruto, 2021-02-22:
@kisvegabor Thank you very much.

MouriNaruto, 2021-03-04:
[image]

@kisvegabor @embeddedt  I have solved the issue. I use the font that @kisvegabor provided, reduce the font glyphs for less binary size, embedded into the source code.

This is the font file I finally used. 
[Font Awesome 5 Free for LVGL.zip](https://github.com/lvgl/lvgl/files/6082668/Font.Awesome.5.Free.for.LVGL.zip)

kisvegabor, 2021-03-04:
>  I have solved the issue. I use the font that @kisvegabor provided, reduce the font glyphs for less binary size, embedded into the source code.

Great!

The smaller binary size was important for you for performance reasons? 

For general purpose usage, I prefer the font with all symbols to allow users to add symbols not included in the build-in list.

MouriNaruto, 2021-03-04:
@kisvegabor 

> The smaller binary size was important for you for performance reasons?

I am interested in writing the most compact implementations by using the least syntaxes and third-party libraries. And I worked for the project team which need for compact implementations.

MouriNaruto, 2021-03-04:
@kisvegabor 

> For general purpose usage, I prefer the font with all symbols to allow users to add symbols not included in the build-in list.

I have provided two versions of Font Awesome 5 Free font. (One is for optimized size, other is for full symbols.)

https://github.com/lvgl/lv_port_windows/blob/master/LVGL.Windows/LVGL.Resource.FontAwesome5Free.h
https://github.com/lvgl/lv_port_windows/blob/master/LVGL.Windows/LVGL.Resource.FontAwesome5FreeLVGL.h

I hope you will like it.

closed 2021-04-13 by MouriNaruto