lvgl/lvgl #2082: slider draws incorrectly if vertical and type=symmetrical

hammockman, 2021-02-16:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Vertical slider with LV_SLIDER_TYPE_SYMMETRICAL behaves incorrectly

### To Reproduce
The micropython simulator exhibits the same mis-behaviour as the PC simulator (CodeBlocks) and the Arduino port:

https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html
```python
import imp, sys
sys.path.append('https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/master/lib')
import display_driver
import lvgl as lv
scr = lv.obj()
s = lv.slider(scr)
s.set_width(20)
s.set_height(100)
s.set_range(-100,100)
s.set_value(0,0)
s.set_type(1)
s.align(scr, lv.ALIGN.CENTER, 0, 0)
lv.scr_load(scr)
```

### Expected behavior

If either the slider is horizontal or the slider type is not symmetrical then the slider behaves as expected.
[image]

### Screenshots or video

[image]

[image]

kisvegabor, 2021-02-17:
@embeddedt You are the most familiar with the news slider (bar) drawer. Could you take a look?

embeddedt, 2021-02-17:
Will investigate when I have a chance.

embeddedt, 2021-02-20:
This should be fixed in #2085.

kisvegabor, 2021-02-22:
Worked well for me!

hammockman, 2021-02-24:
Brilliant, that seems to have resolved the issue. Thanks very much for the prompt attention.

PR #2085 (2021-02-24): fix(bar) correct symmetric handling for vertical bars
### Description of the feature or fix

Fixes #2082.

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [x] Update CHANGELOG.md
- [x] Update the documentation
closed 2021-02-24 by embeddedt