lvgl/lvgl #2089: global-buffer-overflow

incity, 2021-02-23:
[image]

kisvegabor, 2021-02-23:
How can we reproduce it? What is the content of teh text area?

incity, 2021-02-24:
for multi-line textarea, input multi-line text and delete chars will produce it.

kisvegabor, 2021-02-24:
It works well here:

[image]

```

WARNINGS ?= -Werror -Wall -Wextra \
  -Wshadow -Wundef -Wmaybe-uninitialized -Wmissing-prototypes -Wno-discarded-qualifiers \
  -Wno-unused-function -Wno-error=strict-prototypes -Wpointer-arith -fno-strict-aliasing -Wno-error=cpp -Wuninitialized \
  -Wno-unused-parameter -Wno-missing-field-initializers -Wno-format-nonliteral -Wno-cast-qual -Wunreachable-code -Wno-switch-default  \
  -Wreturn-type -Wmultichar -Wformat-security -Wno-ignored-qualifiers -Wno-error=pedantic -Wno-sign-compare -Wno-error=missing-prototypes -Wdouble-promotion -Wclobbered -Wdeprecated  \
  -Wempty-body -Wshift-negative-value -Wstack-usage=20048 \
  -Wtype-limits -Wsizeof-pointer-memaccess -Wpointer-arith \
  -fsanitize=address -fno-omit-frame-pointer
            
CFLAGS ?= -O0 -g3 -I$(LVGL_DIR)/ $(WARNINGS)
LDFLAGS ?= -lSDL2 -lm  -fsanitize=address  -fno-omit-frame-pointer
```

incity, 2021-02-24:
my lv_examples is not the latest, but i think it was not the problem??

embeddedt, 2021-02-24:
@incity Can you reproduce it using one of the examples or a slightly modified version and post the exact code here? Otherwise it will be difficult to debug.

incity, 2021-02-24:
@embeddedt I just use the lv_examples, and do not change any code.

incity, 2021-02-24:
my lv_examples version is the commit a5015dbb38dd56d49bbf7a96bd2e4bd56c10bf29

kisvegabor, 2021-02-24:
Could you try with the latest lvgl and lv_examples?

incity, 2021-02-24:
when i input text, there are some warnings(i has use latest lvgl and lv_exampes) 
[image]

incity, 2021-02-24:
the attachment is my config

kisvegabor, 2021-02-27:
It happens because you have enabled `LV_USE_ARABIC_PERSIAN_CHARS` but font doesn't have Persian characters. 
In the last line on the left there is an empty button. There should be persian character to switch to the persian page.

Do you still see the issue with disabled `LV_USE_ARABIC_PERSIAN_CHARS`.

I can't see it even with your lv_conf.h.

incity, 2021-02-28:
i disable LV_USE_ARABIC_PERSIAN_CHARS, but still has this issue

incity, 2021-02-28:
[image]

kisvegabor, 2021-02-28:
Just guessing: what if you change `lv_txt_utf8_get_byte_id` like this:
```c
static uint32_t lv_txt_utf8_get_byte_id(const char * txt, uint32_t utf8_id)
{
    uint32_t i;
    uint32_t byte_cnt = 0;
    for(i = 0; i < utf8_id && txt[byte_cnt] != '\0'; i++) {    //CHANGED HERE
        uint8_t c_size = _lv_txt_encoded_size(&txt[byte_cnt]);
        byte_cnt += c_size > 0 ? c_size : 1;
    }

    return byte_cnt;
}
```

incity, 2021-02-28:
> Just guessing: what if you change `lv_txt_utf8_get_byte_id` like this:
>
> ```c
> static uint32_t lv_txt_utf8_get_byte_id(const char * txt, uint32_t utf8_id)
> {
>     uint32_t i;
>     uint32_t byte_cnt = [...]

use this can fix the problem

kisvegabor, 2021-03-01:
I've applied the fix.

I have this version of GCC: (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

commit 2618326f (2021-03-08): fix(txt): fix global buffer overflow in lv_txt_utf8_get_byte_id

fixes #2089
closed 2021-03-01 by kisvegabor