lvgl/lvgl #2090: How to support animation format(e.g. gif) through lv_img_decoder_t?

xiaoxiang781216, 2021-02-23:
### Introduce the problem

lv_img_decoder_t doesn't have the fields or callback to provide the mutiple frame data, how to write a portable widget to show the animation?

### Examples and cases

Improve img widget so show the animation.

### Suggested solution

Add animation info(e.g. frame_rate, frame_number, loop) to lv_img_header_t or lv_img_decoder_dsc_t. Add new callback(switch the frame) in lv_img_decoder_t.

embeddedt, 2021-02-23:
I am not sure whether all platforms will be able to keep up with decoding a new image on each frame; we will probably need some type of caching logic.

kisvegabor, 2021-02-24:
I was thinking a lot about how to create the [gif library](https://github.com/lvgl/lv_lib_gif) based on the image decoder but mian issue was how to invalidate the image after a given time.

It was my way of thinking:
1. In the decoder's `open_cb` start a one-shot timer with the time to the next frame and return the first frame
2. When the timer fires, start it again with the time to the next frame, invalidate the image object and update the `decoder_dsc` to send the new frame when LVGL asks.

It seems straight forward but the issue is that the timer can't invalidate the image object because when the image source is decoded it knows nothing about the obejcts. It's just image data to decode. 

Even worse is that the same image data can belong to multiple image objects. With static images, it's not an issue but with animations, it'd mean all the animations with the same image data will be in the same phase.

xiaoxiang781216, 2021-02-25:
> I was thinking a lot about how to create the [gif library](https://github.com/lvgl/lv_lib_gif) based on the image decoder but mian issue was how to invalidate the image after a given time.
>
> It was [...]

Why does image decoder involve the invalidation process? Image decoder only need to support decoding the specific frame passively and report the frame rate and total fame count. Widget(e.g. img) can control the play sequence and timing by self.
If the decoding is the bottleneck, decoder could schedule the decoding in background.

xiaoxiang781216, 2021-02-25:
> I am not sure whether all platforms will be able to keep up with decoding a new image on each frame; we will probably need some type of caching logic.

The top cache can be done by lv_img_cache, and decoder can manage an internal cache for animation.

kisvegabor, 2021-03-01:
> Why does image decoder involve the invalidation process? Image decoder only need to support decoding the specific frame passively and report the frame rate and total fame count.

This way `lv_img` (or a new object such as `lv_imganim`) needs to know the current `frame_id` and pass it down to image caching and finally to the image decoder, right?

xiaoxiang781216, 2021-03-02:
Yes, the current frame is managed by lv_imganim, so the animation and timing is driven by widget self and image decoder just need to decoder the specific frame passively. The interaction between widget and decoder is still one way comunication.

kisvegabor, 2021-03-04:
Sounds good. It requires adding the `frame_id` parameter to several functions:
- `lv_draw_img`
- `lv_img_decoder_open`
- `_lv_img_cache_open`
-

kisvegabor, 2021-04-14:
I've implemented it in #2197 PR but couldn't test it yet.
Let's discuss it in the PR.

closed 2021-04-14 by kisvegabor