lvgl/lvgl #2091: Enhanced image decoder

FASTSHIFT, 2021-02-23:
### Introduce the problem
Enhanced image decoder

### Examples and cases
The current image decoder can only support static pictures, if I want to use dynamic pictures, I need to refer to the practice of [lv_lib_gif](https://github.com/lvgl/lv_lib_gif/blob/master/lv_gif.c), and I cannot use a unified image decoder interface.

### Suggested solution
Consider that the image decoder supports dynamic pictures or videos.
To use like:
```C
lv_obj_t* img = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(img, "/res/mypic.gif");
```

FASTSHIFT, 2021-02-23:
I think it is possible to add a refresh task to the image decoder, notify the external decoder to decode, and notify the img object to update after the decoding is completed.

embeddedt, 2021-02-23:
Duplicate of #2090.

closed 2021-02-23 by embeddedt