lvgl/lvgl #2104: dev branch build break

xiaoxiang781216, 2021-02-26:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

All new PR can't pass the CI now, here is an example:
https://github.com/lvgl/lvgl/pull/2098/checks?check_run_id=1986660527

### To Reproduce

build the last dev branch

### Expected behavior

dev branch can pass CI.

### Screenshots or video

kisvegabor, 2021-03-01:
C/C++ test should work, but the Micropython test fails because Micropython's dev branch wasn't updated yet to v8.

xiaoxiang781216, 2021-03-13:
Ok, let's close it.

closed 2021-03-13 by xiaoxiang781216