lvgl/lvgl #2105: Gauge needle doesn't invalidate correctly

embeddedt, 2021-02-26:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The gauge needle is not invalidating correctly.

### To Reproduce

1. Visit https://lvgl.github.io/lv_sim_emscripten/ (or run `lv_demo_widgets`).
2. Click on the Visuals tab.
3. Scroll down to the gauge.
4. Observe that the gauge needle doesn't fully invalidate when it's close to or in the bottom half.

### Expected behavior

The gauge should invalidate correctly.

### Screenshots or video

[image]

embeddedt, 2021-02-26:
I suspect that this change might be the culprit. I haven't tested it yet.

https://github.com/lvgl/lvgl/commit/2edc95d7cd1c5083aebb6cfd60fcfec5f322c9d1#diff-0c0ce61e2747774d9fd03180f4541a8a85a54856e39afa3a4f44ff22d3c5c796R202

kisvegabor, 2021-03-01:
Thanks for the report, I've pushed a fix.

kisvegabor, 2021-03-01:
58708ca7