lvgl/lvgl #2107: Canvas Memory Usage

AfflatusX, 2021-02-26:
### Introduce the problem
Inspired by https://github.com/lvgl/lvgl/issues/2006 - I built a prototype that utilizes Thorvg and LVGL canvas to draw vector keyframe animations. (we use a different animation library, not lottie but similar idea).

However, using a full ARGB buffer takes WIDTH*HEIGHT*4 bytes - for a 475x475 screen, it's about 900 KB memory. To avoid screen tearing, we already set VDB to 100% of the screen (which is another 900 KB). With this design, drawing full screen animation/canvas will take 1.8 MB which is a huge amount for embedded devices.

I'm wondering if there is a way to "yield" the VDB to canvas drawing and then allowing other objects to be draw normally above it?

### Examples and cases
Allocate a 475*475 ARGB buffer = 900K.
Setting VDB to be the size of the screen to avoid tearing = 900K.

### Suggested solution
Allowing canvas to share the VDB buffer directly so we don't need to allocate the 900K buffer for it.

Apologize in advance if this is not the correct channel to post the question!

embeddedt, 2021-02-26:
Can your rendering logic be adjusted to render only part of the image at a time? If so, the correct approach is to make a custom widget. That will allow you to draw to the VDB yourself and not use the canvas.

@kisvegabor Is there an example for v7 of how to make a custom widget which draws to the VDB without using the `lv_draw` APIs?

AfflatusX, 2021-02-27:
@embeddedt thanks for the reply! it's tricky since I'm trying to draw some SVG so it's really hard to chop that into smaller blocks...

embeddedt, 2021-02-27:
Ah, I see.

The issue is that LVGL isn't designed to present you with a full buffer for rendering; it expects that rendering can be divided into smaller chunks. That was the original rationale behind adding the canvas widget - it allowed for rendering to be done to a normal bitmap and then copied by LVGL to the screen as necessary. Unfortunately, as you've noted, it is also one of the most expensive widgets in terms of RAM usage.

-------

(I'll preface this section by stating that I am not completely familiar with how the rendering logic works.)

Sharing the VDB buffer could be a potential solution, but I don't know if it will work in every case. Consider the following example:

[image]

As you can see, the gauges at the bottom of this page are partially cut off, as I haven't scrolled down enough to reveal them fully. Internally, LVGL is telling them that only their top half needs to be redrawn, and only provides them with enough VDB space to draw the top half. I don't think there is any memory provided for the bottom half to get drawn.

embeddedt, 2021-02-27:
This is an interesting problem; I'll think about it for a bit and see if I can come up with something.

AfflatusX, 2021-02-27:
@embeddedt - that's a very good example to show the limitation of such approach. However, for our product specifically the canvas layer serves as a background of the view - so it does not scroll. We can maybe think of it as a special component that always draws on VDB directly before other components start their drawing process. Just enabling that can enable tons of richer experiences without hurting the RAM usage.

embeddedt, 2021-02-27:
In that case the question for @kisvegabor is whether LVGL guarantees that the entire object will be redrawn in one pass if the VDB is large enough.

AfflatusX, 2021-03-01:
> In that case the question for @kisvegabor is whether LVGL guarantees that the entire object will be redrawn in one pass if the VDB is large enough.

I think that's the definition for VDB (and the reason why 100% VDB prevents tearing), but I could be wrong. Thanks again @embeddedt  for digging into this! Let know if there is anything I can do to help!

kisvegabor, 2021-03-02:
A ThorVG port would be great, so let's dig into this :slightly_smiling_face: 

> In that case the question for @kisvegabor is whether LVGL guarantees that the entire object will be redrawn in one pass if the VDB is large enough.

Definitely no. Let's talk about 2 different scenarios:
1. Only a small portion of a widget has changed (e.g. the needle of the gauge). In this case LVGL tells the widget to redraw only the given area. If this area is larger than the VDB, LVGL splits the area to VDB sized areas and ask the widget to redraw in multiple passes.
2. A larger portion of the screen is invalidated that involves multiple widgets, e.g. a button with a label. (Let's say VDB is large enough the draw the button in one pass.) LVGL will ask the button to redraw itself, and after that, asks the label too. However, if the button has rounded corners, or opacity, its background (e.g. the screen itself) needs to be redrawn before the button. But only the button-sized are of the screen.

So imagine if you have a large SVG image on a canvas and you add a button to it.  If you press the button, only the buttons area will be invalidated (the first byte of the VDB will set the to buttons left top point) but you need to redraw the SVG too. You have the whole VDB, but it doesn't start at the screen 0;0 coordinate.

This way of rendering has advantages (speed, memory usage) but it's custom and can't work well with existing rendering engines (e.g. OpenGL). To mitigate this in v8 I've changed how "true double buffering" works. In v8 even if only 1 pixel changes on the screen, the whole screen will be invalidated. If you have a screen-sized VDBs they will always start at the screen's 0;0 coordinate and you can simply draw there. 
If you set only 1 screen-sized buffer for LVGL you can still use a `rounder_cb` in the display driver to make every invalidated area screen sized. It will also give you full-screen refreshes in one pass.

So seemingly there is a solution but I think the "canvas method" is also reasonable:
- If you want large SVGs probably you have a high-end system with external RAM where the canvas's buffer can be located
- Real-time SVG rendering can be slow
- IMO most people would use SVG icons instead of SVG wallpapers and the size of small canvases is manageable.

---
BTW, in v7 VDB is called display buffer, which is not so straightforward. In v8 it could be renamed to `lv_disp_render_buffer` or so. What do you think?

embeddedt, 2021-03-02:
> In v8 even if only 1 pixel changes on the screen, the whole screen will be invalidated. If you have a screen-sized VDBs they will always start at the screen's 0;0 coordinate and you can simply draw there.

I was confused about this when you mentioned it in #2058 as well. Doesn't this mean that there are a lot of unnecessary computations happening on each frame, or am I missing something?

kisvegabor, 2021-03-02:
> I was confused about this when you mentioned it in #2058 as well. Doesn't this mean that there are a lot of unnecessary computations happening on each frame, or am I missing something?

That's true, but this a more standard method hence it allows using higher-level GPU interfaces e.g. (SDL, OpenGL, OpenGL ES). It turned out if we use these GPUs we have no read and write access to the frame buffer directly and all the LVGL drawers should be reimplemented with the GPU. 

LVGL's SW renderer is not suitable for full HD or larger displays but with GPU they can be easily used. Even if the whole screen needs to be redrawn, it will be faster in the end. 

Besides, with the previous implementation (in v7) it was required to keep the 2 frame buffer in sync. It means when an area was drawn in  fb1 that area needed to be copied to fb2 to let LVGL work on the same image as we have on the screen right now. However if e.g. a tab view was scrolled it meant redrawing the whole screen + copying the whole screen. (I found that copying from ext. RAM to ext RAM is very slow because it needs a read + write for every byte).

If someone needs the v7 behavior he can reimplement it by using 2 screen-sized frame buffers and 1 or 2 LVGL draw buffers and copy the areas into inactive frame buffer manually. And when the last area is sent to `flush_cb` fb1 and fb2 can be swapped and the dirty areas can be copied to keep the frame buffer in sync. In fact, it will be faster than the current v7 method because LVGL rendered into internal RAM buffers which is way faster than rendering into ext RAM buffers. 

If you think it's important we can add an `lv_disp_copy_invalidated areas(disp)` function to make it easier to go back to the old method.

closed 2021-05-14 by kisvegabor