lvgl/lvgl #2109: [v8] NULL pointer access in lv_example_dropdown_3

embeddedt, 2021-02-28:
The dropdown's list has not been initialized yet when this gets called. It doesn't seem to get created till the dropdown first opens.

https://github.com/lvgl/lvgl/blob/8f48cf5cc7b0e68a22b44f6522394b67db382d89/examples/widgets/dropdown/lv_example_dropdown_3.c#L29

kisvegabor, 2021-02-28:
IT should be fixed now.

embeddedt, 2021-03-01:
It was working yesterday, but now the example seems to just be crashing on STM32. I haven't had a chance to check GDB or Valgrind yet.

kisvegabor, 2021-03-01:
I've found a tricky bug in `lv_mem` but fixed a few hours ago. It works well here now.

embeddedt, 2021-03-01:
I forgot to mention something. I get [this error](https://github.com/lvgl/lvgl/blob/23a8c0b68a2ddc1fe1e2a30bd268212c8b0c2fe4/src/lv_misc/lv_mem.c#L269) printed in the log during display initialization because my realloc implementation returns NULL when a size of 0 is passed. The crash happened after I added the following lines above `LV_MEM_CUSTOM_REALLOC`:

```c
    if(new_size == 0) {
        LV_MEM_CUSTOM_FREE(data_p);
        return &zero_mem;
    }
```

... which I've just realized is buggy because `&zero_mem` will get passed to my realloc implementation at a later time, and it's not from the heap. That's probably why it's crashing. Sorry for the confusion.

kisvegabor, 2021-03-01:
I've fixed it as you suggested: 367e3a31

embeddedt, 2021-03-01:
That fixed the first part of the problem. Now the demo loads but crashes somewhere in here when I try to open the dropdown.

```c
#0  lv_anim_del (var=var@entry=0x6009606c, exec_cb=exec_cb@entry=0x0) at lvgl/src/lv_misc/lv_anim.c:160
#1  0x0804e76c in trans_del (obj=0x60095ccc, part=<optimized out>, prop=prop@entry=LV_STYLE_COLOR_FILTER_DSC, tr_limit=tr_limit@entry=0x6009677c) at lvgl/src/lv_core/lv_obj_style.c:840
#2  0x0804f10a in trans_anim_start_cb (a=<optimized out>) at lvgl/src/lv_core/lv_obj_style.c:926
#3  0x08049d3a in anim_task (param=<optimized out>) at lvgl/src/lv_misc/lv_anim.c:464
```

kisvegabor, 2021-03-01:
Does it work with `LV_MEM_CUSTOM 0`?

embeddedt, 2021-03-01:
With LV_MEM_CUSTOM == 0 it immediately crashes and doesn't even show the dropdown. I suspect it's memory corruption.

(I should note that I've used `LV_MEM_CUSTOM == 1` with v7 for months and never had a problem, so I think my heap allocator is fine.)

kisvegabor, 2021-03-01:
The immediate crash is better because it's easier to debug :) 

Can you enable logging with TRACE level? 

I'll try it on embedded hardware today or tomorrow too.

embeddedt, 2021-03-01:
[trace.txt](https://github.com/lvgl/lvgl/files/6063045/trace.txt)

The crash happens at the end of that log, so it seems to succeed in

kisvegabor, 2021-03-01:
I've just fixed a bug with `LV_MEM_CUSTOM == `1: 47b8893f

I've run the stress test with both  `LV_MEM_CUSTOM` 1 and 0 for a couple of minutes without issues on  STM32f769DISCO.

If it still not working try `LV_MEM_AD_JUNK 0` in `lv_mem.c`.

embeddedt, 2021-03-01:
Okay; that seems to have changed the behavior a bit.

I've now tracked it down to a crash when I call `lv_disp_set_rotation`. The end member of the `disp->screens` array (index 3) is getting stomped on by heap allocations. I'm not sure why yet (still tracing). I put `LV_MEM_CUSTOM == 0` for testing.

embeddedt, 2021-03-01:
Found it. :slightly_smiling_face: 

It was a minor issue in `lv_disp_drv_update`. https://github.com/lvgl/lvgl/commit/db7a6e4facc0d47ebc3ba35d7481d983b4f1cdf1

kisvegabor, 2021-03-01:
Yuppi! Thank you :slightly_smiling_face:

embeddedt, 2021-03-01:
The example now runs smoothly! This can be closed.

closed 2021-03-01 by embeddedt

kisvegabor, 2021-03-01:
Another related fix: 122f6c12

embeddedt, 2021-03-01:
Good to know!

kisvegabor, 2021-03-01:
Now v8 is working at least for you, @xiaoxiang781216, and me. It seems all the catastrophic things are fixed. :slightly_smiling_face:

embeddedt, 2021-03-01:
The new scrolling is very impressive!

kisvegabor, 2021-03-01:
Thanks! Have you checked the scroll examples?