lvgl/lvgl #2113: Roller didn't match to the text when release the screen

sk123sos, 2021-03-02:
When I scroll the roller and release it ,some time it may not align the select part with the text.

Here's my code

roller_b61d = lv_roller_create(main_screen, NULL);
    lv_obj_set_size(roller_b61d, 293, 114);
    lv_roller_set_fix_width(roller_b61d,293);
    lv_obj_set_pos(roller_b61d, 462, 280);
    lv_roller_set_options(roller_b61d,
    "1""\n"
    "2""\n"
    "3""\n"
    "4""\n"
    "5""\n"
    "6""\n"
    "7""\n"
    "8"
,LV_ROLLER_MODE_INIFINITE);
    lv_roller_set_visible_row_count(roller_b61d, 5);
    static lv_style_t style_roller_b61d;
    lv_style_init(&style_roller_b61d);
    lv_style_set_text_font(&style_roller_b61d,LV_STATE_DEFAULT,&ali_font_40);
    lv_style_set_text_line_space(&style_roller_b61d,LV_STATE_DEFAULT,20);
    lv_obj_add_style(roller_b61d,LV_ROLLER_PART_BG, &style_roller_b61d);
    static lv_style_t _sty_roller_b61d;
    lv_style_init(&_sty_roller_b61d);
    lv_style_set_bg_color(&_sty_roller_b61d,LV_STATE_DEFAULT,LV_COLOR_MAKE(0x01, 0xa2, 0xb1));
    lv_style_set_pad_top(&_sty_roller_b61d,LV_STATE_DEFAULT,5);
    lv_style_set_pad_bottom(&_sty_roller_b61d,LV_STATE_DEFAULT,5);
    lv_style_set_text_color(&_sty_roller_b61d,LV_STATE_DEFAULT,LV_COLOR_MAKE(0xff, 0xff, 0xff));
    lv_style_set_text_font(&_sty_roller_b61d,LV_STATE_DEFAULT,&ali_font_45);
    lv_style_set_text_letter_space(&_sty_roller_b61d,LV_STATE_DEFAULT,1);
    lv_obj_add_style(roller_b61d,LV_ROLLER_PART_SELECTED, &_sty_roller_b61d);

It seems normal but there are some problems with it
[image]
[image]

sk123sos, 2021-03-02:
The version I'm using is 7.9.0

embeddedt, 2021-03-02:
Try changing `typedef int16_t lv_coord_t` to `typedef int32_t lv_coord_t` in `lv_conf.h`.

sk123sos, 2021-03-03:
> Try changing `typedef int16_t lv_coord_t` to `typedef int32_t lv_coord_t` in `lv_conf.h`.

I have tryed it but it dosen't work

sk123sos, 2021-03-03:
[image]
when this bug happened,the states of roller seems like this and if I touch any space on the screen,the roller will send some event, go on scroll itself and works well.
The states changes after I touch the screen is here
[image]

sk123sos, 2021-03-03:
Seems it didn't send the event LV_EVENT_VALUE_CHANGED

turoksama, 2021-03-04:
It looks like somthing wrong with your indev driver.
Does it return "released" signal actually whenever you moved your finger off the screen? Hope I was wrong about this.

sk123sos, 2021-03-04:
> It looks like somthing wrong with your indev driver.
> Does it return "released" signal actually whenever you moved your finger off the screen? Hope I was wrong about this.

It does send the released event,with the test code it will print 'custom code' when released event was detected.

kisvegabor, 2021-03-08:
Can you reproduce the issue in a simulator?

sk123sos, 2021-03-09:
> Can you reproduce the issue in a simulator?

No I can't ,it seems to be my hardware problem.I used it on a development board with linux,can you give me some advice that what should I check with my project.

sk123sos, 2021-03-09:
> Can you reproduce the issue in a simulator?

Except rollers ,all other widgets work well,it confuses me a lot.

kisvegabor, 2021-03-10:
Please `printf` what happens in your indev's `read_cb`. When the `LV_INDEV_STATE_REL/PR` is set and what are the x/y coordiantes sent to LVGL.

sk123sos, 2021-03-11:
> Please `printf` what happens in your indev's `read_cb`. When the `LV_INDEV_STATE_REL/PR` is set and what are the x/y coordiantes sent to LVGL.

I find that if I scroll the screen ,stop my finger with screen pressed and than release it,this mistake will not happen.
But if I scroll and releas it immediately things go wrong,but two logs here seems the same,the only difference is the time I pressed it.
[image]

sk123sos, 2021-03-11:
> Please `printf` what happens in your indev's `read_cb`. When the `LV_INDEV_STATE_REL/PR` is set and what are the x/y coordiantes sent to LVGL.

I got it,some mistake happened with our hardware lib,I have fixed it ,thanks for you advice

kisvegabor, 2021-03-17:
Glad to hear you have found the issue!

closed 2021-03-17 by kisvegabor