lvgl/lvgl #2114: [v8] Improve layout recalculation efficiency

embeddedt, 2021-03-02:
Currently layouts seem to be recalculated any time an object is created or its size is changed. Ideally, calculation should be deferred to the next render cycle to prevent lag when adding/changing/removing several objects.

> I know at least 4 render strategies (with variants of each):
- immediate render
- postponed render to next event loop
- postponed render to animation frame (fixed frame rate)
- postponed animation frame with full redraw

_Originally posted by @puzrin in https://github.com/lvgl/lvgl/issues/1852#issuecomment-788960310_

puzrin, 2021-03-02:
I guess, current update strategy is sync. In this case, i'd suggest to NOT add freeze/unfreeze methods until you decide 100% to not use more advanced approaches.

Deferred updates are cool, but have one disadvantage for "old school" programmers - it's impossible to read real value after write. For example, set width 70% and try to read in pixels.

embeddedt, 2021-03-02:
> Deferred updates are cool, but have one disadvantage for "old school" programmers - it's impossible to read real value after write. For example, set width 70% and try to read in pixels.

Perhaps we could have an `lv_layout_force_recalc` function which would force the layout to be recalculated early. If it's used sparingly, the performance shouldn't be a major problem.

puzrin, 2021-03-02:
From my experience in webdev, manual refresh fixes less problems than creates :).

I'd prefer to have deferred updates only.

kisvegabor, 2021-03-02:
@embeddedt wrote this in [this comment](https://github.com/lvgl/lvgl/issues/1852#issuecomment-788941953):
> I should also note that this isn't a theoretical issue only - the UI completely freezes for about 3/4 to 1 second each time I press the + button, which would be undesirable if any animations were visible on screen.

With how many widgets have you seen it?

embeddedt, 2021-03-02:
This was with `lv_example_scroll_3`, so just the one list.

puzrin, 2021-03-02:
@embeddedt do i understand right, v6/v7 had no perf issues with such demo but v8 has? Is this regression or old problem?

embeddedt, 2021-03-02:
Yes. This exact demo didn't exist in v7, but adding elements to a list had almost 0 performance penalty. v8 is using the new layout system and adding an element to a list now takes quite a while. So I would consider this a regression.

kisvegabor, 2021-03-02:
LAyouts in v8 should be a lot faster than v7 layouts so there should be another issue here. Have you disabled the asserts and trace level logging in `lv_conf.h`? They make LVGL much slower.

puzrin, 2021-03-02:
@embeddedt What happens, if you hide screen, add all elements you need and display back? That should emulate "poor man deferred update".

embeddedt, 2021-03-02:
I will try disabling the debug features but logging rarely causes this much lag for me.

> What happens, if you hide screen, add all elements you need and display back?

@puzrin I'm not sure that the actual rendering is the problem here - I think the issue might be the repeated calculations I am seeing in the log, where the layout updates multiple times for each object.

embeddedt, 2021-03-02:
I disabled the asserts and now it seems to be a lot faster, which is surprising. I'll try bisecting the settings to see if I can find which assert slows it down so much.

embeddedt, 2021-03-02:
Ok, I stand corrected. Logging was definitely the issue - I removed it, turned the asserts back on, and now elements get added instantly. Nonetheless, I still find the number of recalculations concerning.

puzrin, 2021-03-02:
Let me clarify terms. I think, web is good reference for our case. In browser "rendering" process consists of 2 separate phases:

- calculate layout
- draw

When add/modify/delete element causes huge layout change, it's called "reflow". That's very expensive operation.

- When you insert child to list, with proper styles, that should not cause reflow but can cause redraw.
- If you apply extra efforts to remove background transparency & shadows - then redraw will not happen too (almost)

---

Recalculations may be caused:

- by styles (transparent background or outer shadows)
- by bug
- by lack of deferred updates

Some calculations require several passes (for example, in flexbox). I did not digged demo and can't say exactly what happened.

kisvegabor, 2021-03-02:
Meanwhile, I also tried in on STM32F429. Adding the 3rd list item took 2ms. Adding the 70th list item took 3 ms (no logging, no asserts). 

I also compared v7 with v8 in simply creating 70 list items:
- v7:  258 ms
- v8: 176 ms
- v8 using LV_MEM_CUSTOM 1: 144 ms
- v8 disabling list layout and reenabling it when the items are added: 80 ms

Really there are several layout recalculations:
1. on child creation/deletion
2. child's size change
3. on certain style property changes on parent
4. on parent size change
5. on child property changes (e.g. hidden change)

Some local optimizations can be added to the list, e.g. add the layout to the item only when all children are added and configured. But it was fast anyway because only 2 children were repositioned (an icon and a text). And it happens only once. 

The larger issue is that when a list button is added all list buttons need to be repositioned. When the 70th list button is added, 70 buttons, when the 71th is added all the 71 buttons again. It can be mitigated by manually disabling the layout of the list and reenabling it when all the children are added (as in the last measurement). We can even add a flag to temporarily disable layouting. 

I'm not a big fan of delayed layout recalculation because the user might expect to know the "real" position of the object immediately.

embeddedt, 2021-03-02:
> The larger issue is that when a list button is added all list buttons need to be repositioned. When the 70th list button is added, 70 buttons, when the 71th is added all the 71 buttons again. It can [...]

This is a good idea.

puzrin, 2021-03-02:
> I'm not a big fan of delayed layout recalculation because the user might expect to know the "real" position of the object immediately.

1. IMO it's a bad practice to expect immediate update. Do you have examples when this is really required? I mean, this already is a "past of web", and it is not possible to invent something new in this area.
2. How often sync updates required? Why "almost all" should suffer from 0.01% of people who wish "ancient things"? For most of cases deferred updates "just works".

Actually, deferred update happens in nearest animation frame. It's possible to make "wait_for_animation_frame()", but i'm not sure this is required at all.

> We can even add a flag to temporarily disable layouting.

That's possible, but dirty. There will be endless mistakes on user side. In worst case i would prefer reverse thing - option to temporary disable deferred updates.

kisvegabor, 2021-03-03:
Okay, let's think about deferred updates. Being in sync is really required only in the 0.01% of the cases and there can be an `update_now()` function.

So what we need is a "layout dirty" flag in every object and in every animation frame (screen redraw in our case) check for the flags from the children and update the layout of the child if needed. And go to the parent...

Probably there should be a global layout dirty flag that indicates if there is at least one dirty layout to avoid reiterating every object in every screen redraw.

puzrin, 2021-03-03:
I think, those flags are good start, and can be improved later if required, transparently for user.

The only config option i can suggest is "max animation FPS" (minimal allowed delay in ms between animation frames). Seems nothing more to discuss.

kisvegabor, 2021-03-03:
I've added it and pushed to `dev` (0d38d100). It was simpler than I anticipated.

I found a case where the deferred calculation can be problematic and it's actually the already mentioned `lv_example_scroll_3()`. It should work like this:

[image]

But woked like this:

[image]

Relevant code (the add button"s event):
```c
static void float_btn_event_cb(lv_obj_t * float_btn, lv_event_t e)
{
    if(e == LV_EVENT_CLICKED) {
        lv_obj_t * list = lv_event_get_user_data();
        char buf[32];
        lv_snprintf(buf, sizeof(buf), "Track %d", btn_cnt);
        lv_obj_t * list_btn = lv_list_add_btn(list, LV_SYMBOL_AUDIO, buf, NULL);
        btn_cnt++;

        lv_obj_move_foreground(float_btn);

        lv_obj_update_layout(list); //REQUIRED!!!
        lv_obj_scroll_to_view(list_btn, LV_ANIM_ON);
    }
}
```

The problem is the new button is created on the top of the list and `lv_obj_scroll_to_view` scrolls to the top of the list.
`lv_obj_update_layout` pushes the new item to the bottom and `lv_obj_scroll_to_view` works well after that.

embeddedt, 2021-03-03:
Maybe `lv_obj_scroll_to_view` could update the object's parent's layout? That would help for the majority of cases, and we could add notes in the documentation recommending the use of `lv_obj_update_layout` if there is weird behavior.

puzrin, 2021-03-03:
> I found a case where the deferred calculation can be problematic and it's actually the already mentioned

Thank you for example. I think, problem is that animation should be attached/activated after layout calculated (after animation frame complete). Then we probably need something of this:

1. `subscribe_once(DEFERRED_DONE, activate_animation_handler)` - event on the end of deferreds flush
2. `flush_deferred(screen)` - sync function to flush deferred ops
3. Forced `flush_deferred(screen)` inside of any property read
4. https://developer.mozilla.org/en-US/docs/Web/API/Window/cancelAnimationFrame - api to schedule animations. ~ the same as (1)

What do you think?

> Maybe `lv_obj_scroll_to_view` could update the object's parent's layout?

As far as i understand, this is standalone featue, not from core, place into core only for convenience. In this case i see no problem to force flush deferreds.

kisvegabor, 2021-03-03:
> Maybe lv_obj_scroll_to_view could update the object's parent's layout? That would help for the majority of cases, and we could add notes in the documentation recommending the use of lv_obj_update_layout if there is weird behavior.

Sounds reasonable. `lv_obj_scroll_to_view` always needs recalculated layouts. However, not only the parents' layout needs to be recalculated but the whole screen because an ancestor's layout can change the size of the list's size.

Added here: 376da66d

puzrin, 2021-03-03:
What do we have now, comparing with https://github.com/lvgl/lvgl/issues/2114#issuecomment-789143521? Goal reached?

kisvegabor, 2021-03-03:
@puzrin 
> 1. subscribe_once(DEFERRED_DONE, activate_animation_handler) - event on the end of deferreds flush
2. flush_deferred(screen) - sync function to flush deferred ops
3. Forced flush_deferred(screen) inside of any property read
4. https://developer.mozilla.org/en-US/docs/Web/API/Window/cancelAnimationFrame - api to schedule animations. ~ the same as (1)

Sounds like a good concept. It'd be overkill only for this thing, but let's keep it in mind.

> What do we have now, comparing with #2114 (comment)? Goal reached?

I do a quick test in a few minutes.

kisvegabor, 2021-03-03:
- add 70 list items
  - non deferred:  103 ms
  - deferred: 77 ms

- add 200 list items
  - non deferred: 328 ms
  - deferred: 256 ms

Honestly, it's not a big difference.

kisvegabor, 2021-03-03:
There were somethings wrong with my test code. See the updated result above.

puzrin, 2021-03-03:
Nice. With upgraded allocator should become even more impressive.

kisvegabor, 2021-03-03:
It was with the new allocator :)

kisvegabor, 2021-03-03:
The same 200 items in v7 need 1132 ms.

puzrin, 2021-03-03:
What is the initial speed of v8, without deferreds and without new allocator?

embeddedt, 2021-03-03:
@kisvegabor v8 has almost 4x improvement in performance over v7? You've sold me. :wink:

kisvegabor, 2021-03-04:
@puzrin
> What is the initial speed of v8, without deferreds and without new allocator?

876 ms. It's closer to v7. However, in v7 the time grows much faster (a lot of recursions, extra recalculations) than in v8 (O(N) time). 

@embeddedt 
> v8 has almost 4x improvement in performance over v7? You've sold me. wink

Yes :) But note that it's only for object creation. Rendering speed remained quite the same.

excitedbox, 2021-03-07:
For rendering speed improvements I can only keep pointing you to the iteration problem in the style system. Especially with the new memory allocator it should be even faster when switching to a hashmap from a linked list. 

Any place where you use style->map[i] or style->map[id+1] you could then be using style->map[current_id] and eliminate the loop.

kisvegabor, 2021-03-08:
@excitedbox 

Tomorrow I'll make some measurements on the style systems performance of v8 and report it in the related issue.

---

`lv_example_table_2()` is also an interesting one. It's a table with 1 column and a custom drawer that adds switches. On STM32F429 it was created in 22 ms (with 200 rows) and rendered 2 times faster than a list.

[image]

puzrin, 2021-03-08:
> It's a table with 1 column and a custom drawer that adds switches.

Note, in html5 table is not used for layout anymore. It's used to display table data only. The example above should be flexbox (flexbox for list with separators + flexbox for item content).

Hm... `:first-child` / `:last-child` will be required to remove first/last separator (border). However, for static lists this can be implemented via manual assign of secondary class (not a big deal, but not nice)

kisvegabor, 2021-03-08:
> Note, in html5 table is not used for layout anymore. It's used to display table data only. The example above should be flexbox (flexbox for list with separators + flexbox for item content).

It's no a layout just a simple table. I've mentioned here because we were talking about the speed of object createion. 

> Hm... :first-child / :last-child will be required to remove first/last separator (border). However, for static lists this can be implemented via manual assign of secondary class (not a big deal, but not nice)

I agree.

kisvegabor, 2021-03-09:
@puzrin 
BTW, it's exactly what you have asked for the solder station: a very light-weighted, very customizable list.

puzrin, 2021-03-09:
I'm interested in standard BEM-styled containers (flexbox). All those custom optimizations clutter code. IMO v8 looks promissing.

excitedbox, 2021-03-13:
Actually you wouldn't use flex for that. You would use a div container or block with scroll enabled and overflow hidden or a textarea with scroll. Then you would add a style to the row (bottom border) to add the lines etc. 

Flex is used for breaking content across lines depending on width or making it stretch to fit the space. Grid is used to divide an area into a coordinate system with rows and columns.

A very intuitive CSS framework that can help people understand how CSS is best used is TailwindCSS because their documentation has so many examples it lets you see what each thing is used for really nicely.

c10ud, 2021-03-29:
If I were to show ~1000 items on a grid-like layout (ie. 3 columns and many rows), which would the best way to do this with current (dev) lvgl be?

I was looking into layouts (grid, flex) but also table would be a good match if it could display generic objects or maybe just buttons (image button could be enough)

kisvegabor, 2021-03-29:
You can use any of grid, flex, or table.

For a table with custom cells see [this example](https://github.com/lvgl/lvgl/blob/master/examples/widgets/table/lv_example_table_2.c). It uses very little memory and created very quickly.

Flex should work very well using a row layout with "wrap".

However, grid is not the best choice because auto-placement is not supported so you need to place the items into their cells one-by-one.

kisvegabor, 2021-04-24:
With the postponed layout recalculation is implemented the original issue is solved so I close this now. 

Thank you very much for your contribution! :slightly_smiling_face:

closed 2021-04-24 by kisvegabor