lvgl/lvgl #2115: There is a problem with ellipsis rendering in Arabic text

Changes729, 2021-03-03:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Arabic ellipsis maybe not right.

### To Reproduce
```c
    lv_obj_t* label1 = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_DOT);
    lv_label_set_text(label1, "Hello World!");
    lv_obj_set_width(label1, 150);
    lv_obj_align(label1, NULL, LV_ALIGN_IN_LEFT_MID, 50, -30);

    lv_obj_set_style_local_border_color(label1, LV_LABEL_PART_MAIN,
                                        LV_STATE_DEFAULT, LV_COLOR_BLACK);
    lv_obj_set_style_local_border_width(label1, LV_LABEL_PART_MAIN,
                                        LV_STATE_DEFAULT, 2);

    lv_obj_t* label1_ellipsis = lv_label_create(lv_scr_act(), label1);
    lv_obj_set_width(label1_ellipsis, 60);
    lv_obj_align(label1_ellipsis, NULL, LV_ALIGN_IN_LEFT_MID, 50, -60);

    lv_obj_t* label2 = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
    lv_obj_set_width(label2, 100);
    lv_label_set_text(label2, "مرحبا بالعالم!");
    lv_obj_align(label2, NULL, LV_ALIGN_IN_RIGHT_MID, -350, 30);
    lv_obj_set_style_local_text_font(label2, LV_LABEL_PART_MAIN,
                                     LV_STATE_DEFAULT, &font_arabic);

    lv_obj_set_style_local_border_color(label2, LV_LABEL_PART_MAIN,
                                        LV_STATE_DEFAULT, LV_COLOR_BLACK);
    lv_obj_set_style_local_border_width(label2, LV_LABEL_PART_MAIN,
                                        LV_STATE_DEFAULT, 2);

    lv_obj_t* label2_ellipsis = lv_label_create(lv_scr_act(), label2);
    lv_obj_set_width(label2_ellipsis, 60);
    lv_obj_align(label2_ellipsis, NULL, LV_ALIGN_IN_RIGHT_MID, -350, 60);
```

### Expected behavior
The ellipsis should be shown at the end of the Arabic text.

### Screenshots or video
[image]

Changes729, 2021-03-03:
My font of the Arabic has no graph for '.'

Changes729, 2021-03-03:
I have tried to disable the BIDI rendering in method lv_label_get_letter_on. and the result here seems like what I want to see.

[image]

kisvegabor, 2021-03-04:
If there is no `.` in the fonts what character would you expect to be displayed as ellipses?

Changes729, 2021-03-05:
> If there is no `.` in the fonts what character would you expect to be displayed as ellipses?

I'm sorry I didn't make myself clear.

### code
```cpp
lv_obj_t* label1 = lv_label_create(lv_scr_act(), NULL);
lv_label_set_long_mode(label1, LV_LABEL_LONG_DOT);
lv_label_set_text(label1, "Hello World!");
lv_obj_set_width(label1, 150);
lv_obj_align(label1, NULL, LV_ALIGN_IN_LEFT_MID, 50, -30);

lv_obj_set_style_local_border_color(label1, LV_LABEL_PART_MAIN,
                                    LV_STATE_DEFAULT, LV_COLOR_BLACK);
lv_obj_set_style_local_border_width(label1, LV_LABEL_PART_MAIN,
                                    LV_STATE_DEFAULT, 2);

lv_obj_t* label1_ellipsis = lv_label_create(lv_scr_act(), label1);
lv_obj_set_width(label1_ellipsis, 60);
lv_obj_align(label1_ellipsis, NULL, LV_ALIGN_IN_LEFT_MID, 50, -60);

lv_obj_t* label2 = lv_label_create(lv_scr_act(), NULL);
lv_label_set_long_mode(label2, LV_LABEL_LONG_DOT);
lv_obj_set_width(label2, 100);
lv_label_set_text(label2, "مرحبا بالعالم!");
lv_obj_align(label2, NULL, LV_ALIGN_IN_RIGHT_MID, -350, 30);
lv_obj_set_style_local_text_font(label2, LV_LABEL_PART_MAIN,
                                    LV_STATE_DEFAULT, &font_arabic);

lv_obj_set_style_local_border_color(label2, LV_LABEL_PART_MAIN,
                                    LV_STATE_DEFAULT, LV_COLOR_BLACK);
lv_obj_set_style_local_border_width(label2, LV_LABEL_PART_MAIN,
                                    LV_STATE_DEFAULT, 2);

lv_obj_t* label2_ellipsis = lv_label_create(lv_scr_act(), label2);
lv_obj_set_width(label2_ellipsis, 60);
lv_obj_align(label2_ellipsis, NULL, LV_ALIGN_IN_RIGHT_MID, -350, 60);

lv_obj_t* label3_ellipsis = lv_label_create(lv_scr_act(), label2);
lv_obj_set_width(label3_ellipsis, 70);
lv_obj_align(label3_ellipsis, NULL, LV_ALIGN_IN_RIGHT_MID, -350, 90);
```

### Test Result
the place of the ellipses maybe not right.
[image]

### Excepted Result
maybe should render like this?
[image]

kisvegabor, 2021-03-10:
Thanks!

I'm a little bit busy now and can take a look into it a little later.

kisvegabor, 2021-05-14:
Not stale

closed 2023-12-13 by kisvegabor