lvgl/lvgl #2116: Memory allocator revisit

puzrin, 2021-03-03:
### Introduce the problem

Seems lvgl's internal memory allocator is very old/simple and could be more effective. I know, config allows to replace allocator. But it would be nice to have list of advised alternatives.

### Suggested solution

I only googled "allocator for embedded" and not used those in real projects. But looks interesting:

- https://github.com/rhempel/umm_malloc
- https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/MemMang

Note. I think, after "v8 + deferred updates" allocator performance may become critical. Right now - nothing too worry about too much.

Priority: not urgent, not breaking.

xiaoxiang781216, 2021-03-03:
How about this one:
https://github.com/mattconte/tlsf
More simple(2 files with 1000 lines), but very fast and less fragmentation, here has more information and comparison:
http://www.gii.upv.es/tlsf/main/docs

puzrin, 2021-03-03:
@xiaoxiang781216 any suggestions are welcome.

More projects available by googling [github embedded allocator](https://www.google.com/search?q=github+embedded+allocator).

At first step we need to understand, is it possible to get significant speed/fragmentation benefits by swapping allocator to something well-know. I have no experience to recommend anything final, but at first glance there is room for improve (auto pools for objects, lock-less multi-threading and so on).

kisvegabor, 2021-03-03:
@puzrin 

> https://github.com/rhempel/umm_malloc

See the description [here](https://github.com/rhempel/umm_malloc#detailed-description) There are 20 bytes of metadata on every block. Now it's 4 bytes

> https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/MemMang

`heap_4.c` looks interesting but needs deeper investigation.

@xiaoxiang781216 
> https://github.com/mattconte/tlsf

O(1) time for everything lósounds great!

I'm not sure what this line means in the README: "Low overhead per TLSF management of pools (~3kB)". 3 kb of RAM usage by default?

[__builtin_clz](https://github.com/mattconte/tlsf/blob/master/tlsf.c#L56) might be not implemented on every platform. 

I can't see if it performs auto defragmentation or not.

---

In summary, IMO these are the most valuable feature of lv_mem:
- 4 byte overhead/allocation
- defragmentation by joining adjacent free block
- can give memory usage reports
- no RAM/flash overhead due to extra tables

xiaoxiang781216, 2021-03-03:
> @puzrin
> 
> > https://github.com/rhempel/umm_malloc
> 
> See the description [here](https://github.com/rhempel/umm_malloc#detailed-description.) There are 20 bytes of metadata on every block. Now it's 4 bytes
> 
> > https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/MemMang
> 
> `heap_4.c` looks interesting but needs deeper investigation.
> 
> @xiaoxiang781216
> 
> > https://github.com/mattconte/tlsf
> 
> O(1) time for everything sounds great!
> 
> I'm not sure what this line means in the README: "Low overhead per TLSF management of pools (~3kB)". 3 kb of RAM usage by default?
> 

3KB is for 2GB heap size, the overhead reduce dramatically for the small heap. The heap size is controlled by one macro:
https://github.com/mattconte/tlsf/blob/master/tlsf.c#L237-L244
The code can calculate the right value once we update the macro.

> [__builtin_clz](https://github.com/mattconte/tlsf/blob/master/tlsf.c#L56) might be not implemented on every platform.
> 

The code try to detect the compiler feature, if clz doesn't support the default one is provided:
https://github.com/mattconte/tlsf/blob/master/tlsf.c#L150-L173

> I can't see if it performs auto defragmentation or not.
> 

Yes, it perform defragmentation in the free immediately wich O(1). Overhead per allocation is also 4 bytes.
Here is the detail report and comparison:
http://www.gii.upv.es/tlsf/files/papers/tlsf_slides.pdf

> In summary, IMO these are the most valuable feature of lv_mem:
> 
> * 4 byte overhead/allocation
> * defragmentation by joining adjacent free block
> * can give memory usage reports
> * no RAM/flash overhead due to extra tables

kisvegabor, 2021-03-03:
Thanks, @xiaoxiang781216 . It's impressive. I made some performance comparisons. 100 list items were created (300 objects in total) on ST32F429. Note that the measurements include the time of other calculations too, not only the mallocs
- built in: 119 ms
- tlsf: 119 ms (exactly the same as built-in)
- lvgl: 178 ms

So there is room for optimization. The only question is if we can add a memory monitoring feature to tlsf (free/used size, fragmentation percentage, largest free block). I found it extramly useful and unique in lvgl's memory managers.

xiaoxiang781216, 2021-03-03:
tlsf provide tlsf_walk_pool api can achieve all mentioned statistics:
https://github.com/mattconte/tlsf/blob/master/tlsf.c#L904
and tlsf_check for debuging purpose.

embeddedt, 2021-03-03:
I think having a proper [pooling allocator](https://en.wikipedia.org/wiki/Memory_pool) for things like class data might be important going forward, as I have noticed from the logs that there are a lot of realloc operations happening when an object gets created. I haven't compared with v7 yet so maybe I'm worrying about nothing, but I suspect there were less reallocations needed with `lv_obj_allocate_ext_attr`.

With `lv_example_scroll_3` I had allocated a 64KB heap and approached 90% fragmentation when it ran out of memory.

xiaoxiang781216, 2021-03-03:
Yes, if the same class/struct allocate a lot of time, it's better to manage them by a dedicated pool. But, a good general heap manager is also important since the pool get the memory from heap too.

puzrin, 2021-03-03:
> as I have noticed from the logs that there are a lot of realloc operations happening when an object gets created. I haven't compared with v7 yet so maybe I'm worrying about nothing, but I suspect there were less reallocations needed with `lv_obj_allocate_ext_attr`.

That's strange, new architecture should make less allocations (may be new features?). The only thing to improve - switch from lists to auto-scaled arrays for assigned event listeners. But it's better to do this after allocator upgrade.

embeddedt, 2021-03-03:
> new architecture should make less allocations (may be new features?).

I believe class inheritance requires a reallocation each time you move down from a parent to a child but I could be wrong.

kisvegabor, 2021-03-03:
> tlsf provide tlsf_walk_pool api can achieve all mentioned statistics:
https://github.com/mattconte/tlsf/blob/master/tlsf.c#L904
and tlsf_check for debuging purpose.

Cool! It allows us to mimic `lv_mem_monitor` and `lv_mem_test`. @xiaoxiang781216 it seems you are already familiar with tlsf. Would you interested in adding behind the `lv_mem` API? If so we can add it in v8 too.

> I believe class inheritance requires a reallocation each time you move down from a parent to a child but I could be wrong.
It was improved in v8. In v7 `ext` was reallocated every time the creation went one level deeper. In v8 the instance size is set in the class so it's allocated only once. See [here](https://github.com/lvgl/lvgl/blob/22fcb8009b6d6ee4dfec05e688837b1e4287bac8/src/lv_widgets/lv_label.c#L60).

Besides in v7 all styles were removed and the theme was applied on every level. In v8 it's done [only once](https://github.com/lvgl/lvgl/blob/22fcb8009b6d6ee4dfec05e688837b1e4287bac8/src/lv_core/lv_obj_class.c#L66).

So if there were no problems with the number of reallocations in v7, it should be fine in v8 too. 
 

> With lv_example_scroll_3 I had allocated a 64KB heap and approached 90% fragmentation when it ran out of memory.

Note that `frag =  1 - (biggest_free_block / total_free_size) x 100 %`, so ot increases as the memory usage increases. E.g. if you have 1kB biggest free block, it's 1 % fragmentation with 100 kB total free RAM, but 50% with 2kB total free RAM. However in both cases the largest memory you could allocate was 1 kB.

embeddedt, 2021-03-03:
Okay; I'm probably just paranoid because I have the new mem_monitor enabled and thus I see this statistic on every application.

kisvegabor, 2021-03-03:
Ignorance is bliss... :laughing:

puzrin, 2021-03-03:
Will `tlsf` be good enouth with small objects, `Array[i32]`, length = 1..4. Or it will require additional "nested" pooling allocator?

Described case will happen after migrating from lists to auto-scaled arrays.

xiaoxiang781216, 2021-03-03:
tlsf has four bytes overhead for each allocation(true for any general one), so if we allocate many objects with the same size, the pool is more efficient than the heap since it could be implemented without any overhead. So I propose that:

1. Implement an efficient(fast speed, low overhead and less fragmentation) general heap manager
2. Implement a pool allocator(allocate a big block from heap and manage the small same size block internally)
3. Switch the allocation hotspot from heap to pool

xiaoxiang781216, 2021-03-03:
> Cool! It allows us to mimic `lv_mem_monitor` and `lv_mem_test`. @xiaoxiang781216 it seems you are already familiar with tlsf. Would you interested in adding behind the `lv_mem` API? If so we can add [...]

Yes, we just ported tlsf to https://github.com/apache/incubator-nuttx and replace the default heap manager. Could you open an issue describe what you want for the new heap manager? So I will find some free time to implement it.

puzrin, 2021-03-03:
> tlsf has four bytes overhead for each allocation(true for any general one), so if we allocate many objects with the same size, the pool is more efficient than the heap since it could be implemented without any overhead. So I propose that:

I think, 4 bytes overhead not worth to worry about. If tslf is fast enouth and effectively reuse blocks of the same size, i'd prefer it as generic allocator (with more simple interface).

At first glance, we need allocator, effective with:

- 8K-64K heap size (lvgl should run on stm32f072CBxx with 16K ram & 128K flash)
- 4-100 byte objects (small auto-scaled arrays + widget instances).

If you think, tslf will be ok - i think we should use it and forget about problem.

puzrin, 2021-03-03:
> Not good for so small object(<12B)

I think, that will never become a problem. If we really need to save extra 4 bytes for small arrays, we will find a way to do so. Right now potential problem is not with memory, but with perfomance.

PS. Min "array[1] of i32" size is 8 bytes, because 4 bytes required for meta (length) + align.

excitedbox, 2021-03-03:
Usually memory allocation is not the cause of major slow downs even when using standard malloc and alloc. The problems usually come from the way data is stored in memory. I think it would make more sense to look into how lv stores data instead of how the underlying memory assignment is done. The biggest gains will always be made in data structure.

ie. Using a linked list for iterators will be slow no matter how it is allocated because of massive rewrites when an object gets removed or having to iterate an entire list (multiple times) to find an items. 

This is actually something I found on the wiki for the C++ implementation for iterators earlier today. If an object is removed the whole iterator becomes invalid because they must have used a linked list instead of doubly linked so there is no way to remove an Item without loosing the parent/child connection without rewriting the whole list. I couldn't believe they haven't fixed that yet.

So I would say that using 20 bytes in overhead is not good compared to the current implementation. The solution I had suggested for the style system only had 1 byte of overhead and that was voted too much even though that would have actually brought real improvements in speed.

kisvegabor, 2021-03-03:
Thank you @xiaoxiang781216. I've opened #2123. Let's continue there.

kisvegabor, 2021-03-03:
@excitedbox 
I've also read that usually you shouldn't optimize malloc itself but optimizing for better data structures and try to avoid malloc instead. 

However, malloc is still a critical part of LVGL. The current memory managers are written by me without doing much research. So IMO replacing it with something professional is a clear win regardless of the other data structures.

closed 2021-03-03 by kisvegabor

excitedbox, 2021-03-07:
Yes in that case it would probably be helpful. When I looked over the TLFS docs it looked quite flexible and since Xiaoxiang has experience with it maybe that will make it easier to add future data structure improvements as well.

pscs, 2021-06-09:
I am getting link errors when compiling LVGL v8 for an esp32. It appears that the esp32 heap uses the same TLFS functions, so the linker is complaining about duplicate definitions, eg:

c:/program files (x86)/arduino/hardware/expressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Program Files (x86)\Arduino\hardware\expressif\esp32/tools/sdk/esp32/lib\libheap.a(heap_tlsf.c.obj): in function `tlsf_check':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/../esp-idf/components/heap/heap_tlsf.c:516: multiple definition of `tlsf_check'; C:\Users\paul\AppData\Local\Temp\arduino_build_556623\libraries\lvgl\misc\lv_tlsf.c.o:C:\Users\paul\OneDrive\Documents\Arduino\libraries\lvgl\src\misc/lv_tlsf.c:865: first defined here

Setting LV_MEM_CUSTOM to 1 allows it to compile, but I suspect that's suboptimal.

kisvegabor, 2021-06-10:
Thanks got the report. I've just fixed it by adding `lv_` prefix to public tlsf functions and types.