lvgl/lvgl #2118: Is the lv_chart can change dotted line color?

acer1204, 2021-03-03:
my version is 7.8.1.
I can use style to change line chart line color.
Is the lv_chart can change dotted line color?
[problem](https://www.dropbox.com/s/c4bc0bs5d6rxkpj/FF.png?dl=0)
i try create a limit line in line line chart by use dotted line...
it this dotted line can change color?

kisvegabor, 2021-03-03:
Do you mean changing the color only of the uppermost line?

turoksama, 2021-03-04:
Here I have a solution instead: Use cursor as limiting line you mentioned.
You can change the color by:
```
lv_chart_cursor_t *limit_line= lv_chart_add_cursor(chart, COLOR, LV_CHART_CURSOR_LEFT);
```
What do you think?

acer1204, 2021-03-08:
> Do you mean changing the color only of the uppermost line?

yes

kisvegabor, 2021-05-14:
In v8 you can modify the style of any division line. See this example: https://github.com/lvgl/lvgl/blob/master/examples/widgets/chart/lv_example_chart_2.c

I think it solves the issue so I close it now.

closed 2021-05-14 by kisvegabor