lvgl/lvgl #2119: 优化建议

Ghazigq, 2021-03-03:
1.可否优化lvgl的驱动
目前软件包lcd的buff必须大于一个屏幕大小，lvgl的lcd接口在刷新区域时，采用的是计算buff的偏移然后写入到buff中，可否将驱动更改，刷新区域从buff的起始地址写入，当然这也需要lcd框架驱动的配合
2.RGB格式驱动增加
在使用artpi的多媒体扩展板时，因屏幕为RGB888，需要在lvgl中做一次转换，将ARGB8888转换为RGB888
3.可否以最新的软件包做一个realse版本

xiaoxiang781216, 2021-03-03:
需要用英文提问题，要不老外看不懂。lcd的问题，lvgl已有解决方案，可以参考下面的文档：
https://docs.lvgl.io/dev/en/html/porting/display.html
https://docs.lvgl.io/dev/en/html/overview/display.html

Ghazigq, 2021-03-03:
> 需要用英文提问题，要不老外看不懂。lcd的buf问题，lvgl现有解决方案，可以参考下面的文档：
> [https](https://docs.lvgl.io/dev/en/html/porting/display.html) ://docs.lvgl.io/dev/zh/html/porting/display [。 html ](https://docs.lvgl.io/dev/en/html/porting/display.html)
> [https://docs.lvgl.io/dev/zh/html/overview/display.html](https://docs.lvgl.io/dev/en/html/overview/display.html)

搞错仓库了 抱歉

closed 2021-03-03 by Ghazigq