lvgl/lvgl #2123: Replace LVGL's memory allocator with TLSF

kisvegabor, 2021-03-03:
As discussed in #2116 LVGL's built-in memory manager could be improved in terms of performance and fragmentation. [tlsf](https://github.com/mattconte/tlsf) seems like a good candidate.

At first glance, the built-in current memory allocator could be replaced by tlsf without any API change. Even the memory monitor and memory test features can be kept.

This list summarizes requirements:
- The following top level functions are required: 
  - `lv_mem_alloc` (rename to `lv_malloc`?)
  - `lv_mem_free` (rename to `lv_free`?)
  - `lv_mem_realloc` (rename to `lv_realloc`?)
- These functions are also wrappers to `LV_MEM_CUSTOM_ALLOC/FREE/REALLOC` (set in `lv_conf.h` if `LV_MEM_CUSTOM 1`)
- The size of the memory pool should be configurable via `LV_MEM_SIZE`. The max size can be very large too (even hundreds of MB-s). If possible optimize tlsf for certain ranges via `#if`s. E.g. 0..8kB, 8kb..32kB, 32 kb...256kB, >1MB or so.  
- Manual defrag (now `lv_mem_defrag`) is not required as it should be handled in tlsf.
- Allow allocating 0 sized memory. Now a dedicated pointer is used in this case (called `zero_mem`).
- Reimplement `lv_mem_monitor` to extract at least the free and used size, largest free block and calculate the fragmentation
- Reimplement `lv_mem_test` to check memory integrity.
- Other functions in `lv_mem.c` shouldn't be affected (E.g. `lv_mem_buf_...` , `lv_memcpy`, `lv_memset`, etc)
- `tlsf.c/h` should be added directly as `lv_tlsf` (?) with correct attribution instead of submodules because submodules might be problematic for some package managers or for novice users. This way `lv_mem.c` can be clean containing only tlsf or `LV_MEM_CUSTOM` calls.

@xiaoxiang781216 already has experience with tlsf and mentioned that he interested in adding tlsf. Thank your for this! :slightly_smiling_face:

xiaoxiang781216, 2021-03-04:
> As discussed in #2116 LVGL's built-in memory manager could be improved in terms of performance and fragmentation. [tlsf](https://github.com/mattconte/tlsf) seems like a good candidate.
>
> At first [...]

Let's postone rename to limit the modification only inside misc/lv_mem.c

> * `tlsf.c/h` should be added directly as `lv_tlsf` (?)

I have two candidate:
```
src/extra/tlfs/
src/lv_tlfs/
```
Which one do you like?

> with correct attribution instead of submodules because submodules might be problematic for some package managers or for novice users. This way `lv_mem.c` can be clean containing only tlsf or `LV_MEM_CUSTOM` calls.
> 

BTW, should we format tlsf.c/h to LVGL coding style?

puzrin, 2021-03-04:
may be `src/vendor/tlfs`?

> BTW, should we format tlsf.c/h to LVGL coding style?

I would keep vendor sources as close to original as possible, to keep diff human-readable and simplify verification on upgrades. For the same reasons - 2 commits in history. First - add original files and nothing more, second - update to make things work.

kisvegabor, 2021-03-04:
> src/extra/tlfs/
src/lv_tlfs/
Which one do you like?

I'd put it simply into `src/lv_misc` simlarly to `lv_printf.c/h` which area also external files.

> BTW, should we format tlsf.c/h to LVGL coding style?

I think we should add as few modifications as possible to make updating easier later. 
E.g. an `#if LV_MEM_CUSTOM == 0` is required at the beginning of the file.

xiaoxiang781216, 2021-03-04:
Ok.

xiaoxiang781216, 2021-03-13:
Close, implemented by #2129.

closed 2021-03-13 by xiaoxiang781216