lvgl/lvgl #2134: Why save only the drv's pointer in lv_disp_t?

xiaoxiang781216, 2021-03-13:
### Introduce the problem

In this commit:
https://github.com/lvgl/lvgl/commit/e9cc1c2d4676b3863a85f35a79162ecedd280213#diff-9e08ffb2a7c9db7af1fcaea8e8a7e0c1be80e0f55292d56582959a4d6991e4b1L150
lv_disp_drv_t isn't embedded in lv_disp_t directly, but many other similar driver(e.g. _lv_indev_drv_t),  struct is still embedded in the corresponding device struct. My question is: what's benefit from this change?

### Examples and cases

### Suggested solution

kisvegabor, 2021-03-14:
Oh, I simply forgot to change indevs in the same way... I updated indev and lv_fs as well now.

The reasons behind the change are:
- It allow using extending `lv_disp_drv_t` with a class in C++ an use the class's pointer. See [here](https://forum.lvgl.io/t/change-lvgl-to-use-driver-references-instead-of-copy/3495/9). 
- It makes the driver you have initialized the same as the one passed to `flush_cb` and other fucntions. It causes some confusion now. 
- It's not a big deal, but also save some RAM.

xiaoxiang781216, 2021-03-16:
Understand, thanks for the detailed explanation.

closed 2021-03-16 by xiaoxiang781216