lvgl/lvgl #2143: Signals -> Events (consider unify messeging)

puzrin, 2021-03-16:
### Introduce the problem

LVGL uses term "signals" for pre-defined (AFAIK) message handlers. This cause significant cunfusion IMO - two different terms without "added value" (from user-s point of view).

### Suggested solution

Use events everywhere.

As far as i understand, current separation in coding-driven, not architecture-driven. That's not nice.

---

Initially, this MAY be caused by previous approaches, without classes, with different implementation of subscribers and so on. Nevermind...

Currently, with classes, event distribution may look like:

1. throw to default handler (via class)
2. throw to custom handlers (via instance)
3. bubble up (via instance parents scan), depending on message type OR `emit()` flag (TBD)

AFAIK, most of events have now "slow" sources (like "click"), and not subject to perf loss:

- uncertain event like "draw" now have direct class method, and not needs event emit.
- "expensive" `on_hover` (for mouse) is not needed (and cost of calculation is out of event machinery)
- "expensive" `on_scroll` is expensive in browser too and should be debounced anyway.

So, i think, showing "signals" to users as ordinary "events" should simplify understanding and not cause any problems. The only difference - user will not be able to "unsubscribe" defaults (not a big deal).

---

Probably, every event should be class instance, not sure. But it worth cleanup signals first, prior to dig more deep.

---

Priority:

- Signals are not "fatal", but very annoying for understanding architecture.
- Breaking change (suitable for major version only)
- Not deadly urgent for v8 (if cause release delay), but i'd like to drop those as "annoying". May be for v9, if problem for v8.

Let's discuss possibility of replacement.

kisvegabor, 2021-03-17:
There is one thing that signals do but events don't: signals are called on the ancestor types of the widget too.
E.g. when a slider's style changes `LV_SIGNAL_STYLE_CHG` is sent to
1. `lv_obj`
2. `lv_bar`
3. `lv_slider`

> throw to default handler (via class)

Calling "class's event handler" on every ancestor type seems doable and it might solve the problem.

kisvegabor, 2021-03-17:
If there is consensus with it I can add it to v8 quite quickly.

puzrin, 2021-03-17:
> There is one thing that signals do but events don't: signals are called on the ancestor types of the widget too.
> E.g. when a slider's style changes `LV_SIGNAL_STYLE_CHG` is sent to
> 
> 1. `lv_obj`
> 2. `lv_bar`
> 3. `lv_slider`

I'd like to clarify case first. Do you mean, inherited widgets should inherit/extend class methods, including event handler? I see no "signal specifics" here, just ordinary "class" inheritance.

kisvegabor, 2021-03-17:
So the `lv_event_send()` calls the last derived class's event handler and that event handler calls the base's event handler?

puzrin, 2021-03-17:
> So the `lv_event_send()` calls the last derived class's event handler and that event handler calls the base's event handler?

Short answer: I think yes.

You need to decide, why you extend widget class and what you expect to get. Calling or not calling `super.event_handler()` - details of widget implementation, hidden from everybody else. You just call class method, and don't care what happens inside.

kisvegabor, 2021-03-18:
Done!

Both signals and draw callbacks are merged into events.

puzrin, 2021-03-18:
IMO `.draw()` could stay class method, with direct call.

It exists in every widget, does not needs customization without class extend, and implicit call always has better look.

I mean, messaging is nice feature, but no need to put there everything.

kisvegabor, 2021-03-18:
`draw` works the very same way as events. E.g. in case of a `lv_label` first `lv_obj` draws a rectangle background (if it's not transparent) and the `lv_label` draw the text. 

In addition, moving drawing to events allow a nice "hooking" opportunity to user to add custom drawing. 

So a whole "chain" look like this:
1. draw grand parent class
2. draw parent class
3. draw class
4. call user drawings
5. bubble drawing to parent object (not sure it makes sense here)

puzrin, 2021-03-18:
You missed advantage of classes. In theory, initially all should be methods of class, because "explicit is better than implicit". But:

- many methods are optional.
- some methods needs to be customised per instance.

So, it's rational to do like this:

1. keep persistent not-extendable methods in class
2. move all garbage to message bus (more indirect, but more compact).

`.draw()` belongs to (1) IMO.

kisvegabor, 2021-03-18:
But drawing can be extendable! And it's a huge "power". I already made some projects with v8 and it was very simple to create custom charts and a slider that would be impossible or extremly complicated in another way.

puzrin, 2021-03-18:
Could you explain? May be i missed something. I thought, it's ok to override .draw() when you create new (inherited) class/widget (and there you can call parent's draw if required).

What are cases, when such approach is complicated or unuseable?

kisvegabor, 2021-03-18:
It's possible to overwrite, but it requires a new class instead of simply attaching a new event. It's tedious but not impossible. 

The tricky part is when you want to **extend** the drawing. Take a look at the charts of the new demo widgets. By default, LVGL doesn't support these stacked bar charts or displaying the points value on the press. However, the drawing can be hooked. Events are sent from every bar/point in which all the calculated data is available  (coordinates, styles, etc)  and only some modification or replacement by a  custom drawer is required. Without this feature basically a new chart drawer needed to be implemented (copy/paste + modifications) and maintained or the chart needs to be very complex to support a lot of custom, usually non-generic features.

Another thing is that there are [10 drawing related events](https://github.com/lvgl/lvgl/blob/171f7548d6bc6832c9b4134186d4462b959c5628/src/core/lv_obj.h#L64-L74). If a separate `draw_cb` would be use we needed either 10 methods or a `draw_mode` parameter which is basically an event type.

puzrin, 2021-03-18:
All JavaScript is based on concept of monkey-patching, and i know everything about disadvantages of such approach :). It's ok for quick-hacks in small apps only, and problematic for bigger lts projects with multiple people (because any junior can trash everything with "cool changes").

I would focus on simplifying class extend, to make trivial creating new class with overriden drawer.

> Another thing is that there are [10 drawing related events](https://github.com/lvgl/lvgl/blob/171f7548d6bc6832c9b4134186d4462b959c5628/src/core/lv_obj.h#L64-L74). If a separate `draw_cb` would be use we needed either 10 methods or a `draw_mode` parameter which is basically an event type.

I'm not ready to analyze deep. At first glance, that's over-engeneered (may be z-index could help). And, as i said before, current implementation of extended widgets is more complex than it could be, IMO.

---

Anyway, i think dropping signals to hell is great :+1: , and we can discuss drawer later when widgets refactoring happens. Let's close as solved.

closed 2021-03-18 by puzrin

kisvegabor, 2021-03-18:
> Anyway, i think dropping signals to hell is great +1 , and we can discuss drawer later when widgets refactoring happens. Let's close as solved.

The missing part in my mind was that we can call the events of ancestor widget types and then on user events too. It feels much simpler now.

puzrin, 2021-03-19:
> The missing part in my mind was that we can call the events of ancestor widget types and then the user events too. It feels much simpler now.

I don't understand. What's the problem with this? That's what "event emitter" function should do - call class method then instance list.

kisvegabor, 2021-03-19:
There is no problem, I just meant to say that before this issue I couldn't imagine how can we use events for signals too. 
So thank you for pointing it out! :slightly_smiling_face: