lvgl/lvgl #2144: [v8] Testing

kisvegabor, 2021-03-16:
Hi,

Almost every planned feature is implemented for v8. Now what we need the most is a lot of feedback about bugs and possible regressions. 

v8 is available in the `master` branch of lvgl.  
https://github.com/lvgl/lv_sim_eclipse_sdl/tree/dev is a ready-to-use project with v8. Note that the driver registration part slightly changed. The most important thing is that drivers `lv_indev_drv_t`, `lv_disp_drv_t` and `lv_fs_drv_t` needs to be `static`.

I suggest trying the demo widget first. It should work well on small (240x240) and very large (1280x720) screens.

The followings will be added soon:
- The `copy` parameter is not working yet in the `create` functions
- the mono theme
- updating the docs
- Updating the MicroPython binding

In v8 there is an `examples` directory where you find a lot of examples for widgets, layouts, etc.

If you have minor remarks, feel free to comment here. For bigger issues please open a new issue.
 
Here is a teaser about the new widgets demo on 800x480:

https://user-images.githubusercontent.com/7599318/111334210-a60a8380-8673-11eb-9092-72e1afb3549a.mp4

Thank you very much if you test v8 and tell your experience :slightly_smiling_face:

puzrin, 2021-03-16:
https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_obj_style.h#L68

Questions about styles:

- Is partition still required? AFAIK we discussed ID extension mechanics, and `part` should be outdated.
- Isn't `state` outdated? AFAIK it's part of style.
- I'm a bit confused about direct styles assign and "list of styles" assign. How should i create one list of styles and assign it to multiple objects?

puzrin, 2021-03-16:
I can't catch idea of coloring "methods" of styles (taken from examples):

- lv_style_set_text_color
- lv_style_set_line_color
- lv_style_set_content_color
- lv_style_set_arc_color

And so on. Looks like duplication of `lv_style_set_color`. At least, on web we have only `color` and `background-color` (let's temporary stay aside specials like border, shadow, ...)

amirgon, 2021-03-16:
> The most important thing is that drivers `lv_indev_drv_t`, `lv_disp_drv_t` and `lv_fs_drv_t` needs to be `static`.

Could you clarify this requirement?
In Micropython these structs cannot be created statically.

embeddedt, 2021-03-16:
These structs can't go out of scope; they need to be a static variable or be allocated on the heap.

amirgon, 2021-03-16:
> These structs can't go out of scope; they need to be a static variable or be allocated on the heap.

Oh Ok.
Would need to update the Micropython README - users would have to keep a global reference to these struct to prevent them from being garbage collected.

kisvegabor, 2021-03-16:
@puzrin 
> Is partition still required? AFAIK we discussed ID extension mechanics, and part should be outdated.

Yes, it's required. For example, a slider has "MAIN", "INDICATOR" and "KNOB" parts. Extension possibility means  a custom widget can have custom parts. E.g. "MAIN" and "FOO".

In HTML it's similar to pseudo-elements, more precisely to [::part](https://developer.mozilla.org/en-US/docs/Web/CSS/::part). The parts of the widgets are like a shadow tree.

> Isn't state outdated? AFAIK it's part of style.

You need to assign a style to a given state. E.g. you can have a `red_border` style that can be assigned to any states (PRESSED or FOCUSED).

> I'm a bit confused about direct styles assign and "list of styles" assign. How should i create one list of styles and assign it to multiple objects?

There is no concept for list of styles. I.e. you can not do this: 
```c
list1 = [style1, style2, style3]
lv_obj_add_style_list(obj, list1)
```

But you can:
```c
lv_obj_add_style(obj, LV_PART_MAIN, LV_STATE_DEFAULT, style1);
lv_obj_add_style(obj, LV_PART_MAIN, LV_STATE_DEFAULT, style2);
lv_obj_add_style(obj, LV_PART_MAIN, LV_STATE_DEFAULT, style3);
```

>  Looks like duplication of lv_style_set_color. At least, on web we have only color and background-color (let's temporary stay aside specials like border, shadow, ...)

In HTML there is no native line and arc. `text_color` is `color` in HTML. `content_color` is a simple replacement of `::after` to add a custom text from style to the widget.

 

@amirgon 
> The most important thing is that drivers lv_indev_drv_t, lv_disp_drv_t and lv_fs_drv_t needs to be static.

A detailed explanation is here: https://github.com/lvgl/lvgl/issues/2134#issuecomment-798900524

puzrin, 2021-03-16:
> Yes, it's required. For example, a slider has "MAIN", "INDICATOR" and "KNOB" parts. Extension possibility means a custom widget can have custom parts. E.g. "MAIN" and "FOO".

In web, styles definitions are separate of target assign. For example, i define styles for normal states, clicked, and so on.

At the time of use (assign) i know nothing about internal machinery, i just assign style.

Current API seems not allow such logic.

> In HTML it's similar to pseudo-elements, more precisely to [::part](https://developer.mozilla.org/en-US/docs/Web/CSS/::part). The parts of the widgets are like a shadow tree.

IMO pseudo-elements are not expected to be extended. Usual practice is vendor-prefixes (you can introduce any custom extra property).

- I think, this approach will work here too, if you consider widget as "blackbox", because number of extra properties (like "indocator-color") will not be big.
- If you introduce composite widgets (where sub-nodes are accessible) - then no extra probs required, because devault ones are enough.

---

Should i create separate issue to discuss? Seems, movement of this part goes to non-standard direction, and i don't think this is nesessary.

kisvegabor, 2021-03-16:
> In web, styles definitions are separate of target assign. For example, i define styles for normal states, clicked, and so on.
At the time of use (assign) i know nothing about internal machinery, i just assign style.
Current API seems not allow such logic.

True, however, I remember that it was a big question for me to where to store style parts and states. In styles or in widgets? Our common decision was that the widget is better. If you want I can search for the conversation but I'm 100% sure that you also agreed with it. 

> IMO pseudo-elements are not expected to be extended.

With HTML analogy: they can be extended if someone creates his own element (object) with its own shadow tree and uses `::part` to access shadow elements. E.g. a carousel with special controls.

> I think, this approach will work here too, if you consider widget as "blackbox", because number of extra properties (like "indocator-color") will not be big.

Parts are full featured. E.g. slider knob can have:
- bg color/opa/gradient/...
- border color/opa/image...
- shadow width/offset/...
- outline width/color/opa/...
- content text/font/..

Some properties can't be used, e.g. the arc-related ones. At least not directly. In v8 the user can hook the drawing of parts with events. It allows infinite flexibility. This way even the arc properties can be used if the user draws an arc on the knob based on the current value of the slider.

The gist is, it'd required a massive number of prefixed properties. (basic a duplication of the original ones for every part):

So IMO keeping parts directly addressable by styles is reasonable.

> If you introduce composite widgets (where sub-nodes are accessible) - then no extra probs required, because devault ones are enough.

In a lot of cases, it makes sense to keep the parts virtual. E.g. a table with 100 cells. Creating every cell and label would require 20-30kB RAM. Besides, it's just easier to simply draw what is required. Instead of creating a  slider knob that can move according to strict rules, it's easier to simply draw it according to the current value. Or think about every tick lines of a meter widget (see the network speed meter in the above video)

FYI, the widgets demo uses only 22.5 kB of RAM on STM32F429 which is pretty cool.

puzrin, 2021-03-16:
> True, however, I remember that it was a big question for me to where to store style parts and states. In styles or in widgets? Our common decision was that the widget is better. If you want I can search for the conversation but I'm 100% sure that you also agreed with it.

I'm 100% sure i could not agree with such approach (at least, with what i see in api now). This could be misunderstanding / language barrier and so on. I dit not dived too deep, because v8 changes are huge and it's impossible to keep everything in head.

Anyway, we need to move forward. If it's impossible to fit described changes into v8 time frame, then to v9. If there are some design problems, i'm ready to help. Or, if it's absolutely impossible to make desired changes, i'd like to understand reasons.

> With HTML analogy: they can be extended if someone creates his own element (object) with its own shadow tree and uses `::part` to access shadow elements. E.g. a carousel with special controls.

possible != common practice. I never seen such approach in my practice with different projects. If you mean creating forward/back contrtols via ::before / ::after - that's not creating new parts, that's using existing ones (but we skipped that due simplified DOM implementation).

> The gist is, it'd required a massive number of prefixed properties. (basic a duplication of the original ones for every part):
> 
> So IMO keeping parts directly addressable by styles is reasonable.

I do not agree.

- First, we speak about external widgets (not from core).
- Second, we could analyze each and count. I think, need of extra properties is over-estimated
- Third, IMO you make estimates with "old approach" to widgets (monolytic). While now you can use "composite" ones.

> In a lot of cases, it makes sense to keep the parts virtual. E.g. a table with 100 cells. Creating every cell and label would require 20-30kB RAM. Besides, it's just easier to simply draw what is [...]

Table with 100 cells is not relevant sample. You still can use custom drawer to "just draw what required" and use fake subnodes to carry styles only. For example "knob->tick_lines" will only store color of ticks (all ticks are still drawn at once). All we need is carefully review widgets and split those with proper granularity.

IMO, standard possibilities can be used with ease to reach desired effect, without need to introduce extra custom features.

I do not insist on immediate rework of "extended widgets" (need core features only). But i think, after some efforts, widgets can be redesigned more close to "web".

embeddedt, 2021-03-16:
@kisvegabor There is no doubt that this is one of the *best* demos you have made for LVGL. Fantastic work. :clap:

barbiani, 2021-03-17:
Plays well on the PC. Great stuff!

kisvegabor, 2021-03-17:
@embeddedt @barbiani 

Thanks!

@puzrin 
Seemingly it'd be a long discussion about conceptual things. Thanks for the remarks, but let's postpone these for the next iteration.

puzrin, 2021-03-17:
@kisvegabor No problem. Let's go next.

Flexbox. I see no examples how to use it via styles (i verify presentation management via styles [BEM], and skip other api alternatives).

> In HTML there is no native line and arc. `text_color` is `color` in HTML. `content_color` is a simple replacement of `::after` to add a custom text from style to the widget.

Consider "line" and "arc" like rectangle with line/arc inside. Ordinary "color" (as for any `<div>` content) is enough.

kisvegabor, 2021-03-17:
> Flexbox. I see no examples how to use it via styles

It's doesn't work via stlyes but descriptor. I saw no other framework than CSS where layouts are style property and I agree with. Style is what the library draws (color, radius, border, font, etc).

E.g in QT container with a specific layout can be created: https://doc.qt.io/qt-5/layout.html#laying-out-widgets-in-code

>  (i verify presentation management via styles [BEM], and skip other api alternatives).

Atomic CSS such as Tailwind is the other big party and with reason. BEM is also not perfect, just an option that works in some cases (I'm not a web developer just read a lot about these methodologies)

> Consider "line" and "arc" like rectangle with line/arc inside. Ordinary "color" (as for any <div> content) is enough. 

Hm, sounds interesting, thinking about it. 

> Sample with ::after - can not understand.

It's really like `::after` but you can add only text that can be aligned to the "main" content. I used it in cases like:
- **53** kg where "kg" is `content` aligned to the right of the number and shown with a smaller font.
- display the value of a slider, placed automatically above the knob
- a text above the text area, e.g. "First name"

puzrin, 2021-03-17:
> It's doesn't work via stlyes but descriptor. I saw no other framework than CSS where layouts are style property and I agree with. Style is what the library draws (color, radius, border, font, etc).

Seems we mean different things under "flexbox" term. I thought you agreed to implement strict css subset with minimal deviations (to guarantee predictability). Is it a problem to arrange flexbox via styles?

> It's really like `::after` but you can add only text that can be aligned to the "main" content. I used it in cases like:
>
>  * **53** kg where "kg" is `content` aligned to the right of the number and [...]

- first case is solved by something like `<span>` (CSS `display:inline-block`).
- second case is custom widget feature.
- third case seems to be [placeholder](https://www.w3schools.com/tags/att_input_placeholder.asp) or should be done via `<label>` (`<div>`) before `<input>` (not clear from description).

It looks like you take the same function name, and assign it with random feature for different widgets :).

Problem with all those custom methods - those are unique for lvgl and should be learned separate, without chance to reuse anywhere else. While with strict subset anyone who know html will not ask you questions "how it works?".

kisvegabor, 2021-03-17:
> I thought you agreed to implement strict css subset with minimal deviations

Flexbox itself is really a subset of CSS. In CSS even width/height are style properties. Which is reasonable to some extent but definitely not what our users expect. We can add an option later to allow setting layouts and size/position form style too but I imagine it as supplementary stuff for rare use cases. 

I'd love to hear other's opinions too. Do you think the layout should be style property?

> first case is solved by something like <span> (CSS display:inline-block).
> second case is custom widget feature.
> third case seems to be placeholder or should be done via `<label>` (`<div>`) before `<input>` (not clear from description).

You can do anything with `<span>` but `after` is directly tied to the base element and uses its styles from there which is quite convenient. Anyway, it's a flexible and very simple lightweifhted that comes in handy in some cases.

kisvegabor, 2021-03-17:
@embeddedt 
Will you have time in the near future to update the docs and example build scripts? As the examples are now in lvgl it might be easier to keep things in sync.

alfiezhao, 2021-03-18:
lv_textarea_set_cursor_hidden() found in lv_textarea.h not implemented in lv_textarea.c
where is function lv_obj_set_hidden() or how to hide a object?

alfiezhao, 2021-03-18:
lv_textarea_get_cursor_hidden() not implemented.
Seems there are some functions only existing in header file and not implemented in .c file

alfiezhao, 2021-03-18:
fontconverter not updated for v8

scottandrew, 2021-03-18:
Is the discussion about HTML a conceptual thing? I don't know if I would necessarily try to structure the UI library based on what HTML does and doesn't do. How the UI is structured, terminology and such should be what is best for the library not what some other technology is doing. HTML is not C and how to optimize and structure the code is something that should be library specific.

scottandrew, 2021-03-18:
So master seems to not be building.. 
```
/Users/scottandrew/Sources/Hello_LVGL/lvgl/src/font/lv_font_fmt_txt.c:115:12: error: implicit declaration of function 'lv_mem_get_size' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        if(lv_mem_get_size(LV_GC_ROOT(_lv_font_decompr_buf)) < buf_size) {
           ^
/Users/scottandrew/Sources/Hello_LVGL/lvgl/src/font/lv_font_fmt_txt.c:115:12: note: did you mean 'lv_area_get_size'?
/Users/scottandrew/Sources/Hello_LVGL/lvgl/src/font/../misc/../draw/../misc/lv_area.h:173:10: note: 'lv_area_get_size' declared here
uint32_t lv_area_get_size(const lv_area_t * area_p);
         ^
```

There was also an syntax error in the lv_draw_label.c

```
 lv_disp_buf_t * draw_buf = lv_disp_get_draw_buf(disp);
```

alfiezhao, 2021-03-18:
> lv_textarea_set_cursor_hidden() found in lv_textarea.h not implemented in lv_textarea.c
> where is function lv_obj_set_hidden() or how to hide a object?
Have worked it out:
    if (bVisible)
        lv_obj_clear_flag(widget, LV_OBJ_FLAG_HIDDEN);
    else
        lv_obj_add_flag(widget, LV_OBJ_FLAG_HIDDEN);

alfiezhao, 2021-03-18:
How to set text align in textarea?

xiaoxiang781216, 2021-03-18:
> 
> The followings will be added soon:
> 
> * The `copy` parameter is not working yet in the `create` functions

@kisvegabor why not drop copy support directly? Anyway, it will drop in v9 eventually.

> * the mono theme
> * updating the docs
> * updating the MicroPython binding

xiaoxiang781216, 2021-03-18:
> So master seems to not be building..
>
> ```
> /Users/scottandrew/Sources/Hello_LVGL/lvgl/src/font/lv_font_fmt_txt.c:115:12: error: implicit declaration of function 'lv_mem_get_size' is invalid in [...]

It seems be removed in patch:
https://github.com/lvgl/lvgl/commit/8bc6931f7f8283017e3584c7c353cb6dd59c1e58#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26
@kisvegabor should we bring back lv_mem_get_size? tlfs expose tlsf_block_size just for this.

> 
> There was also an syntax error in the lv_draw_label.c
> 
> ```
>  lv_disp_buf_t * draw_buf = lv_disp_get_draw_buf(disp);
> ```

@scottandrew please try this fix: https://github.com/lvgl/lvgl/pull/2148

kisvegabor, 2021-03-18:
@alfiezhao 
> lv_textarea_set_cursor_hidden() found in lv_textarea.h not implemented in lv_textarea.c
> lv_textarea_get_cursor_hidden() not implemented.

Fixed here: 171f7548

> where is function lv_obj_set_hidden() or how to hide a object?

You can use `lv_obj_add/clear_flag(obj, LV_OBJ_FLAG_HIDDEN)`

> fontconverter not updated for v8

I've just updated it.

kisvegabor, 2021-03-18:
@scottandrew 
> Is the discussion about HTML a conceptual thing? I don't know if I would necessarily try to structure the UI library based on what HTML does and doesn't do. How the UI is structured, terminology and [...]

I see it the same way. "Steal" the idea the makes sense in our case.

kisvegabor, 2021-03-18:
@xiaoxiang781216 
> why not drop copy support directly? Anyway, it will drop in v9 eventually. v8 already make many incompatible change, it isn't too bad to add another one.

I agree it'd better remove it now. My only concern is with this change almost all videos of https://lvgl.academy/ needs to be recorded again. @mair is the author of the courses so let's hook him in. I think our options are:
1. Leave the copy parameter: we postpone the problem to v9, but I'm not sure how many videos need to the recorded due to the already existing changes. (styles and layouts will be outdated for sure). As the default style of v8 looks different the v7 and update might be reasonable anyway.
2. Skip v8 with on lvgl.academy. (v7 is supported for a year from v8 release)
3. Record the videos again. @mair can say it, but probably it's much faster based on the v7 courses.

puzrin, 2021-03-18:
> I see it the same way. "Steal" the idea the makes sense in our case.

I already explained the problem of this approach. If you only scrape ideas, you will spend a huge time for you own architecture design errors. To save time effectively, you need not just scrape ideas, but scrape subset of features without deviations.

Mair, 2021-03-18:
@kisvegabor the right choice would be the one that suits the consumer of the library the best. having said that, backwards compatibility is awesome if you can afford it as it reduces the end users learning curve and makes it easier to upgrade which drives adoption.
if the widgets look different but are programmed in the same way, the videos will not need to change. If the API's are going to be different then we will need to re-shoot the video's. If the decision is made to radically change the API's we should evaluate the effort in the reshoot not only for v8 and v9 but potentially v10 and v11 as well. and run the numbers to see if its worth it. 
In summary, make the decision that best suits the libraries end users then, lets have a discussion about the training

scottandrew, 2021-03-18:
@kisvegabor Still seeing the same build issues as yesterday.. 

```
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/font/lv_font_fmt_txt.c:115:12: error: implicit declaration of function 'lv_mem_get_size' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
        if(lv_mem_get_size(LV_GC_ROOT(_lv_font_decompr_buf)) < buf_size) {
           ^
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/font/lv_font_fmt_txt.c:115:12: note: did you mean 'lv_area_get_size'?
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/font/../misc/../draw/../misc/lv_area.h:173:10: note: 'lv_area_get_size' declared here
uint32_t lv_area_get_size(const lv_area_t * area_p);
```

```
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:674:5: error: unknown type name 'lv_draw_buf_t'; did you mean 'lv_draw_res_t'?
    lv_draw_buf_t * draw_buf = lv_disp_get_draw_buf(disp);
    ^~~~~~~~~~~~~
    lv_draw_res_t
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/../core/lv_obj_draw.h:34:2: note: 'lv_draw_res_t' declared here
}lv_draw_res_t;
 ^
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:674:21: warning: incompatible pointer types initializing 'lv_draw_res_t *' with an expression of type
      'lv_disp_draw_buf_t *' [-Wincompatible-pointer-types]
    lv_draw_buf_t * draw_buf = lv_disp_get_draw_buf(disp);
                    ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:676:57: error: member reference base type 'lv_draw_res_t' is not a structure or union
    int32_t disp_buf_width = lv_area_get_width(&draw_buf->area);
                                                ~~~~~~~~^ ~~~~
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:677:45: error: member reference base type 'lv_draw_res_t' is not a structure or union
    lv_color_t * disp_buf_buf_tmp = draw_buf->buf_act;
                                    ~~~~~~~~^ ~~~~~~~
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:680:43: error: member reference base type 'lv_draw_res_t' is not a structure or union
    disp_buf_buf_tmp += ((pos_y - draw_buf->area.y1) * disp_buf_width) + pos_x - draw_buf->area.x1;
                                  ~~~~~~~~^ ~~~~
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/draw/lv_draw_label.c:680:90: error: member reference base type 'lv_draw_res_t' is not a structure or union
    disp_buf_buf_tmp += ((pos_y - draw_buf->area.y1) * disp_buf_width) + pos_x - draw_buf->area.x1;
                                                                                 ~~~~~~~~^ ~~~~
```

```
/Users/scottandrew/sources/Hello_LVGL/lvgl/src/core/lv_obj_style.h:6:9: warning: 'LV_OBJ_STYLE_H' is used as a header guard here, followed by #define of a different macro
      [-Wheader-guard]
#ifndef LV_OBJ_STYLE_H
```

alfiezhao, 2021-03-19:
> How to set text align in textarea?
In v7 I can use lv_textarea_set_text_align() but in v8 that function is gone and I tried lv_obj_set_style_text_align() but no good luck.
lv_obj_set_style_text_align(lv_textarea_get_label(ta), LV_PART_MAIN, LV_STATE_DEFAULT, LV_TEXT_ALIGN_CENTER);
Is it broken?

scottandrew, 2021-03-19:
Have it building with custom config. I do have a question about custom themes in V8 as I move the code over... My last config had:

```C++
#define LV_THEME_DEFAULT_INCLUDE            "src/theme/smart_rv_theme.h"      /*Include a header for the init. function*/
#define LV_THEME_DEFAULT_INIT               smart_rv_theme_init
#define LV_THEME_DEFAULT_COLOR_PRIMARY      lv_color_hex(0x000000)
#define LV_THEME_DEFAULT_COLOR_SECONDARY    lv_color_hex(0xffffff)
#define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT
#define LV_THEME_DEFAULT_FONT_SMALL &lv_font_montserrat_12
#define LV_THEME_DEFAULT_FONT_NORMAL &lv_font_montserrat_14
#define LV_THEME_DEFAULT_FONT_SUBTITLE &lv_font_montserrat_16
#define LV_THEME_DEFAULT_FONT_TITLE &lv_font_montserrat_18
```

kisvegabor, 2021-03-19:
@Mair 
Thanks. I think the architecture of LVGL getting more and more stable and believe there will be less severe API breaking changes in the upcoming major versions. 
For v9 I was planning mainly graphical effect updates (e.g. better gradient and image transformation support). These are local changes and it seems possible to create wrappers for backward compatibility. 

So if you agree, I suggest dropping the copy paramater now to collect major API breaks in v8, instead of spreading them across versions. 

@scandermore 
Thanks, I fixed them.

In v8 themes are attached to a given display.
It work like this:
```c
lv_disp_t * disp = lv_disp_drv_register(&disp_drv);
lv_theme_t * th = lv_theme_default_init(disp, LV_COLOR_PALETTE_LIGHT_BLUE, LV_COLOR_PALETTE_RED, &lv_font_montserrat_10, &lv_font_montserrat_14, &lv_font_montserrat_18);
lv_disp_set_theme(disp, th);
```

This way it's not required to follow a standard prototype with the theme. Your theme init function can receive as many fonts and colors as you wish.

@alfiezhao 
It works with `lv_obj_set_style_text_align(ta, LV_PART_MAIN, LV_STATE_DEFAULT, LV_TEXT_ALIGN_CENTER);`

scottandrew, 2021-03-19:
is lv_obj_align_mid going away?

kisvegabor, 2021-03-20:
Yes. I found that it's not versatile enough. E.g. what if you want to "mid_align" in X but "normal align" in y. 
You can use  `lv_obj_get_width(obj) / 2` in the offset paramer of the `lv_obj_align` to egt the same result.

scottandrew, 2021-03-20:
Got my project up and running but it seems setting colors on objects to override local style is broken. I have the following helpers in my object class.. 

```C++
void Object::setLineColor(uint8_t part, lv_state_t state, Color color) {
  lv_obj_set_style_line_color(object, part, state, color.lvColor());
}

void Object::setBorderColor(uint8_t part, lv_state_t state, Color color) {
  lv_obj_set_style_border_color(object, part, state, color.lvColor());
}

void Object::setLineWidth(uint8_t part, lv_state_t state, lv_coord_t width) {
  lv_obj_set_style_line_width(object, part, state, width);
}
```

In my arc wrapper I have the following: 

```C++
void TankGuage::setValue(int16_t value) {
  char percent[5] = "";
  char remaining[60] = "";
  sprintf(percent, "%d%%", value);

  int gallonsRemaining = floor(TANKGALLONS * ((100 - value) / 100.0));
  sprintf(remaining, "%d gal\nremaining", gallonsRemaining);

  percentageLabel->setText(percent);
  gallonsRemaingLevel->setText(remaining);

  // depending on the level we change colors.
  if (value < warningLevel) {
    setLineColor(LV_PART_MAIN, LV_STATE_DEFAULT, goodBackground);
    setLineColor(LV_PART_INDICATOR, LV_STATE_DEFAULT, goodIndicator);
  } else if (value < crticalLevel) {
    setLineColor(LV_PART_MAIN, LV_STATE_DEFAULT, warningBackground);
    setLineColor(LV_PART_INDICATOR, LV_STATE_DEFAULT, warningIndicator);
  } else {
    setLineColor(LV_PART_MAIN, LV_STATE_DEFAULT, criticalBackground);
    setLineColor(LV_PART_INDICATOR, LV_STATE_DEFAULT, criticalIndicator);
  }

    Arc::setValue(value);
}
```

But I am also noticing none of my alignment or line items are working either

```C++
  setLineWidth(LV_PART_MAIN, LV_STATE_DEFAULT, 8);
  setLineWidth(LV_PART_INDICATOR, LV_STATE_DEFAULT, 12);
  setPaddingAllSides(LV_PART_INDICATOR, LV_STATE_DEFAULT, -1);
  setBorderWidth(LV_PART_INDICATOR, LV_STATE_DEFAULT, 0);
  setSize(GUAGE_SIZE, GUAGE_SIZE);
  setPaddingAllSides(LV_PART_MAIN, LV_STATE_DEFAULT, 4);
```

Clues?

kisvegabor, 2021-03-21:
In v8 "arc" has its own style properties. See [here](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_style.h#L192-L197).

puzrin, 2021-03-21:
Which properties respond for width/height? For example, i wish create style with `{ width: 30%; height: 50px; }`

scottandrew, 2021-03-21:
> In v8 "arc" has its own style properties. See [here](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_style.h#L192-L197).

Thank you

kisvegabor, 2021-03-21:
> Which properties respond for width/height? For example, i wish create style with { width: 30%; height: 50px; }

Position and size are not style properties. I see that it'd be useful in some cases when you want to create multiple objects with the same size or position but there a lot of things to consider here, therefore I postponed this. Some issues I see:
- Which has higher precedence? `lv_obj_set_width` or `style.width`? 
- How to make it clear which one is being used? It should be straightforward for the user if the values come from style or object attribute.
- How will the many style queries affect the performance?

Anyway, I really see value in it, so could you open an issue to discuss it? If we find a good way to handle this it can go to v8.

embeddedt, 2021-03-21:
> Which has higher precedence? lv_obj_set_width or style.width?

If we are replicating CSS-like behavior then `lv_obj_set_width` should be removed.

> How will the many style queries affect the performance?

We should probably have a way of caching the value so that there is no performance impact.

kisvegabor, 2021-03-21:
> If we are replicating CSS-like behavior then lv_obj_set_width should be removed.

I think `lv_obj_set_width` should stay for 2 reasons:
- It very intuitive and that's what most of the user expects
- HTML also has `width` attribute on some elements

> We should probably have a way of caching the value so that there is no performance impact.

Normally, the object's size is cached in `obj->coords` but if there are a lot of objects (list with 200 buttons) and the position/size needs to be got for all the buttons and their children, layout recalculations might be slow.

puzrin, 2021-03-21:
> * Which has higher precedence? `lv_obj_set_width` or `style.width`?
> ...

IMO this is incorrect question: "how to add new feature into existing deviated state". Just drop deviations and do subset of css.

If you need legacy like `lv_obj_set_width()`, it can be done via local style. But i would drop such method as confusing, because lvgl already have methods for local styles.

> It should be straightforward for the user if the values come from style or object attribute.

Everything come from styles (external or local). There are NO other "object attribute". All user need is read CSS docs if any questions.

> Anyway, I really see value in it, so could you open an issue to discuss it? If we find a good way to handle this it can go to v8.

No problem. Could you clarify topic please? Only width/height or all CSS deviations?

embeddedt, 2021-03-21:
> It very intuitive and that's what most of the user expects

Why not have it be a wrapper around the object's local `width` style property in that case?

> HTML also has width attribute on some elements

As far as I know the `width` attribute is deprecated and is very rarely used in modern HTML5 apps as the `width` style property does the same thing. I personally have never needed to use it.

puzrin, 2021-03-21:
> I think `lv_obj_set_width` should stay for 2 reasons:
> 
>  * It very intuitive and that's what most of the user expects

IMO `lv_style_local_width()` (i don't remember exact name) is good enougth. Also note, your "intuitive" `lv_obj_set_width` can cause deeply wrong  understanding how things work.

>  * HTML also has `width` attribute on some elements

Nobody uses this feature.

kisvegabor, 2021-03-21:
> Could you clarify topic please? Only width/height or all CSS deviations?

Can I ask for both in 2 issues? :slightly_smiling_face: 

> It very intuitive and that's what most of the user expects
>> Why not have it be a wrapper around the object's local width style property in that case?

>> IMO lv_style_local_width() (i don't remember exact name) is good enougth. Also note, your "intuitive" lv_obj_set_width can cause deeply wrong understanding how things work.

It's very different how all the UI framework (at least I've what seen) works. I'm pretty sure the vast majority of the users expect to set width/height/x/y on an object basis and they think styles are about ~colors. 

Providing a wrapper could work at first look, but I think the work `style` can not be in the function name.  

> Also note, your "intuitive" lv_obj_set_width can cause deeply wrong understanding how things work.

How and why? 

> HTML also has width attribute on some elements
>> As far as I know the width attribute is deprecated and is very rarely used in modern HTML5 apps as the width style property does the same thing. I personally have never needed to use it.

>> Nobody uses this feature.

Okay, let's drop this point. :slightly_smiling_face:

embeddedt, 2021-03-21:
For anyone having issues viewing the demo video, you can try it live here: https://lvgl.github.io/lv_sim_emscripten/

kisvegabor, 2021-03-21:
@embeddedt Awesome!

egonbeermat, 2021-03-21:
What is the functional equivalent of lv_obj_get_ext_attr() in v8? Additionally, is there any more information to peruse regarding migration from v7? The provided examples seem too basic to drill into for more complex coding scenarios. Thanks in advance!

scottandrew, 2021-03-22:
void lv_arc_set_adjustable(lv_obj_t * arc, bool adjustable) doesn't seem to exist in the C file. Is there any chance that this is getting implemented? If not I can try to issue a pull request. I just want the arcs for display.

kisvegabor, 2021-03-24:
> What is the functional equivalent of lv_obj_get_ext_attr() in v8? Additionally, is there any more information to peruse regarding migration from v7? The provided examples seem too basic to drill into for more complex coding scenarios. Thanks in advance!

We are updating the docs so the description will come.
If you need to add custom data to a widget you can do 3 things:
1. Use the `user_data`
2. Create a new `class`. See for [lv_led](https://github.com/lvgl/lvgl/blob/master/src/extra/widgets/led/lv_led.c) as an example. 
3. If you need only a few flags you  can use `LV_OBJ_FLAG_USER_1/2/3/4` with `lv_obj_add/clear_flag()`

kisvegabor, 2021-03-24:
> void lv_arc_set_adjustable(lv_obj_t * arc, bool adjustable) doesn't seem to exist in the C file. Is there any chance that this is getting implemented? If not I can try to issue a pull request. I just want the arcs for display.

I've removed it intentionally. If you need an arc only to display data you should use `lv_meter` which is way more versatile. If you really want to use `lv_arc` you can remove the knob's styles to make is transparent: `lv_obj_remove_style(arc, LV_PART_KNOB, LV_STATE_ANY, NULL)`

scottandrew, 2021-03-24:
lv_obj_set_auto_realign appears to be gone. So is it suggested to use one fo the layouts if you need to keep text centered in a parent as it changes? My text is not center signed. I have set the width to match the arc.

[image]

embeddedt, 2021-03-24:
@kisvegabor 

> Will you have time in the near future to update the docs and example build scripts? As the examples are now in lvgl it might be easier to keep things in sync.

I'm a little bit stretched on time at the moment but I will try to get around to it as soon as I can.

kisvegabor, 2021-03-25:
> I'm a little bit stretched on time at the moment but I will try to get around to it as soon as I can.

No problem, thanks for the update. 
The `index.rst` files are not updated yet. Do you have any special requirements with them? If so I can update them accordingly.

kisvegabor, 2021-03-25:
Some important update on v8:
- The `copy` parameter is removed from the `create` functions. 
- `lv_obj_align()` now saves the "align" and updates the object's alignment if it's size changes
- `lv_obj_align()` has not `base` parameter anymrore, you can use `lv_obj_align_to` to align to an other obejct
- `lv_obj_set_x/y` works relative to the set "align". E.g. if `LV_ALIGN_CENTER` is used, `x=1` shift the object from the center
- `x`, `y`, `width`, `height`, are now style properties in the background. You can keep using `lv_obj_set_x/y/width/height` as usual but you can also set the coordinates in a style.
- layout properties are also moved to style properties but their are simple wrappers to set the layout on an object directly.  See the flex and grid examples.   
- `content` style properties are removed
- One update still comes soon: the default sizes should come from styles. Now they are set with `lv_obj_set_width/height` which sets the size in local styles therefore the size can't be overwritten by "normal" styles.

kisvegabor, 2021-03-25:
> lv_obj_set_auto_realign appears to be gone. So is it suggested to use one fo the layouts if you need to keep text centered in a parent as it changes? My text is not center signed. I have set the width to match the arc.

See my comment above. With the latest version you can `lv_obj_align(label, LV_ALIGN_CENTER, 0, 0)` and it will remain aligned. Or shorter: `lv_obj_center(label)`

alfiezhao, 2021-03-25:
@kisvegabor 
lv_obj_set_style_text_align(ta, LV_PART_MAIN, LV_STATE_DEFAULT, LV_TEXT_ALIGN_CENTER);
has no effect on text alignment of textarea widget.
But I just tried lv_obj_align(lv_textarea_get_label(ta), LV_ALIGN_CENTER, 0, 0) and now the text alignment is working.

[334 more replies]

closed 2021-06-01 by kisvegabor