lvgl/lvgl #2145: Keyboard/button matrix - expensive update on key press

jupeos, 2021-03-17:
### This issue...
...relates specifically to the button matrix but was found using the keyboard. When a key is pressed the whole of the `btnm` is invalidated in addition to/instead of, just the region of the key. This results in a very sluggish keyboard experience. This is due to the signal handler `lv_btnmatrix_signal` making a call to the "base class" `ancestor_signal(btnm, sign, param)` where it passes in the `btnm`. If you drill down this results in a `lv_obj_invalidate` operating on the whole button matrix, i.e. half the screen for a keyboard.
Note: This issue relates to this [inquiry](https://forum.lvgl.io/t/keyboard-input-slow/4276?u=jupeos).

### To Reproduce
Simulator or real v7.11.0 and earlier.
Take any example that has a button matrix and place a bp on ln:852 of `lv_btnmatrix.c`. Press a button and drill down until you get to `lv_obj_invalidate_area` and check the region size.

### Expected behavior
Only the region of the key that is pressed should be invalidated.

As a hack to fix this for a demo that I have, I added ` if(sign != LV_SIGNAL_PRESSED) {` around the call to `ancestor_signal` which did work but is probably not the correct solution.

kisvegabor, 2021-03-18:
I've checked that and it really happens. The reason is when the state of the button matrix goes to PRESSED LVGL checks if there are any changes in the new state. It sees that there is a  change because the button looks differently. However, at this low-level LVGL doesn't know anything about the internal of the widget, only knows that something is changed. In lack of more accurate information, it redraws teh whole widget. 

Unfortunately, it hasn't changed in v8 either. :slightly_frowning_face: It'd require a mechanism to let the widget handle invalidation on state change (instead of a generic handler) but I can't imagine an efficient way to handle this now.

---

Why does it take so much time to redraw the keyboard? If it's noticeable then the screen changes and other transitions should be slow as well.

embeddedt, 2021-03-19:
> It'd require a mechanism to let the widget handle invalidation on state change (instead of a generic handler) but I can't imagine an efficient way to handle this now.

Is there a reason why an event handler would be too slow? As far as I know, objects only change state when they are pressed and released, not on each frame.

kisvegabor, 2021-03-19:
Sorry, "effective" wasn't a good word here. I meant a simple and easy to maintain and approach. 

I've played with it in v8 and finally found a solution. Some widgets need to be slightly updated though but it's working. 

The refreshed areas are shown with random colors:
[image]

embeddedt, 2021-03-19:
@kisvegabor Is this refresh visualization a feature of v8? This would be extremely useful for render profiling.

kisvegabor, 2021-03-19:
It's there since v5 or v6 :slightly_smiling_face: 

It can be enabled in `lv_refr.c` by `#define MASK_AREA_DEBUG 1`. Do you think it's worth adding it lv_conf.h?

embeddedt, 2021-03-19:
Oh, wow! I wish I knew about this earlier! I think this would be useful to have in `lv_conf.h`.

jupeos, 2021-03-22:
@kisvegabor 
> I've played with it in v8 and finally found a solution. Some widgets need to be slightly updated though but it's working.

Any solution for V7?

kisvegabor, 2021-03-22:
This part of v7 is quite complicated. Therefore adding this would be quite complicated and might introduce a lot of bugs. Besides, it's the first time this problem came up so seemingly this problem doesn't affect a lot of users.  

Would it be difficult for you to change to v8?

jupeos, 2021-03-22:
I'm working on a product going to market shortly and want to keep a stable code base. V8 seems too risky at this time.
As a client of your (excellent!) library I would like to see long term support (1 year I believe for LVGL) for older releases.
As I'm sure you have experienced it can be very frustrating when a vendor of a product recommends an upgrade to a newer release as the solution to a bug.

However, maybe a compromise here would be to review the hack that I implemented and see what the negative effects might be. If they are minimal (or can be managed) then this could suffice.

This is the code that I used to get things operational in `lv_btnmatrix_signal`

```
    if(sign == LV_SIGNAL_PRESSED) {
        res = LV_RES_OK;
    }else{
        /* Include the ancient signal function */
        res = ancestor_signal(btnm, sign, param);
    }
```

kisvegabor, 2021-03-22:
> I'm working on a product going to market shortly and want to keep a stable code base. V8 seems too risky at this time.

In this case I also suggest sticking with v7.

> As a client of your (excellent!) library I would like to see long term support (1 year I believe for LVGL) for older releases.

We really have 1 year release for old releases. See [here](https://docs.lvgl.io/latest/en/html/intro/index.html#version-support).
And v8 isn't out yet so v7 is the current active version. 
The problem is that its not really a bug but a conceptual thing. v7 just invalidates the whole widget on state change. When I added it to v8 some widgets became broken. I could fix them, but we can not allow that the custom widgets of people who use v7 suddenly also became broken.

> As I'm sure you have experienced it can be very frustrating when a vendor of a product recommends an upgrade to a newer release as the solution to a bug.

I perfectly understand. It's only an option if you just started the project.

> However, maybe a compromise here would be to review the hack that I implemented and see what the negative effects might be. If they are minimal (or can be managed) then this could suffice.

Sure!

Your hack should do the job. There is only one thing: this way `LV_SIGNAL_PRESSED` is not executed, and therefore the first press is not recognized. LVGL keeps sending `LV_SIGNAL_PRESSING` so the button press will be detected 30 ms later. 
I think you shouldn't worry about it, I just mentioned it because it might be better to know. :)

jupeos, 2021-03-22:
Thanks for the prompt response. I'll work with what I have right now and look to upgrade to V8 down the road.
Keep up the good work!

kisvegabor, 2021-03-24:
Thanks!
Good luck with your project! :slightly_smiling_face:

closed 2021-03-24 by kisvegabor

bjfhskj, 2024-05-27:
I have encountered the same problem recently with v8.3.4,my keyboard was taking ~500ms to respond to a key press,how can I accelerate it?