lvgl/lvgl #2147: 90% CPU usage with simple chart

BlueIan, 2021-03-17:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce
This is the code the runs in the FreeRTOS task modified to run in the simulator. The only difference is the hardware initialization. 

```c

static void task_chart_update(lv_task_t * param);

uint16_t dataPointValue;
lv_chart_series_t * ser1;
lv_obj_t * chart_simple;

int main(int argc, char** argv)
{
	

	/*Initialize LittlevGL*/
	lv_init();

	/*Initialize the HAL for LittlevGL*/
	hal_init();

	lv_obj_t * screen_1 = lv_obj_create(NULL, NULL);

	lv_scr_load(screen_1);

	// Create a chart to update
	// ----------------------------------------
	chart_simple = lv_chart_create(lv_scr_act(), NULL);
	lv_obj_set_size(chart_simple, 200, 150);
	lv_obj_align(chart_simple, NULL, LV_ALIGN_CENTER, 0, 0);
	lv_chart_set_type(chart_simple, LV_CHART_TYPE_LINE);

	// Add data series
	// ----------------------------------------
	ser1 = lv_chart_add_series(chart_simple, LV_COLOR_RED);
	
	// Add task to update chart
	// ----------------------------------------
	lv_task_create(task_chart_update, 100, LV_TASK_PRIO_LOW, NULL);

	while (1)
	{
		
		lv_task_handler();
		//vTaskDelay(10);
		Sleep(10);       
	}

	return 0;
}
	

static void task_chart_update(lv_task_t * param)
{
	(void)param;    /*Unused*/

	// Generate new data point.
	// ------------------------------
	dataPointValue += (rand() % 10);

	if (dataPointValue > 100)
		dataPointValue = 0;

	// Plot point
	// ----------------------------------------
	lv_chart_set_next(chart_simple, ser1, dataPointValue);
}
```
-->

### Expected behavior

### Screenshots or video

kisvegabor, 2021-03-18:
I've tried the same chart on STM32F429 (180MHz) and added new points in every 100 ms. The CPU usage was 6-7%.

It's great that you have tried with empty `flush_cb`, that would have been my first question. :slightly_smiling_face: 

Some questions:
- Where are the draw buffers of LVGL located? If in external RAM, the low memory bandwidth can be the reason for the low performance.
- Have you enabled compiler optimization? 
- Just to be sure, can you try without FreeRTOS?

BlueIan, 2021-03-19:
Hi kisvegabor, thanks for your reply.  The 6-7% is in the range I would have expected. I changed the optimization level from O0 to O1 without any noticeable change.  (While I continued checking with higher optimization, I have introduced a problem with the I2C locking up that won't go away when returning to O0.  I am confident this unrelated but will needs to be resolved. Probably a conductive cookie crumb or similar :)  )
Anyway, all the ram is internal and runs at the CPU clock.   I will test without FreeRTOS next.  

Assuming the no improvement with further optimization and running without RTOS (will confirm shortly), where should I start looking?  I usually trace this kind of problem with a scope by toggling GPIO pins at start/end of functions.   Are there any functions that are candidates for tracing like this?

BlueIan, 2021-03-19:
Hi kisvegabor, thanks for your reply.  The 6-7% is what I would have expected. I changed the optimization level from O0 to O1 without any noticeable change.  (While I continued checking with higher optimization, I have introduced a problem with the I2C locking up that won't go away when returning to O0.  I am confident this unrelated but will needs to be resolved. Probably a conductive cookie crumb or similar :)  )
Anyway, all the ram is internal and runs at the CPU clock.   I will test without FreeRTOS next.  

Assuming the no improvement with further optimization and running without RTOS (will confirm shortly), where should I start looking?  I usually trace this kind of problem with a scope by toggling GPIO pins at start/end of functions.   Are there any functions that are candidates for tracing like this?

Thanks,
Ian

kisvegabor, 2021-03-19:
You should trace `_lv_disp_refr_task()` in `lv_refr.c`.

It's probably no the issue, but it might happen that I2C read functions can't work wait for a timeout causing high CPU load.
So I suggest tracing `_lv_indev_read_task()` too in `lv_indev.c`.

Also be sure `LV_USE_LOG` and `LV_USE_DEBUG` is set to 0 in `lv_conf.h`.

BTW, what is the size of the display buffers of LVGL?

BlueIan, 2021-03-19:
The display buffer is 480 * 10,  defined in LVGL FreeRTOS task as 
static lv_color_t LCD_buf_1[LV_HOR_RES_MAX * 10];

Hope to have more info soon

embeddedt, 2021-03-19:
If you have more RAM I would suggest increasing the display buffer size as much as possible. This can significantly reduce rendering overhead.

BlueIan, 2021-03-23:
Thank you both for your suggestions.

I have some results.  
Biggest improvement came from increasing display buffer from 10 lines to 100 lines (of 320).
Also increased the SPI clock rate from 12MHz to 32 MHz.
Currently I am getting 33FPS @30%CPU.  This still seams high.

I am attaching two scope images that show three traces.  In order they are:
1) (yellow) This is a 1ms pulse every time lv_task_handler(); is called. There is a 10ms FreeRTOS delay between calls.
2) (blue) This is signal is turned on/off at the enrty/exit of _lv_disp_refr_task()  
3) (pink) This is turned on at the start of the DMA flush operation and turned off when  lv_disp_flush_ready() is called.

[image]

In the above image, the display shows 1 second of operation.  The top trace is showing lv_task_handler() bing called every 10ms.  The second trace shows _lv_disp_refr_task()  executing 17 times in one second.  Each call usually takes 25ms.

[image]

This second image clearly shows the DMA flush operation starting near the end of _lv_disp_refr_task()  execution and ending 35-40ms later. That matches the expected transfer time for 420x100x16 bits.  

I don't think the FPS display is accurate in this case.   

So the flush operation is not extending the operation of _lv_disp_refr_task() in any way that I can see.

I am hoping to reduce the lvgl processing time some more - closer to the 6% you quote for the comparable 180MHz ST part.

What should I look at next?

BlueIan, 2021-03-23:
Any more suggestion on where to look?

BlueIan, 2021-03-26:
Bump?
Just need a bit of direction here...

kisvegabor, 2021-03-26:
First I'd like to clarify what is measured by my LVGL As CPU load. It measures the time spent in `lv_task_handler` from a 500 ms window. So the flush operation not measures as it happens in the background.

From the tace, it's clear that the flush doesn't block LVGL's rendering. Furthermore, it also can be seen the chart is refreshed in on chunk (else you would see "flush"-es during disp_refr) so probably increasing the buffer size wouldn't help. 

So as you also concluded the problem is that the rendering is too slow. You mentioned that you have tried O1 optimization with no difference. Can you try O3? 
Anyway, optimization should approx. double the performance. Be sure to clean your project after setting the optimization level. I've noticed that MCUXpresse doesn't rebuild the project automatically.

embeddedt, 2021-04-20:
I've tried the example provided by @BlueIan on my custom STM32F7 project (uses DMA2D for flushing) and found this:

For v7 with `-Og`:
* DMA2D used for rendering and flushing: 15% CPU
* Software rendering: 15% CPU

The difference is probably small due to the size of the chart - DMA2D tends to offer improvement mainly when refreshing larger areas.

For v7 with `-Os`:
* Software rendering: 17% CPU

For v7 with `-O3`:
* Software rendering: 13-14% CPU

Interestingly changing the optimization made little difference and I was unable to get to 6% CPU in any of these cases on v7. I am running LVGL on an OS, but it is the only task running so I doubt that the OS is the problem.

Out of curiosity, I tried it on v8 with `-O3`:
* Software rendering: 8% CPU

@kisvegabor Which version did you use to measure?

kisvegabor, 2021-04-20:
@embeddedt I already don't remember :sweat_smile:

- Have you enabled CPU caching?
- HAve you set the chart size is 200x150?

embeddedt, 2021-04-20:
Yes; both caches are enabled, and the chart size is 200x150. It's strange because I think my board should give similar results to yours. I will try testing the baremetal demo soon.

kisvegabor, 2021-04-20:
I'll test it again tomorrow too.

embeddedt, 2021-04-20:
It seems that the overhead of my OS is greater than I thought: on bare metal the CPU usage is approximately 6-7%. This is with `-Os` optimization.

kisvegabor, 2021-04-20:
Oh, it's a big surprise to me as well. Now it's similar to my measurements, right?

embeddedt, 2021-04-20:
Yes; it seems that it matches your measurements now. My project currently has the working buffer in SDRAM so that may be part of my problem.

Still, I am not sure why @BlueIan is getting much worse results. A CPU speed of 150MHz vs mine at 200MHz is not enough of a difference to explain the 30% usage vs 6%. Maybe his RAM is slower as well? It would also be worth checking the optimization even though that didn't help me.

closed 2021-05-14 by kisvegabor