lvgl/lvgl #2150: lvgl label long mode(LV_LABEL_LONG_DOT) bug

FlyLu, 2021-03-20:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

[image]

How can I fix this isssue

### To Reproduce

```C
void lv_label_test(void)
{
    lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);

    char text[] = "l ongmodetest";

    lv_label_set_long_mode(label, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label, 100, 20);
    lv_label_set_text(label, text);
}
```

### Expected behavior

l ongmod...

### Screenshots or video

hnsqlisai, 2021-03-20:
The same problem

embeddedt, 2021-03-20:
Which version are you using: 7.11.0 or 8.0 prerelease?

Try setting the text before changing the long mode and size.

FlyLu, 2021-03-22:
> Which version are you using: 7.11.0 or 8.0 prerelease?
> 
> Try setting the text before changing the long mode and size.

Hi， version 7.7.2 (info "dev")

I tried what you said

[image]

kisvegabor, 2021-03-24:
It happens below the "DOT" works like this:
1. LVGL uses the current width of the label the "calculates" the lines the text with line wrapping 
2. Checks which letter is on the x2,y2 coordinate of the label
3. Trun 3 characters into dots from the found letter.

Illustration (`|` and `-` shows the size of the label):
```
--------------
| hello this |
| is a long  |  
--------------
| text
```

turns into:
```
--------------
| hello this |
| is a l...  | 
--------------
```
In your case:
```
--------
| l    |
--------
| modet|
| est  |
```
turns into:
```
--------
| l... |
--------
```

To handle this, it would require a tricky line wrapping algorithm: wrap normally, except in the last line where there should be no wrapping.

FlyLu, 2021-03-24:
@kisvegabor Thank you for your reply

I am a little confused, why the following string display is expected

```C
char text[] = "l ong mode test";

lvgl show:  l ong mode...
```

[image]

embeddedt, 2021-03-24:
You're setting the label width to 100px; I suspect this is not wide enough for that string to fit.

kisvegabor, 2021-04-14:
Unfortunately, I can't imagine a simple fix for it at this moment so I close this issue for now. :slightly_frowning_face:

closed 2021-04-14 by kisvegabor