lvgl/lvgl #2152: Implement height/width via styles

puzrin, 2021-03-21:
### Introduce the problem

Currently obj height/width are out of styles.

1. That's deviation from css
2. That prohibits some obvious use cases (assign `{ width: 30%; height: 50px}` to cells).

### Suggested solution

1. Implement height/width via styles
2. If legacy wrappers required - do those via local styles set call
3. (optional) Consider add `min-height` / `min-width` / `max-height` / `max-width`

kisvegabor, 2021-03-21:
I tend to be open to this. 
It also opens the doors for different sizes in different states. 

What about `left/right/top/bottom`?

puzrin, 2021-03-21:
> What about `left/right/top/bottom`? They can replace `lv_obj_align` to some extent.

I did not analyzed all API. But my answer is always the same - do exactly as CSS does, and drop all custom methods. If you think something undoable - ask me.

left/right/top/bottom are more part of "position: absolute". In other case margin-* would be fine. That's very simplified explanation. In other words, implementing of left/right/top/bottom is not critical at first step.

kisvegabor, 2021-03-21:
Actually, `margin` is not supported now but there is `row gap` and `column gap` (also from CSS).

`position: absolute` is very important because it's the "default" when there is no layout. That is the user can freely position objects or apply a layout.

kisvegabor, 2021-03-21:
I'll make some experiemts tomorrow with 
- width
- height
- left
- right
- top
- bottom

added to styles.

puzrin, 2021-03-21:
> Actually, `margin` is not supported now but there is `row gap` and `column gap` (also from CSS).

That's not replacement. Some part of margin and padding should be supported. And that's more important than left/right/top/bottom.

puzrin, 2021-03-21:
https://developer.mozilla.org/en-US/docs/Web/CSS/position - here are good samples for `position` (with left/right/top/bottom)

After thinking a bit... may be margin can be postponed, but flexbox should be inspected more deep (now it looks far from CSS).

kisvegabor, 2021-03-21:
I agree to postpone the margin. It might be useful in some cases but not mandatory.

Regarding flexbox, here what I've changed:
- no child-level alignment (align-self) for simplicity
- no flex-basis: it's used only if grow/shrink is also used. Assume it's zero, so if grow is used children simply fill the available space.
- no order: too complicated
- no shrink: too complicated, and assumes there is flex-basis to shrink from. 
- changed the name of justify-content, align-items, and align-content because I think no one knows what they without reading the cheatsheet.

So what it can:
- row, column direction with or without wrap and revers
- grow
- start an item in a new track (not in CSS but very useful)
- almost all the alignment options flexbox has: start, end, center, space-around, space-evenly, space-between

If we move position to styles we will have more flexibility but it's very important to keep flexbox simple even at the cost of deviation from CSS. I mean especially the names of alignment options.

puzrin, 2021-03-22:
> * no child-level alignment (align-self) for simplicity
> * no flex-basis: it's used only if grow/shrink is also used. Assume it's zero, so if grow is used children simply fill the available space.
> * [...]

I think, that's reasonable, if you can support "fill gap to 100%" feature:

```
btn btn ....gap.... btn
```

That's usually done via setting gap to 100% and enable shrinking (other cells have fixed width or auto-width by child content).

> 
> * changed the name of justify-content, align-items, and align-content because I think no one knows what they without reading the cheatsheet.

I agree, finding all names without cheatsheet is difficult. But keeping names allow refer to css docs & reuse power of flexbox tutorials for free. Where i can see current names? I'd like to understand, what looks difficult from your point of view and you wish to improve.

> start an item in a new track (not in CSS but very useful)

What is that?

kisvegabor, 2021-03-22:
> I think, that's reasonable, if you can support "fill gap to 100%" case:

Oh, we rarely agree so quickly. :smile: 
Yes, you can do it with a placholder element with "grow(1)".
BTW grow(1), grow(2), etc also works.

> I agree, finding all names without cheatsheet is difficult. But keeping names allow refer to css docs & reuse power of flexbox tutorials for free. Where i can see current names? I'd like to understand, what looks difficult from your point of view and you wish to improve.

We can document the CSS counterparts in the docs and in the comments of teh functions.

The current names are here: https://github.com/lvgl/lvgl/blob/a85f713d3a35bf6357abb82771165d2aa872650c/src/extra/layouts/flex/lv_flex.h#L85-L92

>> start an item in a new track (not in CSS but very useful)
> What is that?

It allows forcing wrap manually on a certain child.

puzrin, 2021-03-22:
> We can document the CSS counterparts in the docs and in the comments of teh functions.

Every extra overcomplication is bad. You will force users read & merge in head 2 docs instead of one.

This could work in different case - if you implement CSS feature "as is" AND add "your vision of problem" second. As extension, not as replacement.

> It allows forcing wrap manually on a certain child.

Still don't understand. I think in domain of "native" CSS flexbox (via css properties, assigned to both parent and children). It doesn't needs anything else IMO. Probably feature is not flexbox-specific but layout-specific?

kisvegabor, 2021-03-22:
> Every extra overcomplication is bad. You will force users read & merge in head 2 docs instead of one.

As you might see so far usually I'm open to discuss anything and consider any approach but in this case, I'm very convinced that using the obscure CSS names would be a bad idea and I really want to stick with more self-descriptive names.

Anyway, I wish it were our biggest question to decide :slightly_smiling_face: 
Let's talk about it later if it really turns out to be a problem. 

> Still don't understand. I think in domain of "native" CSS flexbox (via css properties, assigned to both parent and children). It doesn't needs anything else IMO. Probably feature is not flexbox-specific but layout-specific?

See this for example: https://stackoverflow.com/questions/19574851/how-to-specify-an-element-after-which-to-wrap-in-css-flexbox

---

I made a great progress with rebasing all the coordinates to styles. Everything works again including (flexbox and gird).

I wasn't sure about one thing: how to handle the default size of the widgets? Normally there is an `lv_obj_set_size()` in the constructor but there problems with it:
- It allocates a local style with width and height by default which means 36 extra byte usage per widget
- As the width/height is stored in the local style a normal style ("class") can't overwrite it

The obvious direction would be to define the width/height in the theme's style.  What do you think?

puzrin, 2021-03-22:
> See this for example: [stackoverflow.com/questions/19574851/how-to-specify-an-element-after-which-to-wrap-in-css-flexbox](https://stackoverflow.com/questions/19574851/how-to-specify-an-element-after-which-to-wrap-in-css-flexbox)

CSS has :nth-child or may be grid should be used. I would drop this feature until we have stable base.

> The obvious direction would be to define the width/height in the theme's style. What do you think?

IMO the obvious derection is to read CSS spec :), what are default values. AFAIK:

- `<div>` (block) has { width: 100%, height: auto }
- `<span>` (inline, inline-block) has { width: auto, height: auto }

Theme can override defaults if required, but should not be "replacement" of defaults.

> As the width/height is stored in the local style a normal style ("class") can't overwrite it

CSS has description of classes priority. Nobody forces you use local styles. For example, i use ordinary styles (according to BEM methodology). Locals (`style='...'` attribute) is usually applied for special cases like animations.

For our case, priority is:

- local style
- user styles
- default values

Note, at the time of "classes merge" result may be not numeric (auto, persents). Then those intermediate values are processed by layout manager.

kisvegabor, 2021-03-22:
> CSS has :nth-child or may be grid should be used. I would drop this feature until we have stable base.

It's not that important now, let's discuss it later.

> IMO the obvious derection is to read CSS spec :), what are default values. AFAIK:

What about `<input>` etc? There are some elements that do have a default size. I assume they come from the browser's style sheets.
 
> As the width/height is stored in the local style a normal style ("class") can't overwrite it

By this I meant, if the widget's constructor uses `lv_obj_set_size`, the width/height will be stored in local style.

embeddedt, 2021-03-22:
> What about <input> etc? There are some elements that do have a default size. I assume they come from the browser's style sheets.

Something like this is probably a good candidate for the theme's styles.

puzrin, 2021-03-23:
> What about `<input>` etc? There are some elements that do have a default size. I assume they come from the browser's style sheets.

https://stackoverflow.com/a/11846241/1031804 some defaults are crasy. I recommend to ignore special cases and orient to `100%` or `auto`, depending on type.

Also, i suggest to google for "css reset" - that's a list of rules, to normalize behavior in all browsers.

Tweaking defaults for some objects like `<input>` is not a problem. Override those in theme is not a problem too, if done properly (ask me about concrete cases).

For lvgl-specific widgets like `knob` - set defaults to anything resonable + tune in theme if required.

> By this I meant, if the widget's constructor uses lv_obj_set_size, the width/height will be stored in local style.

My answer is "widget's constructor should not set styles without needs". Any special cases should be discussed separate. I think it worth care about memory of base objects only (nobody needs 100 knobs, while `<div>` & `<span>` are user wide).

kisvegabor, 2021-03-23:
Ok, I'll check the default sizes in more detail soon.

There were some issues with layout updating but now the widgets demo (with a lot of nested layouts) is working too.

As the layout calculations are delayed and coordinates are cached rendering performance remained the same. However, the memory usage of the demo has increased from 23kB to 30kB on STM32F429. It's because storing the width + height in local style needs 28 bytes per object.  If the object is a grid item 64 extra bytes per widget (grid pos, span, place horizontally and vertically, 6 extra property). I'm not sure we should worry about it, I've mentioned it for the sake of completeness. 

An issue I ran into:
```c
obj = lv_btn_create(parent, NULL);
lv_obj_set_size(obj, 100, 100);   //Add width, height local style preprties
lv_obj_remove_style_all(obj);     //Remove the theme styles
```

However `lv_obj_remove_style_all` also removes `width` and `height` which is quite counterintutive.

puzrin, 2021-03-23:
> However, the memory usage of the demo has increased from 23kB to 30kB on STM32F429. It's because storing the width + height in local style needs 28 bytes per object. If the object is a grid item 64 [...]

I suggest to postpone all kinds of metrics and optimizations and focus on spec following first. Because trying to do both in the same time is a nightmare. I'm 100% sure, after you improve current api, many problem disappears, and optimizing the rest will be not a problem.

> However `lv_obj_remove_style_all` also removes `width` and `height` which is quite counterintutive.

I think, this comes from your current expectations from styles. For me, expecting css subset behaviour, no problem.

I understand, breaking habits is not nice. But situation looks similar with fonts rework - benefits are huge.

kisvegabor, 2021-03-23:
I was thinking about `left/right/top/bottom` and realized that they are no powerful enough. E.g. you can't do align to the top and horizontally center but with 10 px offset to the right.

Now we have `lv_obj_align()` which is very easy to use and supports all these alignments. I propose:
- adding `align_type` (e.g. top_left, bottom_mid) as style property with top_left default value
- considering x and y as an offset in the alignment

BTW, now we can add a very-similar-to-CSS `::after` and `::before`.

embeddedt, 2021-03-23:
> E.g. you can't do align to the top and horizontally center but with 10 px offset to the right.

I think it is possible with CSS transforms. It looks like this:

```css
.element {
    position: absolute;
    top: 0;
    left: 50%;
    transform: translateX(-50%) translateX(10px);
}
```

I'm not sure that this is really needed for most layouts. With flexbox and margin there is less of a need to manually position objects.

Alternatively, `lv_obj_align` could manually set `top` and `left` for the object's current size only.

puzrin, 2021-03-23:
>  was thinking about `left/right/top/bottom` and realized that they are no powerful enough. E.g. you can't do align to the top and horizontally center but with 10 px offset to the right.

See recipe from @embeddedt & add `margin-right: -10px` :)

Usually, to center floating block `{ left: 50%; margin-left: -50% }` used.

> These are no my habits. I'm trying to think with other people's minds, who just want to create 3 buttons and 3 texts and place them somewhere. These simple scenarios should be straightforward and work without extra "hacks".

I whould advice people with "straightforward scenarios for 3 buttons" stay with lvgl v5 :).

Please, do non-deviated api first, then extend with "simplifying helpers". In this case you are guaranteed to "not break architecture again and again".

kisvegabor, 2021-03-25:
> I think it is possible with CSS transforms. It looks like this:

That true, but IMO it'd be better with only `left/right` without extra hacks. 

> I'm not sure that this is really needed for most layouts. With flexbox and margin there is less of a need to manually position objects.

It can work centering or other common alignments but would be tricky for e.g. "bottom right with -10;-20 offset"

> Alternatively, lv_obj_align could manually set top and left for the object's current size only.

I see 2 problems with it:
- Not updated automatically if the object's size changes (e.g. label's text)
- Not updated automatically if the parent's size changes

> See recipe from @embeddedt & add margin-right: -10px :)
Usually, to center floating block { left: 50%; margin-left: -50% } use

I think it also way too overcomplicated for such a simple task.

I really believe this "align" feature is very simple and usefule so finally I added it. I am quite sure it won't cause any trouble. But if so, it will be a meaningful lesson to me...

kisvegabor, 2021-03-25:
I've removed the `content` style properties because they can be replaced by adding custom drawers to the events.

puzrin, 2021-03-25:
> I think it also way too overcomplicated for such a simple task.

This is usually to center absolutely positioned blocks (floating out of layout). For regular blocks there are multiple possibilities https://www.w3schools.com/css/css_align.asp

You asked how to do crazy thing (center + offset), and got crazy answer :). IRL things are not so bad, especially with flexbox.

kisvegabor, 2021-03-26:
I think we could talk about it very long, but if there is no serious objection I think we can go with `align`. :slightly_smiling_face: 

---

BTW, what do you think what is the CSS best practice to align 2 absolute positioned items next to each other? 
Concrete example: in a lot of UI designs there is a huge temperature number and a small "°C" text aligned to the right top corner of the numbers. But as the number changes "°C" needs to be realigned. 
I usually do it with `lv_obj_align(unit_label, num_label, LV_ALIGN_OUT_TOP_RIGHT, 0, 0)`. A better solution would be to create a transparent parent for these 2 labels with a grid or flexbox but it seems overkill. Do you have any ideas?

puzrin, 2021-03-26:
> BTW, what do you think what is the CSS best practice to align 2 absolute positioned items next to each other?

AFAIK, that's common case for "floating menu" (while content scrolled, menu sticks on top)

Obvious approach is put those into container, and do absolute positioning of container only.

kisvegabor, 2021-05-14:
width, height, max-width, max-height, min-width, min-height, x, y are available in styles now so I close this issue.

closed 2021-05-14 by kisvegabor