lvgl/lvgl #2155: Remove copy parameter from all create functions

xiaoxiang781216, 2021-03-21:
### Introduce the problem

The copy parameter is not working yet in the create functions for the current mainline. And the community agree to drop this argument in v8 release, instead to v9.

### Examples and cases

### Suggested solution

So let's remove this argument now.

kisvegabor, 2021-03-21:
> And the community agree to drop this argument in v8 release, instead to v9.

@mair, are you ok with this approach?

Mair, 2021-03-24:
@xiaoxiang781216 and @kisvegabor If it needs to happen then OK but, there is going to be some cross over where the videos aren't going to make sense. We may need to pull the plug on the course until the videos are redone

kisvegabor, 2021-03-24:
> We may need to pull the plug on the course until the videos are redone

Isn't it enough to inform the users that the videos are about v7?

kisvegabor, 2021-03-25:
Done.

closed 2021-03-25 by kisvegabor

Mair, 2021-03-25:
@kisvegabor 
> Isn't it enough to inform the users that the videos are about v7?

I suppose you're right. but I can imagine that once v8 is released people will hold off buying an outdated course

kisvegabor, 2021-03-26:
Yes, maybe. But  I think it's still better than closing the academy at all.