lvgl/lvgl #2171: outline discussion

incity, 2021-03-29:
I review the outline drawing code and found that:
1. when i want to draw 1px outline, i have to set the outline_pad to 1.
2. when i set outline width to 1, and set outline_pad to 0, the outline width will become 2px.

but  if i want to draw 1px outline with zero padding,  how can I achieve it?

```
    lv_obj_t * btn = lv_btn_create(h, NULL);
    lv_btn_set_fit2(btn, LV_FIT_NONE, LV_FIT_TIGHT);
    lv_obj_set_width(btn, lv_obj_get_width_grid(h, disp_size <= LV_DISP_SIZE_SMALL ? 1 : 2, 1));
    lv_obj_t * label = lv_label_create(btn, NULL);
    lv_label_set_text(label ,"Button");

    lv_obj_set_style_local_radius(btn, 0, LV_STATE_DEFAULT, 1);
    lv_obj_set_style_local_bg_color(btn, 0, LV_STATE_DEFAULT, LV_COLOR_GREEN);
    lv_obj_set_style_local_border_width(btn, 0, LV_STATE_DEFAULT, 0);
    lv_obj_set_style_local_outline_width(btn, 0, LV_STATE_DEFAULT, 1);
    lv_obj_set_style_local_outline_color(btn, 0, LV_STATE_DEFAULT, LV_COLOR_RED);
    lv_obj_set_style_local_outline_opa(btn, 0, LV_STATE_DEFAULT, LV_OPA_COVER);
    lv_obj_set_style_local_outline_pad(btn, 0, LV_STATE_DEFAULT, 1);
```

[image]

```
static void draw_outline(const lv_area_t * coords, const lv_area_t * clip, const lv_draw_rect_dsc_t * dsc)
{
    if(dsc->outline_opa <= LV_OPA_MIN) return;
    if(dsc->outline_width == 0) return;

    lv_opa_t opa = dsc->outline_opa;

    if(opa > LV_OPA_MAX) opa = LV_OPA_COVER;

    /*Get the inner radius*/
    lv_area_t area_inner;
    lv_area_copy(&area_inner, coords);

    /*Extend the outline into the background area if it's overlapping the edge*/
    lv_coord_t pad = (dsc->outline_pad == 0 ? (dsc->outline_pad - 1) : dsc->outline_pad);
    area_inner.x1 -= pad;
    area_inner.y1 -= pad;
    area_inner.x2 += pad;
    area_inner.y2 += pad;

    lv_area_t area_outer;
    lv_area_copy(&area_outer, &area_inner);

    lv_coord_t width = (dsc->outline_pad == 0 ? (dsc->outline_width + 1) : dsc->outline_width);
    area_outer.x1 -= width;
    area_outer.x2 += width;
    area_outer.y1 -= width;
    area_outer.y2 += width;

    draw_full_border(&area_inner, &area_outer, clip, dsc->radius, true, dsc->outline_color, dsc->outline_opa,
                     dsc->outline_blend_mode);
}
```

embeddedt, 2021-03-29:
Original discussion here: https://github.com/lvgl/lvgl/pull/2067/files/74c15ab10bdcb1fb34b727a1d3b4a3219077ebaa#r574547073

I'm not sure that it's possible to use the outline this way with the current implementation.

incity, 2021-03-29:
okay, so i can set outline width to 1 and outline_pad to -1 to  get 1px outline with zero padding.
But it's kind of weird...

incity, 2021-03-29:
sorry, i was wrong,  set outline width to 1 and outline_pad to -1 will get 1px outline with -1 padding. still can not get 0 padding

kisvegabor, 2021-03-29:
I've read the conversation again and this is not clear to me:

>@kisvegabor
> Or maybe it'd be even better to no change the width at all. We can say that pad=0 means on the edge.
>
> @embeddedt
> I think they both have to be changed together, otherwise the switch border looks a lot smaller than it should, so I went with your original suggestion (change only when pad == 0).

@embeddedt Why the outline width affects the visible borde width? The outline grows outside. So once it's drawn exactly "on" the edge it will cover only the edge pixels of the border independently of its width.

embeddedt, 2021-03-29:
I'm now forgetting exactly what I meant :laughing:, but I think I discovered that if only the position was changed (and not the width), the outline appeared to thinner by 1 pixel because the switch content covered it.

kisvegabor, 2021-03-31:
Actually, the border will look a little bit thinner because the outline is drawn above the border. 

So what about "*We can say that pad=0 means on the edge."*?

incity, 2021-04-01:
maybe the border means still on the widget, but outline means outside the widget. It's easier to understand. 
so outline_pad should means the distance between border and outline

kisvegabor, 2021-04-06:
So we have 3 options:
1. In which pad = 0 causes artifacts
2. In which the outline can look thinner if it has the same color as the background or the border
3. In which width = 1 is not possible. 

I still vote for 2). We can add it in v8 to keep the backward compatibility of v7. I think in the practice it means only +1 width if it looks too thin. 

What do you think?

embeddedt, 2021-04-06:
Okay, let's do 2, as we can easily adjust the default width of the outline in the theme for v8.

kisvegabor, 2021-04-12:
I've added it.

closed 2021-04-12 by kisvegabor

incity, 2021-04-13:
@kisvegabor @embeddedt in v7 still can not get 1px width of outline?

embeddedt, 2021-04-13:
I'm not sure there is a way to change the behavior in v7 without affecting existing applications, so probably not, unfortunately.

kisvegabor, 2021-04-13:
I also think that it'd affect existing apps so we shouldn't add it to v7. :slightly_frowning_face:

embeddedt, 2021-04-13:
@incity For v7, you can probably just apply [the change](https://github.com/lvgl/lvgl/commit/a6d3eff4db282c3c74c578bf363ed2eb30fce5b4) yourself to your codebase, as I don't think that section of the rendering code has changed much between v7 and v8.

commit a6d3eff4 (2021-04-12): fix(draw_outline) do not change the outline_width based on the outline_pad
fixes: #2171