lvgl/lvgl #2176: [v7] Button matrix does not initialise alignment member (undefined behaviour)

detly, 2021-03-31:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version - *no, see below*
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) - *not really possible*

I tried to check if this was present in `master` but the code structure's a bit different and I couldn't find the relevant part of the data structure.

Since it's undefined behaviour specifically in C that doesn't usually lead to externally visible behaviour, the simulator isn't really going to show this.

### Describe the bug

Button matrices do not initialise their `ext->align` member, leading to undefined behaviour.

### To Reproduce
```c
lv_obt_t * bm = lv_btnmatrix_create(parent, NULL);
lv_btnmatrix_set_align(bm, LV_LABEL_ALIGN_LEFT);
```

Run code that does this under Valgrind and it will flag an access to `ext->align` in `lv_btnmatrix.c` at line 435 (depending on revision).

This is because the `align` parameter of `lv_btnmatrix_set_align()` is compared to `ext->align`, where `ext` is the result of `lv_obj_get_ext_attr(btnm)`, but `ext->align` was never set. You will also see a similar issue if you never set the alignment and try to draw the widget.

A workaround is:

```c
lv_btnmatrix_ext_t * ext = lv_obj_get_ext_attr(bm);
ext->align = LV_LABEL_ALIGN_AUTO;
lv_btnmatrix_set_align(bm, LV_LABEL_ALIGN_LEFT);
```

### Expected behavior

No undefined behaviour.

embeddedt, 2021-03-31:
Fixed for v7 in https://github.com/lvgl/lvgl/commit/08b3fe1f11c8f6e77de3de46683b38e91d83c430.

embeddedt, 2021-04-14:
I reviewed the v8 button matrix implementation and it doesn't store the alignment the same way, so I think this bug won't apply there. Closing as I think this is solved for v7.

closed 2021-04-14 by embeddedt