lvgl/lvgl #2177: line width is 0 by default for widget line

ZhaoQiang-b45475, 2021-03-31:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Hi,
The default line width of line is 0, so if user create a line and don't set line width,
the widget will not be shown which will make them confused.

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2021-03-31:
Do you use v7 or v8 from `master`?

ZhaoQiang-b45475, 2021-04-06:
Hi Kisvegabor,
I am using tag v7.11.0

kisvegabor, 2021-04-08:
Thanks, fixed here: 30251017

kisvegabor, 2021-05-14:
As the issue should be fixed I close it.

If it's still not working as expected please comment here and reopen the issue.

commit 30251017 (2021-04-08): fix(material theme) add default line style with 1 px width
fixes #2177
closed 2021-05-14 by kisvegabor