lvgl/lvgl #2178: Margin and Padding Documentation is confusing

cebrax, 2021-03-31:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The documentation, API documentation and code comments for "lv_obj_set_width_margin", "lv_obj_set_height_margin", "lv_obj_set_width_fit" and "lv_obj_set_height_fit".

[image]
[image]

In summary they are explained as:
``
new_width = left_margin + object_width + right_margin
new_width = left_pad + object_width + right_pad
``

However, in code lv_core\lv_obj.c line 939 and 953:
``
The object width will be `obj_w = w - margin_left - margin_right`
``
``
The object height will be `obj_h = h - margin_top - margin_bottom`
``

kisvegabor, 2021-04-02:
It really sounds complicated but looks correct.

Form:
> Styles can add Margin to the objects. Margin tells that "I want this space around me". To set width or height reduced by the margin lv_obj_set_width_margin(obj, new_width) or lv_obj_set_height_margin(obj, new_height). In more exact way: new_width = left_margin + object_width + right_margin.
[image]

```
new_width = left_margin + object_width + right_margin

Can be rearranged as:
object_width = new_width - left_margin - right_margin;
```
[image]
Tells teh same:
```
obj_w = w - margin_left - margin_right
```

The difference is one tells how the `object_width` is calculated, and the other tells how the width you are set is composed.

cebrax, 2021-04-19:
Oh, I see that now.. But shouldn't this be consistent across different documentation sources?

kisvegabor, 2021-04-21:
I agree it'd be better to follow the same way of thinking in both places.
Could you send a Pull request with to `widgets/obj.html`?

embeddedt, 2021-04-21:
@kisvegabor I haven't disabled the v7 build script yet so it will be necessary to update the Markdown files.

closed 2021-05-26 by lvgl-bot