lvgl/lvgl #2183: Linemeter - rectangle width setting

lollopes, 2021-04-01:
Would it be possible to change the width of the rectangle in the linemeter ? I am able to change de width only for the animation part but not for when a value is not being set.

kisvegabor, 2021-04-01:
Which rectangle do you mean? Please send an image and mark the rectangle.

lollopes, 2021-04-01:
Hi Gabor, 

The red lines you can see are created during the animation when the set_value method is called. I have done this by applying a style with a line_width of 2 to the linemeter. Nevertheless, it only changes it when the value is set but not for the default status. In other words, in the picture below I would like to have the grey rectangle as small as the red lines. 

[image]

Thanks for the fast reply

kisvegabor, 2021-04-01:
You can do it with 
```c
    lv_obj_set_style_local_scale_end_line_width(lmeter, LV_LINEMETER_PART_MAIN, LV_STATE_DEFAULT, 1);
```

Or 
```c
    lv_style_set_scale_end_line_width(&style, LV_STATE_DEFAULT, 1);
```

lollopes, 2021-04-01:
That's perfect. I guess you can close the issue. Thank you for your time

closed 2021-04-01 by lollopes

kisvegabor, 2021-04-01:
You are welcome :)