lvgl/lvgl #2184: lv_linemeter don't show value 0 correctly

asypost, 2021-04-01:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version v7.11.0
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

lv_linemeter don't show value 0 correctly.
[image]
you can see a  block on the left bottom which should not be exists. 

### To Reproduce

```c
meter = lv_linemeter_create(lv_scr_act(), NULL);
lv_obj_set_width_fit(meter, LV_HOR_RES_MAX);
lv_obj_set_height(meter, LV_VER_RES_MAX - 18);
lv_linemeter_set_scale(meter, 142, 12);
lv_linemeter_set_range(meter, 0, 800);
lv_obj_set_y(meter, -15);
lv_obj_set_x(meter, 13);
lv_linemeter_set_value(meter, 0);
```

### Expected behavior

The meter should not show any block  when the value set to zero.

### Screenshots or video
See the picture provided up.

kisvegabor, 2021-04-06:
I've fixed it in `release/v7`

closed 2021-04-06 by kisvegabor

asypost, 2021-04-07:
But I'm using v7.11.0

kisvegabor, 2021-04-07:
You can safely update lvgl from relase/v7. There are only a few bug fixes.

asypost, 2021-04-07:
OK,thanks.I was confused by the meaning of release/v7,now I know that is a branch not a RELEASE in github releases page.

kisvegabor, 2021-04-07:
Just to be sure, I meant this branch: https://github.com/lvgl/lvgl/tree/release/v7

asypost, 2021-04-08:
> Just to be sure, I meant this branch: https://github.com/lvgl/lvgl/tree/release/v7

Thanks a lot.

asypost, 2021-04-19:
BTW,when will it be available as stable release?

kisvegabor, 2021-04-21:
See here https://github.com/lvgl/lvgl/issues/2144#issuecomment-822327936

commit 15976bdb (2021-04-06): fix(indev) provide raw old points to read_cb, not rotated points (#2180)

fixes #2184