lvgl/lvgl #2188: png not rendering properly

lollopes, 2021-04-06:
Hi, 

I have been using https://lvgl.io/tools/imageconverter to conver a png image to a .c file. 

The image I want to load is shown below:

[image]

When I load the image in my c file I end up with the following result 

[image]

Do you have any suggestions of why this is happening ?

embeddedt, 2021-04-06:
The PHP converter currently in use on the website doesn't handle indexed PNGs properly. Converting with this converter should work: https://embeddedt.github.io/lv_img_conv/

kisvegabor, 2021-04-06:
@embeddedt 
What's the current state of your converter? It seems only the binary format is missing.
Does dithering work?

embeddedt, 2021-04-06:
Everything (including dithering) should be working, except for binary support. At the moment there are a number of assumptions that the output is going to be to a C file, so I need to refactor things a bit for binary output to work (haven't had a chance to yet).

lollopes, 2021-04-07:
Hi, I used the converter suggested by embeddedt but nothing changed at first. I was trying to render a black and white arrow as you could see from the start of the thread, I managed to produce the right .c file by selecting "indexed 2 colors" for the color format and by allowing dithering. I believe we can close the issue. 

Thank you for your fast answers and support.

closed 2021-04-07 by lollopes