lvgl/lvgl #2190: Displayed area is abnormal

zhwxChaos, 2021-04-08:
### Describe the bug
Run the program on STM32F429.
 Collect data, display it to the graph, then clear the graph after reaching a certain amount, collect it again, and keep looping. 
Suddenly the screen display is abnormal and becomes a "snowflake screen". 
Tracking found that the display area is out of bounds.

[image]

lvgl version:V7.7.0

Hi @kisvegabor , Do you know what caused it?

kisvegabor, 2021-04-08:
It seems like a memory corruption somewhere.

Please update to v7.11, and enable all asserts in lv_conf.h.

zhwxChaos, 2021-04-09:
> enable all asserts in lv_conf.h.

When an error occurred, I enable all assert but  LV_USE_ASSERT_MEM_INTEGRITY.  And Useless information in the print log.

`#define LV_USE_DEBUG        1
#if LV_USE_DEBUG

#define LV_USE_ASSERT_NULL      1
#define LV_USE_ASSERT_MEM       1
#define LV_USE_ASSERT_MEM_INTEGRITY       0
#define LV_USE_ASSERT_STR       0
#define LV_USE_ASSERT_OBJ       1
#define LV_USE_ASSERT_STYLE     1

#endif /*LV_USE_DEBUG*/`

and 
`#define LV_USE_LOG      1
#  define LV_LOG_LEVEL    LV_LOG_LEVEL_WARN
#  define LV_LOG_PRINTF   1`

kisvegabor, 2021-04-09:
`LV_USE_ASSERT_MEM_INTEGRITY` should be enabled too.

closed 2021-05-14 by kisvegabor