lvgl/lvgl #2191: lv_list_clean won't clean the items of list

szsheep, 2021-04-08:
in my project, I need to clean the list sometime.
while I use lv_list_clean to clean list, and then I add new items in list,
however, while I focus some btn item, program crashed in lv_list file.

I found that, lv_list_clean just to clean the obj of scrollable of list, not 
the btn in list. 

so, I change my way to use lv_list_remove to remove btn item one by one,
and it did well.

hence, I deeply doubt that lv_list_clean function was doing the right job.

Please have a check, thanks a lot

kisvegabor, 2021-04-08:
Hi,
Does it work if you add this to `lv_list_clean`:
```c
    lv_list_ext_t * ext = lv_obj_get_ext_attr(list);
#if LV_USE_GROUP
     ext->last_sel_btn = NULL;
#endif
     ext->act_sel_btn = NULL;
```

kisvegabor, 2021-05-14:
I close this issue because there was no activity here for a while.

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2021-05-14 by kisvegabor

bhupiister, 2022-05-14:
I am getting this error when i try to clean the list and add new elements.

Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400ece12  PS      : 0x00060f30  A0      : 0x801025f0  A1      : 0x3ffb5320
A2      : 0x3ffbe180  A3      : 0x3ffb5454  A4      : 0x3ffb545c  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffb545c  A8      : 0x0000ffff  A9      : 0x0000ffff
A10     : 0x3ffbe180  A11     : 0x00000000  A12     : 0x0000808e  A13     : 0x0000808e
A14     : 0x3ffca46c  A15     : 0x00000080  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40085f3d  LEND    : 0x40085f45  LCOUNT  : 0x00000027

PC: 0x400ecd7e: lv_draw_label at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_draw/lv_draw_label.c line 123
EXCVADDR: 0x00000000

Decoding stack results
0x400ecd7b: lv_draw_label at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_draw/lv_draw_label.c line 119
0x40102559: lv_label_design at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_widgets/lv_label.c line 1105
0x400e9e1e: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 625
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9e63: lv_refr_obj at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 665
0x400e9ed6: lv_refr_obj_and_children at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 567
0x400ea162: lv_refr_area_part at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 503
0x400ea2d9: lv_refr_area at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 459
0x400ea459: lv_refr_areas at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 381
0x400ea486: _lv_disp_refr_task at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_core/lv_refr.c line 199
0x400f5fcd: lv_task_exec at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_misc/lv_task.c line 398
0x400f60c5: lv_task_handler at .pio/libdeps/m5stack-core2/lv_arduino/src/src/lv_misc/lv_task.c line 142
0x400d77ea: SetupUI(void*) at src/main.cpp line 3426

@kisvegabor Is this what you are suggesting?

```
void lv_list_clean(lv_obj_t *list)
{
    lv_list_ext_t *ext = lv_obj_get_ext_attr(list);
#if LV_USE_GROUP
    ext->last_sel_btn = NULL;
#endif
    ext->act_sel_btn = NULL;

    LV_ASSERT_OBJ(list, LV_OBJX_NAME);

    lv_obj_t *scrl = lv_page_get_scrollable(list);
    lv_obj_clean(scrl);
}
```

embeddedt, 2022-05-14:
I believe that's correct, yes.

bhupiister, 2022-05-14:
On further investigation it was revealed that the problem is not while clearing the list but when the list it too long and take time to render. Best solution I found was to take semaphore while creating list(adding new elements to it) and only after that give back semaphore to lv task handler. 
Problem occurs when lv task handler tries to update while the elements are still being added to the list.
I'll post my working code here tomorrow morning.

embeddedt, 2022-05-15:
It is mandatory to use a mutex or semaphore in any location where you access LVGL from multiple threads, as the library doesn't have any thread safety in it.