lvgl/lvgl #2192: how to swipe down  son obj  to father obj

rongyunxiang, 2021-04-12:
### Introduce the problem

### Examples and cases

 father page: obj
 lv_obj_t son_obj = lv_cont_creat(obj, NULL);
 lv_obj_t son_a = lv_cont_creat(obj, NULL);
 lv_obj_t son_b= lv_cont_creat(obj, NULL); 
...

lv_obj_set_event_cb(son_obj , callback);

void callback()
{
 ...
goto page (son_obj);
lv_obj_del_async(obj);
...
}

son obj page:
lv_obj_set_event_cb(obj, gestureCapture);

### Suggested solution
I tried to  not set full screen:
 lv_obj_align(obj, NULL, LV_ALIGN_IN_TOP_LEFT, 0, 10);  
It  works better ,but when I use mouse to pull down from y 10-50 , it doesn't work

rongyunxiang, 2021-04-13:
solved

embeddedt, 2021-04-14:
Since the issue is solved I will close it.

closed 2021-04-14 by embeddedt